stlink-org/stlink #289: STM32F072 discovery kit and unknown chip id

derleder, 2015-01-07:
Hello, I try to get the current version of stlink working with the STM32F072 discovery kit (http://www.st.com/web/en/catalog/tools/FM116/SC959/SS1532/PF259724?icmp=stm32f072-discovery_pron_pr_jan2014&sc=stm32f072discovery-pr) but the output says the following:

john@qwerty:~/src/stlink-priv$ ./st-util -m -v99
2015-01-07T16:58:39 DEBUG src/stlink-common.c: stlink current mode: dfu
2015-01-07T16:58:39 INFO src/stlink-usb.c: -- exit_dfu_mode
2015-01-07T16:58:39 DEBUG src/stlink-common.c: *** stlink_exit_dfu_mode ***
2015-01-07T16:58:40 DEBUG src/stlink-common.c: stlink current mode: mass
2015-01-07T16:58:40 DEBUG src/stlink-common.c: *** stlink_enter_swd_mode ***
2015-01-07T16:58:40 DEBUG src/stlink-common.c: *** looking up stlink version
2015-01-07T16:58:40 DEBUG src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)
2015-01-07T16:58:40 DEBUG src/stlink-common.c: stlink pid     = 0x3748
2015-01-07T16:58:40 DEBUG src/stlink-common.c: stlink version = 0x2
2015-01-07T16:58:40 DEBUG src/stlink-common.c: jtag version   = 0x11
2015-01-07T16:58:40 DEBUG src/stlink-common.c: swim version   = 0x0
2015-01-07T16:58:40 DEBUG src/stlink-common.c:     notice: the firmware doesn't support a swim interface
2015-01-07T16:58:40 INFO src/stlink-common.c: Loading device parameters....
2015-01-07T16:58:40 DEBUG src/stlink-common.c: *** stlink_core_id ***
2015-01-07T16:58:40 DEBUG src/stlink-common.c: core_id = 0x0bb11477
2015-01-07T16:58:40 DEBUG src/stlink-common.c: *** stlink_read_debug32 0 is 0xe0042000
2015-01-07T16:58:40 DEBUG src/stlink-common.c: *** stlink_read_debug32 0 is 0x40015800
2015-01-07T16:58:40 WARN src/stlink-common.c: unknown chip id! 0
2015-01-07T16:58:40 DEBUG src/stlink-common.c: *** stlink_close ***

With the STM32 ST-LINK Utility for Windows it works using SWD as Connection Protocol and Connect Under Reset as Connection Mode (Device ID is 0x448). 
According to the output I guess stlink knows to use SWD and as there is just a no-reset flag the default behaviour should be Connect Under Reset. 
Does anyone have any advice to get stlink working with this discovery board?

ziutek, 2015-02-13:
Similar problem with STM32L1-Discovery after st-link firmware update. Proper chip ID is 0x00000416 but now st-util says it is 0x20002d28:

2015-02-13T13:29:16 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)
2015-02-13T13:29:16 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)
2015-02-13T13:29:16 DEBUG src/stlink-common.c: *** looking up stlink version
2015-02-13T13:29:16 DEBUG src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)
2015-02-13T13:29:16 DEBUG src/stlink-common.c: stlink pid     = 0x3748
2015-02-13T13:29:16 DEBUG src/stlink-common.c: stlink version = 0x2
2015-02-13T13:29:16 DEBUG src/stlink-common.c: jtag version   = 0x17
2015-02-13T13:29:16 DEBUG src/stlink-common.c: swim version   = 0x0
2015-02-13T13:29:16 DEBUG src/stlink-common.c:     notice: the firmware doesn't support a swim interface
2015-02-13T13:29:16 INFO src/stlink-common.c: Loading device parameters....
2015-02-13T13:29:16 DEBUG src/stlink-common.c: *** stlink_core_id ***
2015-02-13T13:29:16 DEBUG src/stlink-common.c: core_id = 0x2ba01477
2015-02-13T13:29:16 DEBUG src/stlink-common.c: *** stlink_read_debug32 20002d28 is 0xe0042000
2015-02-13T13:29:16 WARN src/stlink-common.c: unknown chip id! 0x20002d28
2015-02-13T13:29:16 DEBUG src/stlink-common.c: *** stlink_close ***

st-flash, st-info works without any problem:

$ st-info --chipid
Error: could not open stlink device
Error: could not open stlink device
0x0416

derleder, 2015-02-16:
I was working with that one before and know that it's working despite the unknown chip id. Yet it receives an id != 0 which isn't such a big issue unlike for the STM32F072.

I moved on to the STM32F4-Discovery due to multiple reasons (and it's working great) so there's no need for a solution for me anymore but there was a little walkaround for the id problem: I used a virtual windows machine and connected with the STM32 ST-LINK Utility to the chip and during that state it was possible to flash and debug it (with Eclipse) in Linux once - after that you need to reconnect with the utility. It's not good with all the glitching between 2 Systems but it helped in the first place

mrechte, 2015-07-26:
Hello. I have the same issue with STMF4 Discovery (STM32F429) which reports chipid 0x0419

xor-gate, 2016-05-14:
Could you try with latest release `v1.2.0` ?

xor-gate, 2016-05-20:
Closing due to inactivity, chipid 0x477 is supported. Test with `v1.2.0` or `master`. Feel free to open a new issue if the problem still remains. Thank you for your contribution.

closed 2016-05-20 by xor-gate