stlink-org/stlink #301: STM32F042 has 6k RAM, not 4.

rewolff, 2015-03-20:
My STM32F042 has 6k of RAM according to table 2 of the datasheet. 

This is so small I don't want to fork, patch, pull-request. So, here is the patch: 
http://prive.bitwizard.nl/sramsize_fix.diff

texane, 2015-03-20:
patch applied, thanks

rewolff, 2015-03-21:
My processor crashes currently... This of course would certainly happen if I'm using 6.5k out of 4k available. So double check: 

printf "%08x\n", *(int*) 0x20001800
Cannot access memory at address 0x20001800

I cannot access memory at mem_start+6k. 

And it is not that ST-UTIL is limiting access to beyond 6k, just because I patched it yesterday: 

(gdb) printf "%08x\n", *(int*) 0x200017f0
227543d4
(gdb) printf "%08x\n", *(int*) 0x200007f0
080000c0

Normally you'd say: trust the datasheet. But all documentation indicated that the '042 doesn't have an USB bootloader. Now that I have the chips...  it does...

closed 2015-03-21 by rewolff

tonyarkles, 2015-05-20:
You can't access 0x20001800 because that is the word *after* the 6k of ram. 0x200017fc is the last word of the 6k

rewolff, 2015-05-20:
yeah! the "cannot access" was the "expected behaviour", it is a way to verify that the processor refuses access at extactly 6k. Then the fact that it allows access at 6k minus a bit means it really does have 6k. And it often happens that they implement say 4k RAM and then this RAM is mirrored for say 128k because they don't (need to) decode the A12 - A16 address lines. So that's why I double checked that location 0x200017f0 contained different data than 0x200007f0. (I checked at "a few words before the end" instead of "the very last".