stlink-org/stlink #303: st-util -m goes in endless loop if port is in use.

rewolff, 2015-03-25:
I forgot to close the older stutil before starting a new one. I then got an endless loop of: "bind: port already in use". 

I've implemented the following workaround: 

diff --git a/gdbserver/gdb-server.c b/gdbserver/gdb-server.c
index b1cea5c..3b5bd33 100644
--- a/gdbserver/gdb-server.c
+++ b/gdbserver/gdb-server.c
@@ -209,7 +209,8 @@ int main(int argc, char** argv) {
 #endif
 
     do {
-        serve(sl, &state);
+        if (serve(sl, &state)) 
+               sleep (1); // don't go bezurk if serve returns with error
 
         /* Continue */
         stlink_run(sl);

which: 
* restarts immediatly if gdb closes the connection normally
* still continues to try if an error occurs (I'm not sure if ALL return (1) 's in serve() are errors. Existing behaviour should continue to work if say gdb stops unexpectedly. )
* doesn't peg the CPU at 100% if the situation occurs that I had: serve returns error because the bind didn't work.

xor-gate, 2016-05-05:
This is a unacceptable "hack-fix", please test with latest tools in master and latest stlink firmware (2.27). Open a new issue with more details if this problem still remains. Thanks!

closed 2016-05-05 by xor-gate

rewolff, 2016-05-05:
Good stuff that you're going through the old bug reports and closing them when no longer relevant. But maybe my fix is a bit "quick and dirty" but it provides a good lead for a better fix should that be required. 
Motivating me to provide such a patch does not start with closing the issue and calling it unacceptable. 

Just tested: 
  2016-05-05T11:50:54 WARN src/stlink-usb.c: libusb_claim_interface() failed
Problem fixed (differently). Good.

xor-gate, 2016-05-05:
I'm sorry but hardcoded sleeps are not the way to fix problems, never... I understand it doesn't motivate by closing, but someone had to comment on you much much earlier (when fixed/improved) and say the handling of libusb errors was fixed. Sorry!

Much better it is now printing warnings, still I'm not happy with the message itself because people dont even known stlink tools use libusb. As they just want to flash/debug their application. It would be better to print something like: `WARN Stlink found, but unable to claim (probably already in use?)`

rewolff, 2016-05-05:
Agreed. The new message was immediately clear to me, but I'm not average. I don't see why "wait-for-device-to-become-available" is so horribly wrong. But yes, for consistency with expectations and other programs, throwing a fatal error is more appropriate. 

Anyway, apparently the "loop" was there for a reason, not pegging the CPU but still retrying is a minimally invasive solution. Oh. well. Closed. finished. done! :-)

texane, 2016-05-05:
Hi,

@Rewolff: I agree with you. We must encourage people to submit patches,
and this particular answer was a bit brutal. You did react well, thanks.
Some
developers take it very bad when criticized on their code :)

That being said, I am sure there was no bad intention in Jerry's decision.
The
list of open issues is very long, the task is not easy and repetitive ...
but largely
welcome.

At the end, the goal is that everyone enjoy what he is doing, so thank you
to
both.

xor-gate, 2016-05-05:
Yes @texane, i'm wrapping up the long +140 issues to see how we can improve and solve things with current code/situation. And there was no bad intention by closing this issue. I'm going to fixup the warning message to be more convenient.

rewolff, 2016-05-05:
@xor-gate keep up the good work! Having the issue-list too long makes it impossible to scan for important things. I did the same "cleaning up the issue list"  (but several times more issues) for the hugin project. At first it seems easy to just test-and-verify the issues, but that becomes impossible after a while and a few issues too many.

xor-gate, 2016-05-05:
Yeah @rewolff, that is why i started labeling like crazy, else it is unmaintainable. Don't hesitate to open new issues when finding something quirky as you and others help is highly appreciated! 

Kind regards,
Jerry