stlink-org/stlink #305: Option: ignore flash size reported by chip

LupinIII, 2015-03-26:
I have a small dev board from ebay (like this http://www.ebay.com/itm/321569700934 ; available from many sellers) with a F103C8T6. According to the specs it should have 64k of flash. That is also what st-flash reports when writing to it. 
But, and I don't know why this is actually possible, when I use the the "Flash Loader Demonstrator" from ST anf flash via the UART (using the bootloader pins) I can select a target "STM32F1_Med-density_128K" from a dropdown list (the only other entry being "STM32F1_Med-density_64K" for a F103C8T6). It is no problem to flash a bin (hex works of course, too, but with a bin it's easy to tell how much you really write) that is larger than 64k. It verifies the contents of the flash and whatever is written to the flash also works after that.
If I try to flash the same bin file with st-flash I get these messages:
INFO: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
INFO: Attempting to write 112892 (0x1b8fc) bytes to stm32 address: 134217728 (0x8000000)
ERROR: addr too high
and the write fails. Obviously it sees a bin that's too large for the flash and stops. But as seen before at least for the chips I have it's totally possible to write more than that sucessfully.

So it would be nice to have an option to either ignore the reported flash size completely or specify it on the command line (so verifying could still work and not stop reading prematurely)

cabo, 2015-09-20:
Here's my little hack for this:

```
diff --git INDEX/src/stlink-common.c WORKDIR/src/stlink-common.c
index 972c9f3..9f62252 100644
--- INDEX/src/stlink-common.c
+++ WORKDIR/src/stlink-common.c
@@ -475,6 +475,9 @@ int stlink_load_device_params(stlink_t *sl) {
         }
     } else {
         sl->flash_size = flash_size * 1024;
+        /* HACK for STM21F103C8T6 */
+        if (sl->flash_size == 64 * 1024 && getenv("C8T6HACK"))
+          sl->flash_size = 128 * 1024;
     }
     sl->flash_pgsz = params->flash_pagesize;
     sl->sram_size = params->sram_size;
```

So you call e.g., st-util as

```sh
C8T6HACK= st-util
```

and magically you get

```
INFO src/stlink-common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
```

I did this as an environment variable because setting an option may require changes in tons of Makefiles, build environments etc.; it is much less work to just set the environment variable before starting those.

cabo, 2015-10-06:
BTW, that hack is now available in my fork https://github.com/cabo/stlink
(I'm not doing a pull request yet because I'd like to understand whether and how much that hack should be generalized to other chips.)

potozi, 2016-05-13:
I have same problem with F100C6T6 which should have 4KB of RAM (and don't have TIM4) like written in datasheet. But he have 8KB RAM and TIM4. stlink detects this chip like Low-density with 4KB RAM and i can't access addresses higher than 0x20000FA0 in gdb.

I recompile source without this commit and it solved my problem.
https://github.com/texane/stlink/commit/d04fcf5cb68161d08ff98cd9d92b0d81b9242ce2

xor-gate, 2016-05-13:
@potozi thanks for reporting!

Testato, 2016-11-22:
i can confirm that also my BluePIll have 128KBytes

lesto, 2018-01-16:
same issue with a device id 0x413 rev Z, (STM32F407VG), it report 512kB, but from datasheet, websist and testing, it is both 512kB and 1MB flash. (program and verify from stllink utility ignore the flash size and report a successful flash, so despite both program reporting 512kB flash, it is 1MB).

Is there any switch in the standard program to ignore the flash size?

xor-gate, 2018-01-16:
No flash size must normally be read from the chip before making any decision, but currently texane/stlink tools don't do that.

lesto, 2018-01-16:
the st-flash tool refuse to flash the chip for me, V1.4.0

`st-flash 1.4.0
2018-01-16T16:13:08 INFO src/common.c: Loading device parameters....
2018-01-16T16:13:08 INFO src/common.c: Device connected is: F4 device, id 0x10016413
2018-01-16T16:13:08 INFO src/common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 16384 bytes
2018-01-16T16:13:08 INFO src/common.c: Attempting to write 429204 (0x68c94) bytes to stm32 address: 134348800 (0x8020000)
2018-01-16T16:13:08 ERROR src/common.c: addr too high
stlink_fwrite_flash() == -1
`

xor-gate, 2018-01-16:
@lesto don't recycle this issue. Please open a new issue and describe the details please.

Nightwalker-87, 2020-02-26:
Reopening as related to #516.

closed 2020-03-31 by Nightwalker-87