stlink-org/stlink #312: st-util problem

vmishev, 2015-05-17:
I've set up everything and here what shows up:

setting new configuration (0 -> 1)
2015-05-17T21:26:23 WARN src/stlink-usb.c: libusb_set_configuration() failed

I have fedora 21 machine and nucleo stm32f411re. what's wrong

framlin, 2016-01-24:
I have the same problem with Ubuntu 15.04 and a Nucleo-F401RE. Do you have any idea, what to do, to fix this?

xor-gate, 2016-05-03:
Probably you are not in the `dialout` or `plugdev` linux group as this is required for raw access because of libusb.

xor-gate, 2016-05-11:
@vmishev and @framlin could you test with latest tools from the master branch and verify if it works or is still broken?

xor-gate, 2016-05-20:
Closing due to inactivity of original issuer. Please test with latest `v1.2.0` or `master`. Feel free to open a new issue if there are still problems. Thank you for your contribution.

Nightwalker-87, 2020-03-22:
@vmishev @framlin: I know this is old, but can you tell us, if you could manage to get it working with Release v1.2.0?

Nightwalker-87, 2020-04-07:
As it appears, there is a feasible solution to this problem in #427.
Since the provided information in this issue is not sufficient and there is no possibility to reproduce, we can not investigate this any further. Thus I am closing this ticket now.

closed 2020-04-07 by Nightwalker-87