stlink-org/stlink #314: Failing to load on STM32F0Discovery

andrey-gvrd, 2015-05-22:
I'm trying to load an example from libopencm3. This is what happends when using gdb:

	[andrey@localhost miniblink]$ arm-none-eabi-gdb miniblink.elf
	GNU gdb (GDB) 7.6.2
	Copyright (C) 2013 Free Software Foundation, Inc.
	License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
	This is free software: you are free to change and redistribute it.
	There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
	and "show warranty" for details.
	This GDB was configured as "--host=x86_64-unknown-linux-gnu --target=arm-none-eabi".
	For bug reporting instructions, please see:
	<http://www.gnu.org/software/gdb/bugs/>...
	Reading symbols from /home/andrey/Fun/rust/libopencm3-examples/examples/stm32/f0/stm32f0-discovery/miniblink/miniblink.elf...done.
	(gdb) tar extended-remote :4242
	:4242: Connection timed out.

And with st-util nothing happens after trying to load the binary (miniblink.elf is in the folder):

	[andrey@localhost stlink]$ sudo ./st-util
	[sudo] password for andrey: 
	2015-05-22T21:39:36 INFO src/stlink-usb.c: -- exit_dfu_mode
	2015-05-22T21:39:36 INFO src/stlink-common.c: Loading device parameters....
	2015-05-22T21:39:36 INFO src/stlink-common.c: Device connected is: F0 device, id 0x20006440
	2015-05-22T21:39:36 INFO src/stlink-common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
	2015-05-22T21:39:37 INFO gdbserver/gdb-server.c: Chip ID is 00000440, Core ID is  0bb11477.
	2015-05-22T21:39:37 INFO gdbserver/gdb-server.c: Target voltage is 2924 mV.
	2015-05-22T21:39:37 INFO gdbserver/gdb-server.c: Listening at *:4242...
	load miniblink.elf

I'm on Fedora 21.
MCU on the board is STM32F051.

xor-gate, 2016-05-04:
Does this problem still exist when testing with latest master?

xor-gate, 2016-05-17:
Closing due to inactivity of issuer, please test against latest tools in `master` or `v1.2.0` (at time of writing). Feel free to open a new issue if the problem still exists.

Nightwalker-87, 2020-03-24:
@andrey-gvrd: What version of the `st-link tools` did you use? Please provide feedback to us.
I just saw there was a fix for the F0 in v1.4.0.

Nightwalker-87, 2020-07-18:
Can someone with this or a similar device please test this with branch `v1.3.0-patch` ?

Nightwalker-87, 2021-03-21:
As far as I can tell, the program has to be called with `load ./$ProgramName$` in this case as the reference to the current path would be missing else-wise. This is very likely an expected behaviour and not a bug.

Looking at this, we can safely close this issue as resolved.

closed 2021-03-21 by Nightwalker-87