stlink-org/stlink #317: FreeRTOS support

gduteil, 2015-06-11:
Hi,

First of all thank you this great tool.

I'm working on a FreeRTOS support (https://github.com/gduteil/stlink)
There are probably some issues, but it starts to work.

It has been tested on STM32F407 with FreeRTOS 8.2.0.

If people are interested in testing it, please let me know how it goes.

Thanks !

Guizmo

nvdl, 2015-06-12:
I can test it. Will soon order one STM32 Discovery board.

xor-gate, 2017-03-01:
This feature request is almost two years old and never merged back. I will now close this due to inactivity. Feel free to open a new issue or a pull request to contribute. Thanks in advance!

Nightwalker-87, 2020-03-24:
Is there _any_ interest for this in the community?
Please give some feedback.

Nightwalker-87, 2020-03-30:
Since no further demand has been registered for this feature, this ticket is now finally closed.

closed 2020-03-30 by Nightwalker-87