stlink-org/stlink #318: Multiple Programmers

trichner, 2015-06-30:
I have multiple STM32F0Discovery Boards wired up to an USB hub and would like to specify which one to flash (I'm using `st-flash main.bin 0x800000` right now).

Is there a way to specify the USB port to use? or even an actual selection?

texane, 2015-06-30:
Hi,

you can use the STLINK_DEVICE environment variable
to specify which device to use. VID:PID format.

trichner, 2015-07-14:
But all stlink programmers have the same VID:PID, they only differ in Bus/Device

Example `lsusb` dump:
```
Bus 002 Device 012: ID 0483:3748 STMicroelectronics ST-LINK/V2
...
Bus 001 Device 014: ID 0483:3748 STMicroelectronics ST-LINK/V2
```

trichner, 2015-07-23:
Well, if anybody ever has the same need and is looking for a solution: `OpenOCD` can use multiple devices/programmers by using the `hla_serial` command.

shkhaliq, 2016-04-07:
Any updates on this issue?

shkhaliq, 2016-04-08:
I got a workaround. Special thanks to @xor-gate for the `st-probe` tool. Use that tool to get the serial number.
 Note the last byte is a null character, So if the serial number is `483f6e06703f53483324096700` use `483f6e06703f534833240967`.

One of the caveats of getting the serial number is that you have to make sure the programmer is connected to the board and its in a connectable state. 

After that just invoke `st-flash --serial <serial_no> <your_command_here>`.

xor-gate, 2016-04-08:
@shkhaliq could you post your st-probe output, maybe it prints a spurious 00 as last byte? It seems I also have the same which adds the extra 00 byte:

```
jjacobs@workbench ~/repos/github/stlink/build $ ./st-probe
Found 2 stlink programmers
563f6706677256563113066700
         flash: 262144 (pagesize: 2048)
          sram: 65536
        chipid: 0x0414
         descr: F1 High-density device
493f6d06667048513615106700
         flash: 1048576 (pagesize: 16384)
          sram: 196608
        chipid: 0x0413
         descr: F4 device
```

xor-gate, 2016-05-02:
For your convenience `st-probe` has been merged into `st-info` -> `st-info --probe` and the hla serial string is also printed. When only one programmer is attached you can use `st-info --hla-serial`. Have fun! Maybe this issue can be closed then @trichner ?

trichner, 2016-05-02:
I ended up forking OpenOCD and fixing the horrible binary input there: https://github.com/trichner/openocd

xor-gate, 2016-05-14:
We need some reworking and documentation here.

GagoSoft, 2017-01-10:
Hallo 
the serialnumber is a very nice thing, but it works only on my STM32F0-discovery ST-Link V2 ... 
the standalone ST-Link V2 and the Nucleo-board-ST-LINK hast other serial number-formats.
by the way, the serial-numbers are usefull in st-util too. I integrated the serial for F0-Discovery into the gdb-server.c . It works on my single board, but it's not well testet...
`32a33
> #define SERIAL_OPTION 127
38a40,41
> static bool serial_specified = false;
> static char serialnumber[28] = {0};
76c79,84
<             ret = stlink_open_usb(st->logging_level, st->reset, NULL);
---
>             if(serial_specified){
>                 ret = stlink_open_usb(st->logging_level, st->reset, serialnumber);
>             }
>             else{
>                 ret = stlink_open_usb(st->logging_level, st->reset, NULL);
>             }
96a105
> 	  {"serial", required_argument, NULL, SERIAL_OPTION},
116a126,127
>         "  --serial <serial>\n"
>         "\t\t\tUse a specific serial number.\n"
171a183,195
>                 break;
>             case SERIAL_OPTION:
>                 printf("use serial %s\n",optarg);
>                             /** @todo This is not really portable, as strlen really returns size_t we need to obey and not cast it to a signed type. */
>                 int j = (int)strlen(optarg);
>                 if(j % 2 != 0) return -1;
> //                 memset(serialnumber, 0, sizeof(serialnumber));
>                 for(size_t k = 0; j >= 0 && k < sizeof(serialnumber); ++k, j -= 2) {
>                     char buffer[3] = {0};
>                     memcpy(buffer, optarg + j, 2);
>                     serialnumber[12 - k] = (uint8_t)strtol(buffer, NULL, 16);
>                 }
>                 serial_specified = true;`

greets Gabriel

xor-gate, 2017-01-10:
@GagoSoft there is already issue #417 about this problem but no fix yet. It should no be fixed in gdb-server.c but in the library itself.

GagoSoft, 2017-01-10:
the serial-number problem for the nucleo is quite easy, and allready solved in my local sources. I will post my patch soon.
st-info fails reading the serial of both of my standalone-ST-Link-V2. and leave them with a yellow LED ?!

GagoSoft, 2017-01-10:
here my patches for gdbserver.c and flash_opts.c for the serials of my nucleo-boards
[gdbserver.c.diff_and_flash_opts.c.diff.zip](https://github.com/texane/stlink/files/696088/gdbserver.c.diff_and_flash_opts.c.diff.zip)
the flash_opts.c.diff affects only two lines of code...

xor-gate, 2017-01-10:
Please make a pull request from your changes (https://help.github.com/articles/about-pull-requests/). Then you get the credits and the history is tracked.

GagoSoft, 2017-01-10:
sorry, I've never used git before ... :(
I just downloaded the sources from the git-website.
Do I have understand the act right?
* "git clone ... " the sources to my computer or make a branch first?
and then?

xor-gate, 2017-01-10:
* Fork the repository to your github account in the webgui (right corner of the texane/stlink project)
* Clone the git repository to your computer from your account
* Create a feature branch
* Commit your work in the branch
* Push the branch to github
* Create a pull request of the branch

I know this sounds like a lot of work (and to learn) for just a few lines of code, but that is the way how distributed contributions work with git (in comparision to subversion). This is some nice reading about git workflows: https://git-scm.com/book/en/v2/Distributed-Git-Distributed-Workflows

GagoSoft, 2017-01-10:
> st-info fails reading the serial of both of my standalone-ST-Link-V2. and leave them with a yellow LED ?!
...this behaviour is only seen, when no target is connected to the ST-Link (the chipid can't be read and the initialization of the st-Link fails). Not really a Problem.

xor-gate, 2017-01-10:
Hi all @GagoSoft has supplied a feature to control multiple programmers simultaneous. If there are any problems feel free to comment here.

WayneKeenan, 2017-07-23:
If `st-info --probe` can detect, say 4, programmers can it also display each serial the output for convenience?

xor-gate, 2017-07-24:
@WayneKeenan that is a limitation currently how probe is implemented. I have written [xor-gate/stlink2](https://github.com/xor-gate/stlink2) which is able to show the serials even when there are no microcontrollers attached.

Nightwalker-87, 2020-03-16:
Needs verification and a more detailled review.

Nightwalker-87, 2020-03-17:
Closed by #398 & #541.

commit 0106c538 (2017-01-10): Support serial numbers in st-util and st-flash (#541)

Support multiple stlink programmers by passing the serial number as a commandline argument. Fixes #318.
closed 2020-03-17 by Nightwalker-87