stlink-org/stlink #322: st-util returns "unknown chip id! 0" / STM32F429

mrechte, 2015-07-26:
Hello.
Board is STM32F4 Discovery kit for STM32F429

$ st-util
2015-07-26T16:28:00 INFO src/stlink-common.c: Loading device parameters....
2015-07-26T16:28:00 WARN src/stlink-common.c: unknown chip id! 0
$ st-info --chipid
0x0419

stlink build from 1.1.0 source

fbxy, 2015-11-25:
I'm facing the same problem in Mac OS X.

mrechte, 2015-11-25:
The problem is Nutx specific. Have a look here: http://www.nuttx.org/doku.php?id=wiki:howtos:jtag-debugging

xor-gate, 2016-05-14:
Could some of you verify it still exists with `v1.2.0` or `master` ?

kubark42, 2016-05-17:
I still get this after checking out latest master and copy-pastaing **all** text from the readme:

```
st-util
2016-05-17T14:47:32 INFO /Users/kenz/Documents/sandbox/stlink/src/common.c: Loading device parameters....
2016-05-17T14:47:32 WARN /Users/kenz/Documents/sandbox/stlink/src/common.c: unknown chip id! 0x3001641b
```

P.S. There doesn't seem to be a `--version` flag for st-flash.

xor-gate, 2016-05-17:
@kubark42 yes I know about `--version` as most people would expect. This is currently not implemented because of not very nice implemented commandline argument parsing and will be resolved in milestone `v2.0.0`. 

I'm trying to dig up the chipid `0x3001641b` from the datasheets but i'm unable to find it. Could you help me out?

kubark42, 2016-05-17:
@xor-gate Crap, I was trying to update this before anyone got a chance to resolve. I hadn't realized I was using an STM32F4Discovery board that still had its jumpers in place. It's no surprise that this was a GIGO situation.

This result should make you happier:

### STM32F427

```
 st-util
2016-05-17T14:57:24 INFO /Users/kenz/Documents/sandbox/stlink/src/common.c: Loading device parameters....
2016-05-17T14:57:24 INFO /Users/kenz/Documents/sandbox/stlink/src/common.c: Device connected is: F42x and F43x device, id 0x20016419
2016-05-17T14:57:24 INFO /Users/kenz/Documents/sandbox/stlink/src/common.c: SRAM size: 0x40000 bytes (256 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 16384 bytes
2016-05-17T14:57:24 INFO /Users/kenz/Documents/sandbox/stlink/src/gdbserver/gdb-server.c: Chip ID is 00000419, Core ID is  2ba01477.
2016-05-17T14:57:24 INFO /Users/kenz/Documents/sandbox/stlink/src/gdbserver/gdb-server.c: Target voltage is 2881 mV.
2016-05-17T14:57:24 INFO /Users/kenz/Documents/sandbox/stlink/src/gdbserver/gdb-server.c: Listening at *:4242...
```

### STM32F429
```
$ st-util
2016-05-17T14:58:05 INFO /Users/kenz/Documents/sandbox/stlink/src/usb.c: -- exit_dfu_mode
2016-05-17T14:58:05 INFO /Users/kenz/Documents/sandbox/stlink/src/common.c: Loading device parameters....
2016-05-17T14:58:05 INFO /Users/kenz/Documents/sandbox/stlink/src/common.c: Device connected is: F42x and F43x device, id 0x10036419
2016-05-17T14:58:05 INFO /Users/kenz/Documents/sandbox/stlink/src/common.c: SRAM size: 0x40000 bytes (256 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 16384 bytes
2016-05-17T14:58:05 INFO /Users/kenz/Documents/sandbox/stlink/src/gdbserver/gdb-server.c: Chip ID is 00000419, Core ID is  2ba01477.
2016-05-17T14:58:05 INFO /Users/kenz/Documents/sandbox/stlink/src/gdbserver/gdb-server.c: Target voltage is 2880 mV.
2016-05-17T14:58:05 INFO /Users/kenz/Documents/sandbox/stlink/src/gdbserver/gdb-server.c: Listening at *:4242...
```

So I can confirm that I'm not seeing the problem on an STM32F429.

xor-gate, 2016-05-17:
Yeah, those things happen. Still we are error-creating-humans 👍. Thanks for testing. I have cross-referenced about the commandline arguments and edited your post. @kubark42 basicly this means we can close this issue then?

closed 2016-05-17 by xor-gate

kubark42, 2016-05-17:
@xor-gate Thanks for the understanding, and glad after sorting it out I could help test. It does indeed look like there is full st-link support for the STM32F42x chips.

xor-gate, 2016-05-17:
Great!