stlink-org/stlink #331: Can't use StLink on OS X: libusb timeouts

rhaamo, 2015-08-18:
Hi,

I got approximately one response for -a very long shot of runs-, so StLink CANT be used on OSX...

```
➜  stlink git:(master) ✗ ./st-info --descr
libusb_handle_events() timeout
[!] send_recv
^C
```

Even tried stable libusb or HEAD, no success.

According to wiki https://github.com/texane/stlink/wiki "After downgrading the firmware, flashing works ST-Link/V2 debugger with downgraded V2.14.3 firmware".

Tried that and it doesn't works, also 2.14 is really old...

Also "openocd" doesn't works too.
(And that's not a problem with my adapter, it works perfectly on windows using stlink or linux with openocd or stlink)

Any plans to take a look about this issue ? Since this make any development on OS X impossible unless playing with two laptops only for StLink...

mlu, 2015-08-18:
I am using stlink on OSX 10.9.5. The board is a STM32F7 Discovery with the supplied firmware, and both flashing from st-flash and debugging with st-util works.
So stlink does run well on some combinations of OSX and firmware versions.

rhaamo, 2015-08-18:
Well here it doesn't works at all.

Also as you can probably understand, sticking with an *old* firmware and using recent MCU isn't the best recommended way to help working with them.

mlu, 2015-08-18:
I will test some other boards later today and also check firmware versions

My libusb setup from brew:

libusb: stable 1.0.19 (bottled), HEAD
Library for USB device access
http://libusb.info
/usr/local/Cellar/libusb/1.0.19 (11 files, 368K)
  Poured from bottle
From: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/libusb.rb
==> Options
--universal
	Build a universal binary
--with-default-log-level-debug
	Build with default runtime log level of debug (instead of none)
--without-runtime-logging
	Build without runtime logging functionality
--HEAD
	Install HEAD version

libusb-compat: stable 0.1.5 (bottled)
Library for USB device access
http://www.libusb.org/
/usr/local/Cellar/libusb-compat/0.1.5 (12 files, 132K) *
  Poured from bottle
From: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/libusb-compat.rb
==> Dependencies
Build: pkg-config ✘
Required: libusb ✔
==> Options
--universal
	Build a universal binary

rhaamo, 2015-08-18:
i only have "libusb: stable 1.0.19, HEAD // Built from source" here, also tried --HEAD but that doesn't changed anything.
My StLink is git-cloned and built handy (not using brew even if it seems to exist).

I've previously tried fw:
  - 2.J14.S0 (from wiki dropbox link)
  - 2.J15.S0 (same)
  - 2.J16.S0 (same)
  - 2.J17.S0 (same)
  - 2.J21.S0 (same)
  - 2.J24.S0 (from StLink windows utility, official)

No success at all. First plug in mac works, latests runs fails, openocd doesn't even get working on first launch.

```
➜  stlink git:(master) ✗ ./st-info --descr
F4 device (low power) - stm32f411re
➜  stlink git:(master) ✗ ./st-info --descr
libusb_handle_events() timeout
[!] send_recv
--stuck here, can only ^C--
```

I have only a stm32f4-discovery, I can try to see if a friend have another boards with on-board splittable StLink tonight.

I doesn't know StLink nor libusb code but if provided patchs/commits/whatever to see/display debugs things that can help i can do it.

mlu, 2015-08-18:
For some more debug info try
st-util -v99

rhaamo, 2015-08-18:
First run:

```
➜  stlink git:(master) ✗ ./st-util -v99
2015-08-18T10:49:09 DEBUG src/stlink-common.c: stlink current mode: dfu
2015-08-18T10:49:09 INFO src/stlink-usb.c: -- exit_dfu_mode
2015-08-18T10:49:09 DEBUG src/stlink-common.c: *** stlink_exit_dfu_mode ***
2015-08-18T10:49:09 DEBUG src/stlink-common.c: stlink current mode: mass
2015-08-18T10:49:09 DEBUG src/stlink-common.c: *** stlink_enter_swd_mode ***
2015-08-18T10:49:09 DEBUG src/stlink-common.c: *** looking up stlink version
2015-08-18T10:49:09 DEBUG src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)
2015-08-18T10:49:09 DEBUG src/stlink-common.c: stlink pid     = 0x3748
2015-08-18T10:49:09 DEBUG src/stlink-common.c: stlink version = 0x2
2015-08-18T10:49:09 DEBUG src/stlink-common.c: jtag version   = 0x15
2015-08-18T10:49:09 DEBUG src/stlink-common.c: swim version   = 0x0
2015-08-18T10:49:09 DEBUG src/stlink-common.c:     notice: the firmware doesn't support a swim interface
2015-08-18T10:49:09 INFO src/stlink-common.c: Loading device parameters....
2015-08-18T10:49:09 DEBUG src/stlink-common.c: *** stlink_core_id ***
2015-08-18T10:49:09 DEBUG src/stlink-common.c: core_id = 0x2ba01477
2015-08-18T10:49:09 DEBUG src/stlink-common.c: *** stlink_read_debug32 10006431 is 0xe0042000
2015-08-18T10:49:09 DEBUG src/stlink-common.c: *** stlink_read_debug32 200c000 is 0x1fff7a20
2015-08-18T10:49:09 INFO src/stlink-common.c: Device connected is: F4 device (low power) - stm32f411re, id 0x10006431
2015-08-18T10:49:09 INFO src/stlink-common.c: SRAM size: 0x20000 bytes (128 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 16384 bytes
2015-08-18T10:49:09 DEBUG src/stlink-common.c: *** stlink_reset ***
2015-08-18T10:49:09 INFO gdbserver/gdb-server.c: Chip ID is 00000431, Core ID is  2ba01477.
2015-08-18T10:49:09 DEBUG src/stlink-common.c: *** reading target voltage
2015-08-18T10:49:09 DEBUG src/stlink-common.c: target voltage = 2860mV
2015-08-18T10:49:09 INFO gdbserver/gdb-server.c: Target voltage is 2860 mV.
2015-08-18T10:49:09 INFO gdbserver/gdb-server.c: Listening at *:4242...
^C2015-08-18T10:49:17 DEBUG src/stlink-common.c: *** stlink_run ***
2015-08-18T10:49:17 DEBUG src/stlink-common.c: *** stlink_exit_debug_mode ***
2015-08-18T10:49:17 DEBUG src/stlink-common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2015-08-18T10:49:17 DEBUG src/stlink-common.c: *** stlink_close ***
```

Second run (not disconnected, just after ^C):
```
➜  stlink git:(master) ✗ ./st-util -v99
libusb_handle_events() timeout
[!] send_recv
2015-08-18T10:50:34 DEBUG src/stlink-common.c: stlink mode: unknown!
^C
```

Usb reported as:
```
            STM32 STLink:

              Product ID: 0x3748
              Vendor ID: 0x0483  (STMicroelectronics)
              Version: 1.00
              Serial Number: Iÿw?PgHW4? 
              Speed: Up to 12 Mb/sec
              Manufacturer: STMicroelectronics
              Location ID: 0x14130000 / 11
              Current Available (mA): 1000
              Current Required (mA): 100
              Extra Operating Current (mA): 0
```

Each time it "crashes" it just stay as-is, no output, nothing, can just ^C and reconnect and retry

rhaamo, 2015-08-18:
I've rebuilt libusb with **--with-default-log-level-debug**, here is the output of **st-info --descr** on first run and second run : https://gist.github.com/rhaamo/15d7cb2f7f4f7dc5c810

mlu, 2015-08-18:
I cannot replicate your problems on my system, tried with STM32F4 Discovery straight out of the package, works, upgraded to latest STLink firmware works.

(I do find some other weirdness but thats for a different thread)

rhaamo, 2015-08-19:
Tried a STM32L0538-DISCO (stock fw i think), and a StLink/v2 clone and same problem everytime, i got timeouts after second run...

mlu, 2015-08-19:
What osx version ?

rhaamo, 2015-08-20:
10.11 beta 4, i'm updating to beta 5 and will retry tonight.

avrhack, 2015-10-03:
I have the same issue with 10.11 release version. Basically st-link will not work with El Capitan which is a rather major issue as I can no longer do anything since upgrading.

mlu I'm guessing you're still on 10.10 or 10.9?

mlu, 2015-10-03:
Yes I am still at 10.9.5

cquartier, 2015-10-12:
just wanted to throw in here and say I'm experiencing the exact same problem with my STLink v2 mini. OSX 10.11 (15A284). I'm using OpenOCD, custom hardware with an STM32F411 core. Works once and then I have to power cycle the STLink in order to program again.

sajimon, 2015-10-21:
Same here. 
To add new info, my the STLink V2 have same behaviour on Ubuntu Linux 14.04 just after upgrade to El Capitan (on the same machine, mac mini) - so it may be something messed up on efi level.

Cheong2K, 2015-10-22:
Same problem, using OSX 10.11.1

zlowred, 2015-10-26:
As per libusb developers this is not a libusb issue https://github.com/libusb/libusb/issues/111

mcuee, 2015-11-24:
I have re-opened libusb issue 111.

mcuee, 2015-11-26:
As mentioned in https://github.com/libusb/libusb/issues/111, the work-around seems to be read the ST-Link version twice. Please try that. As per the report in the OpenOCD mailing list, that seems to sort out the problem.

mcuee, 2016-01-26:
As per report in libusb ticket, this is an Apple USB Stack bug and Apple Mac OS X A10.11.4 beta seems to sort out this issue.

Cheong2K, 2016-01-29:
Confirmed, fixed in OSX 10.11.4 beta.

user-don, 2016-02-10:
Is it worth updating to the beta on my main machine for this fix? Wondering if there are other stability issues to keep in mind.

xor-gate, 2016-05-03:
As mentioned by @Cheong2K i'm closing this for now. Reopen when there are still problems related to this issue.

closed 2016-05-03 by xor-gate

husnoo, 2016-07-04:
Same problem on 10.11.5 (15F34).

```
uname -a
Darwin Nawals-MacBook-Air.local 15.5.0 Darwin Kernel Version 15.5.0: Tue Apr 19 18:36:36 PDT 2016; root:xnu-3248.50.21~8/RELEASE_X86_64 x86_64

```
```
st-util -v99
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: stlink current mode: debug (jtag or swd)
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: stlink current mode: debug (jtag or swd)
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** stlink_reset ***
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** looking up stlink version
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: stlink pid     = 0x3748
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: stlink version = 0x2
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: jtag version   = 0x17
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: swim version   = 0x4
2016-07-04T11:06:46 INFO /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: Loading device parameters....
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** stlink_core_id ***
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: core_id = 0x00000000
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** stlink_read_debug32 e0042000 is 0xe0042000
2016-07-04T11:06:46 WARN /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: unknown chip id! 0xe0042000
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** stlink_close ***

```

xor-gate, 2016-07-04:
@husnoo thank you for your feedback, please open a new issue for this. Thanks in advance!