stlink-org/stlink #333: Add utility to flash to STM32 data EEPROM area

dflogeras, 2015-08-18:
It would be great to have a command line utility to flash a binary file directly to the data EEPROM in STM32 devices (ie address starting 0x8080000 on the STM32Lxx devices). I use this to store things like a device serial number, which is separate and distinct from the program EEPROM version.  Currently I write a small program to store what I need in the data EEPROM, then replace it with the real firmware, but it would be good to eliminate this step and just use stlink to do it.

niallp, 2015-12-15:
A related feature would be the capability to erase the EEPROM space, the bulk erase presently only clears the main FLASH.

manuvelayudhan, 2018-02-27:
Had a similar requirement for an STM32L051 device. While searching for methods to do the same, came across the following..
https://www.purplealienplanet.com/node/56

Based on the approach mentioned in that page, made some modifications in order to use the existing flash write functions to program the EEPROM. What I did was,

1. Added EEPROM base address, size and page size fields to the device parameter structures.
2. In main(), after ruling out flash & RAM as destinations for write operation, one more check is performed to see if it is a valid EEPROM address for the connected device.
3. If yes, the flash params (base, size and page size) of the _stlink structure (sl) are overwritten with the EEPROM params and the existing flash write APIs are used for writing to EEPROM.

I was able to successfully write to the EEPROM of the following STM32L0 devices
     a. STM32L031K6 (Nucleo 32 board)
     b. STM32L051K8 (custom board)

The changes are available in the fork I created.
     https://github.com/manuvelayudhan/stlink

If someone can review the same to verify if this is the right approach to the problem, it will be helpful.

Possible improvements are

1. Instead of using the existing flash write APIs, a new set of dedicated EEPROM write APIs can be added.
2. Current version will work for STM32L0 category 2/category 3 devices only.

niallp, 2018-06-25:
@manuvelayudhan  your fork works for me to allow for eeprom writing, would be nice to see it pulled into mainline.

manuvelayudhan, 2018-06-26:
Not sure if the current implementation is the best possible / correct one.. Any suggestions to improve it? Or should I just create a pull request from the current state?

niallp, 2018-06-26:
@manuvelayudhan ... I haven't looked too deeply but it seems a fairly minor tweak for useful functionality, all I'd suggest is merging in master and making sure nothing breaks in the latest (didn't see anything obvious).

Nightwalker-87, 2020-03-20:
@dflogeras @manuvelayudhan @niallp: I believe this feature is still missing. What is the current state of this for all of you?

dflogeras, 2020-03-20:
I agree this is unimplemented, but I have no opinion on how to proceed.  In recent times, when I've needed said function, I've had access to a windows computer and just used ST's utility to do so.

Nightwalker-87, 2020-03-20:
I see, but I may be useful for others. This makes me feel like we should leave this open, but of course can keep you out of the topic for the ongoing conversation.

niallp, 2020-03-23:
On 2020-03-20 9:02 a.m., Nightwalker-87 wrote:
>
> I see, but it may be useful for others. This makes me feel like we 
> should leave this open, but of course can keep you out of the topic 
> for the ongoing conversation.
>
I've used the patch by Manu when I've needed it though haven't touched 
it for a while. It would be nice to see the functionality in the mainline.

             ... Niall

Nightwalker-87, 2020-03-23:
This should better go into the code. I'm not so happy if such workarounds remain present for too long. As it reads, this could also be an ordinary feature.

Nightwalker-87, 2021-05-02:
@Ant-ON: Is there still a reasonable way of implementing this after the recent contributions to the `st-flash` source code?
Unfortunately @manuvelayudhan seems to be inactive by now, but the patch is still present in his fork.

Ant-ON, 2021-05-04:
@Nightwalker-87 Doesn't look like a good implementation https://github.com/manuvelayudhan/stlink/commit/fc1b5fb2188f3bff548536dbabdffe4283546326

Nightwalker-87, 2021-05-04:
Hm, ok. Thanks for the feedback. Maybe someone else has a better idea then.

Nightwalker-87, 2022-08-27:
Closing this issue due to inactivity and in relation to the given feedback above.

closed 2022-08-27 by Nightwalker-87