stlink-org/stlink #334: USB hard reset causes total failure on STM32L053-Disco

lcrocker, 2015-08-19:
I'm using an STM32L053 Discovery board, connected via USB to a PC running Ubuntu 14.04.

An earlier version of st-flash and other tools were working for me, but when I built and installed the latest version, everything failed. "st-info --chipid" didn't print anything, st-flash always failed with chip-id 0 error, GUI wouldn't connect. Did the "git bisect" thing and narrowed it down to the 8271b325... commit on Aug 15, "Make _stlink_usb_reset use hardreset". Reversed that patch on the head and everything's happy again: GUI connects, st-info --chipid shows 0x0417, st-flash erase erases.

The comment at the head of _stlink_usb_reset is "TODO - not convinced this does anything..." Doesn't exactly inspire confidence. At any rate, whatever setup that was intended to fix isn't my setup, so you'll have to make that change conditional on something. Sorry, I have no clue what that might be.

mlu, 2015-08-20:
I think this might be connected to the fact that the original _stlink_usb_reset() really does not reset and run the target like a push of the reset button, and that st-info and st-flash always calls reset when opening usb as discussed in #332 .
My suggestion, is we make the reset when opening usb an active choise option but we keep the hard reset so it is possible to actually reset the target.

mlu, 2015-08-20:
We can try the following changes and see if they help, with the patch "Make _stlink_usb_reset use hardreset" applied, the idea is to understand what is actually going on .
In stlink/flash/main.c:144
`    {
        sl = stlink_open_usb(o.log_level, 1);`

to 
`    {
        sl = stlink_open_usb(o.log_level, 0);`

In src/st-info.c:line 54
`    if (sl == NULL)
        sl = stlink_open_usb(0, 1);`
to
`    if (sl == NULL)
        sl = stlink_open_usb(0, 0);`

lcrocker, 2015-08-20:
This works on my 053. You also have to make the same change to line 533 of gui/stlink-gui.c. I've never used "st-term", so I don't know about that one.

mlu, 2015-08-20:
I will look into this and check all resets, and also see how we can add optional resets. But its good the code works :)

stevenhoneyman, 2015-08-20:
Just wanted to add the USB hard reset change doesn't work with my board/st-link either - F215 at the moment

mlu, 2015-08-21:
Pleas tell what program you are using, st-util or st-falsh. You can specify the command line. They follow slightly different paths inside stlink-common.

stevenhoneyman, 2015-08-21:
It's st-util that I'm using, with the Arch Linux provided arm-none-eabi-gdb

mlu, 2015-08-21:
I just added a Testing branch. Your problem might be helped by the patch at gdb-server.c:704

```
     // Some kinds of clock settings do not allow writing to flash.
     stlink_reset(sl);
+    stlink_force_debug(sl);
 
     for(struct flash_block* fb = flash_root; fb; fb = fb->next) {
         DLOG("flash_do: block %08x -> %04x\n", fb->addr, fb->length);```

stevenhoneyman, 2015-08-21:
Unfortunately not:

```sh
$ ./st-util 
2015-08-21T12:58:20 INFO src/stlink-usb.c: -- exit_dfu_mode
2015-08-21T12:58:20 INFO src/stlink-common.c: Loading device parameters....
2015-08-21T12:58:20 INFO src/stlink-common.c: Device connected is: F2 device, id 0x20036411
2015-08-21T12:58:20 INFO src/stlink-common.c: SRAM size: 0x20000 bytes (128 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 131072 bytes
2015-08-21T12:58:20 INFO gdbserver/gdb-server.c: Chip ID is 00000411, Core ID is  2ba01477.
2015-08-21T12:58:20 INFO gdbserver/gdb-server.c: Target voltage is 3232 mV.
2015-08-21T12:58:20 INFO gdbserver/gdb-server.c: Listening at *:4242...
2015-08-21T12:58:54 INFO gdbserver/gdb-server.c: GDB connected.
2015-08-21T12:58:56 INFO src/stlink-common.c: Attempting to write 16384 (0x4000) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x4000
Flash page at addr: 0x08000000 erased
2015-08-21T12:58:56 INFO src/stlink-common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2015-08-21T12:58:56 INFO src/stlink-common.c: Starting Flash write for F2/F4/L4
2015-08-21T12:58:56 INFO src/stlink-common.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 16384
2015-08-21T12:59:00 ERROR src/stlink-common.c: flash loader run error
2015-08-21T12:59:00 ERROR src/stlink-common.c: run_flash_loader(0x8000000) failed! == -1
```

stevenhoneyman, 2015-08-21:
A hard reset of the board, and starting gdb again and it loaded, if that makes any difference?

mlu, 2015-08-21:
Hard to tell, if it keeps loading then thats good, if hard board reset and restart of gdb is necessary all the time then it doesnt help.

stevenhoneyman, 2015-08-21:
It'll flash over and over until I try and do anything (i.e. run/debug) if that's what you meant.

```
Loading section .isr_vector, size 0x184 lma 0x8000000
Loading section .text, size 0x408 lma 0x8000184
Loading section .data, size 0x4 lma 0x800058c
.gdbinit:4: Error in sourced command file:
Error finishing flash operation
```
(board reset)
```
(gdb) load
Loading section .isr_vector, size 0x184 lma 0x8000000
Loading section .text, size 0x408 lma 0x8000184
Loading section .data, size 0x4 lma 0x800058c
Start address 0x8000184, load size 1424
Transfer rate: 1 KB/sec, 474 bytes/write.
(gdb) load
Loading section .isr_vector, size 0x184 lma 0x8000000
Loading section .text, size 0x408 lma 0x8000184
Loading section .data, size 0x4 lma 0x800058c
Start address 0x8000184, load size 1424
Transfer rate: 1 KB/sec, 474 bytes/write.
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
0x2000001e in _susrstack ()
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
0x2000001e in _susrstack ()
(gdb) load
Loading section .isr_vector, size 0x184 lma 0x8000000
Loading section .text, size 0x408 lma 0x8000184
Loading section .data, size 0x4 lma 0x800058c
Error finishing flash operation
```

xor-gate, 2016-05-14:
Hi all, could you please test the tooling from `master` because there have some changes been made to the loader of `STM32L0`.

xor-gate, 2016-06-02:
Closing due to lack of original issuer feedback. Please don't hesitate to open a new issue if there are still problems, thank your for your contributions.

Nightwalker-87, 2020-02-27:
Still lacking verification.

Nightwalker-87, 2020-03-24:
@lcrocker: What is the current state of this? What version of the stlink tools was used when this first appeared? Has this been fixed someday? If yes, it would be good to know where. Can you help?