stlink-org/stlink #342: STM32F746G-Discovery. Failed to unlock flash.

rabidvan, 2015-10-09:
andrey@localhost:~$ st-flash write ~/eclipse/blink_led.bin 0x08000000
2015-10-09T17:29:08 INFO src/stlink-common.c: Loading device parameters....
2015-10-09T17:29:08 INFO src/stlink-common.c: Device connected is: F7 device, id 0x10016449
2015-10-09T17:29:08 INFO src/stlink-common.c: SRAM size: 0x50000 bytes (320 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
2015-10-09T17:29:08 INFO src/stlink-common.c: Attempting to write 15336 (0x3be8) bytes to stm32 address: 134217728 (0x8000000)
2015-10-09T17:29:08 WARN src/stlink-common.c: Failed to unlock flash!
EraseFlash - Sector:0x0 Size:0x8000
Flash page at addr: 0x08000000 erased
2015-10-09T17:29:08 INFO src/stlink-common.c: Finished erasing 1 pages of 32768 (0x8000) bytes
2015-10-09T17:29:08 INFO src/stlink-common.c: Starting Flash write for F2/F4/L4
2015-10-09T17:29:08 INFO src/stlink-common.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 15336
2015-10-09T17:29:09 INFO src/stlink-common.c: Starting verification of write complete
2015-10-09T17:29:09 ERROR src/stlink-common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
andrey@localhost:~$

opyate, 2016-01-23:
@rabidvan which version of `texane/stlink` did you build?
I'm using `1.1.0` from https://www.archlinux.org/packages/community/x86_64/stlink/ and getting:

    $ sudo st-util
    [sudo] password for opyate: 
    2016-01-23T12:05:39 INFO src/stlink-common.c: Loading device parameters....
    2016-01-23T12:05:39 WARN src/stlink-common.c: unknown chip id! 0x10016449

I'm using the same discovery board as you.

xor-gate, 2016-05-14:
@rabidvan, @opyate could you test on `v1.2.0` then we can close this because the loaders have been reworked and some fixes have been done.

rabidvan, 2016-05-14:
Yes, now it works.

xor-gate, 2016-05-14:
Hi @rabidvan thanks for verification, your contribution is very much appreciated!

closed 2016-05-14 by xor-gate