stlink-org/stlink #343: OpenOcd broken on OS X ElCaptain

lmarty-sevenhugs, 2015-10-11:
Hi all, 

OpenOcd is broken after upgrade of my mybokkpro from Yosemite to ElCaptain. 
Any ideas how to fix this error :Error: read version failed
here is the log 

GNU ARM Eclipse 64-bit Open On-Chip Debugger 0.9.0-dev-00355-gcb3893d (2015-03-20-20:05)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.sourceforge.net/doc/doxygen/bugs.html
jtag_ntrst_delay
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Error: read version failed
in procedure 'init' 
in procedure 'ocd_bouncer' 
in procedure 'transport'
in procedure 'init'

Thanks

pavelrevak, 2015-10-11:
this problem is not only for openocd, but also this app texane/stlink.

seems it is problem somewhere in libusb.

first run is OK, when try to run stlink second time then it timeouts while reading first answer (get version) from ST-link device. need physically disconnect and connect back to get it working...

pavelrevak, 2015-10-18:
Duplicate #331

zlowred, 2015-10-26:
As per libusb developers this is not a libusb issue so it doesn't make sense to expect the fix from libusb side https://github.com/libusb/libusb/issues/111

yoyo-dev, 2015-11-05:
you're totally right that reseting USB of ST-link before downloading SW is a great workaround.

Cheong2K, 2015-11-09:
@yoyo-dev,
What did you mean?

mcuee, 2015-11-24:
I have re-opened libusb issue 111.

mcuee, 2015-11-24:
If it is just this "Error: read version failed", I think I mentioned in the libusb issue 111 that you can try the work-around to read the version twice. But I guess there may be more issues.

mcuee, 2015-11-26:
As mentioned in libusb/libusb#111, the work-around seems to be read the ST-Link version twice. Please try that. As per the report in the OpenOCD mailing list, that seems to sort out the problem.

lcoh817, 2015-12-16:
@mcuee, could you explain how and where to read the ST-Link version?

justLV, 2015-12-16:
Small ST-Link issue. After programming once the LED will be green. Unplugging and replugging in solves the problem temporarily and the LED goes back to red. Unintuitive, I know...

mcuee, 2015-12-16:
I am not an expert in ST-Link protocol. But you can refer to the source
codes in stlink or openocd.

Stlink: https://github.com/texane/stlink/blob/master/src/stlink-usb.c#L184
OpenOCD:
http://repo.or.cz/openocd.git/blob/HEAD:/src/jtag/drivers/stlink_usb.c#l484

mcuee, 2015-12-16:
BTW, based on all the information so far, we (libusb project) think
this is a Mac OS X issue.

mcuee, 2016-01-26:
As per report in libusb ticket, this is an Apple USB Stack bug and Apple Mac OS X A10.11.4 beta seems to sort out this issue.

Cheong2K, 2016-01-29:
Confirmed, fixed in OSX 10.11.4 beta.

ngwese, 2016-02-15:
For those with an STLink 2.1 device I would avoid doing a firmware upgrade if you have something working. I moved from firmware V2J26M13 to V2J26M15 (on a Nucleo-F429ZI). The M13 firmware seemed to work okay but the M15 firmware completely breaks stlink under 10.11.3.

I upgraded my machine to 10.11.4 beta 3 in hope of a fix but no dice. The 10.11.4 beta improves the stlink behavior (no more segfault) but it still fails with:
```
% st-info --descr
libusb_handle_events() timeout
[!] send_recv
libusb_handle_events() timeout
[!] send_recv
libusb_submit_transfer(-6)
[!] send_recv
libusb_submit_transfer(-6)
[!] send_recv
...
```

Just to be clear I don't think this is an stlink problem but a firmware problem which is connected to the problems in the Mac OSX usb stack.

jsjeong, 2016-03-22:
Today, Mac OS X 10.11.4 is released officially.
But it seems not fixed.

halt-hammerzeit, 2016-04-06:
My stm32vldiscovery isn't working - it just exits to the terminal command line:

```
➜  ~ /usr/local/Cellar/open-ocd/0.9.0/bin/openocd -f /usr/local/Cellar/open-ocd/0.9.0/share/openocd/scripts/board/stm32vldiscovery.cfg
Open On-Chip Debugger 0.9.0 (2015-11-15-05:39)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
in procedure 'init' 
in procedure 'ocd_bouncer'

➜  ~ 

```

halt-hammerzeit, 2016-04-09:
@jsjeong My situation was fixed after the installation of the OS X stlink driver
https://github.com/texane/stlink/tree/master/stlinkv1_macosx_driver
See my comment for the proof.
I'm on El Capitan 10.11.4

ngwese, 2016-04-16:
After upgrading to 10.11.4 (final) and upgrading the STLink firmware on my Nucleo-F429ZI to V2J27M15     things are once again working (without the stlinkv1 kernel driver referenced above).

xor-gate, 2016-05-11:
Is there any progress/testing going on for this issue? Please test with latest tools from the master branch.

xor-gate, 2016-05-17:
As mentioned by @halt-hammerzeit and @ngwese it should work on latest 10.11 OSX. Thanks all for you feedback and discussion. Feel free when there are other problems to open a new issue.

closed 2016-05-17 by xor-gate