stlink-org/stlink #344: Macos : error darwin_claim_interface

kaidokert, 2015-10-13:
Just putting this in here for others to find, as i was pulling my hair out.
Im on Mac/Yosemite, initially installed everything through brew.

( http://davidyamnitsky.com/blog/2013/11/14/stm32-mac/ and http://nondisplayable.ca/2015/07/10/stm32f0-discovery-setup-mac-os-x.html )

Have a STM32VLDISCOVERY ( rev MB913 ) board at hand. No matter what i try, reinstall uninstall unload kexts etc, nothing works. I keep getting this error:
```st-util
    libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
    2015-10-12T19:52:41 WARN src/stlink-sg.c: libusb_claim_interface() failed
    2015-10-12T19:52:41 ERROR src/stlink-sg.c: Could not open stlink device
```
Also this garbled message in 'sudo dmesg'
```USBMSC Identifier (non-unique): H\M-C\M-?j```

Reboots, unplugging everything and inspecting kext load configs etc doesnt help. Rebuilding everything manually from git master doesnt help too

Eventually plugged in another discovery board, STM32F0DISCOVERY ( MB1034 ) and suddenly everything works. Even stranger, now the old board now works too.

xor-gate, 2016-05-14:
Could you test also with latest `v1.2.0` ?

xor-gate, 2016-05-20:
Closing due to inactivity of original issuer. Feel free to open a new issue when the problem still remains with `v1.2.0` or `master`. Thank you for your contribution.

closed 2016-05-20 by xor-gate