stlink-org/stlink #346: Read protect bits

AntonLytkinIwdel, 2015-11-08:
Do you plan to make this functional?

texane, 2015-11-08:
no plan yet, but any contribution welcome !

aric1987, 2017-07-23:
https://github.com/aric1987/stlink/
test ok! for  stm32f103c8t6

ceremcem, 2019-06-26:
@aric1987 Why don't you create a PR for this?

Nightwalker-87, 2021-05-02:
@ceremcem:  @aric1987 seem to be inactive by now.
I don't know yet, if his patch is still compatible with the current codebase. One needs to check.

Nightwalker-87, 2021-05-02:
@Ant-ON: As I believe to read from the code that these protect bits are also included in the option bytes, as the whole area is written at once. Is this correct?

Ant-ON, 2021-05-04:
@Nightwalker-87 Yes, on F0/F1/F3:
```c
  /* Retrieve current values */
  ret = stlink_read_debug32(sl, FLASH_OBR, &optiondata);
  if (ret) {
    return ret;
  }
  ret = stlink_read_debug32(sl, FLASH_WRPR, &protection);
  if (ret) {
    return ret;
  }
```

PR #1112 (2021-05-10): Support for writing option bytes on STM32F0/F1/F3
Implementation to write option bytes for STM32F0/F1/F3.

(Closes #346, Closes #458, Closes #808, Closes #1084)
closed 2021-05-10 by Nightwalker-87