stlink-org/stlink #352: STM32L476 Discovery ...  unknown chip id! 0

navesta, 2015-12-01:
On 64-bit Ubuntu 14.04, did the following:
1- Installed GUN ARM tool chain, following https://launchpad.net/~terry.guo/+archive/ubuntu/gcc-arm-embedded
2- Installed 32 bit libraries,  sudo apt-get install lib32z1 lib32ncurses5 lib32bz2-1.0
3- Installed usblib, sudo apt-get install libusb-1.0
4- Got the latest st-link and built it, i.e., (
git clone git://github.com/texane/stlink.git
$ sudo apt-get install libusb-1.0
$ sudo apt-get install autoreconf
$ cd stlink 
$ ./autogen.sh
$ ./configure
$ make
$ sudo cp 49-stlinkv2-1.rules /etc/udev/rules.d
$ sudo udevadm control --reload-rules
$ sudo udevadm trigger
)
And am getting this:
stlink$ sudo ./st-util -v99
2015-12-01T12:49:20 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)
2015-12-01T12:49:20 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)
2015-12-01T12:49:20 DEBUG src/stlink-common.c: *** looking up stlink version
2015-12-01T12:49:20 DEBUG src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)
2015-12-01T12:49:20 DEBUG src/stlink-common.c: stlink pid     = 0x374b
2015-12-01T12:49:20 DEBUG src/stlink-common.c: stlink version = 0x2
2015-12-01T12:49:20 DEBUG src/stlink-common.c: jtag version   = 0x18
2015-12-01T12:49:20 DEBUG src/stlink-common.c: swim version   = 0xb
2015-12-01T12:49:20 INFO src/stlink-common.c: Loading device parameters....
2015-12-01T12:49:20 DEBUG src/stlink-common.c: *** stlink_core_id ***
2015-12-01T12:49:20 DEBUG src/stlink-common.c: core_id = 0x00000000
2015-12-01T12:49:20 DEBUG src/stlink-common.c: *** stlink_read_debug32 0 is 0xe0042000
2015-12-01T12:49:20 DEBUG src/stlink-common.c: *** stlink_read_debug32 0 is 0x40015800
2015-12-01T12:49:20 WARN src/stlink-common.c: unknown chip id! 0
2015-12-01T12:49:20 DEBUG src/stlink-common.c: *** stlink_close ***

Cheers,

jeremytregunna, 2015-12-03:
I'm also seeing the same problem with the latest code.

jeremytregunna, 2015-12-04:
My issue was worked around by using ST's windows utility to erase flash and load a known good image onto it. After this, openocd started seeing the device properly again.

xor-gate, 2016-05-03:
Please test with latest stlink (master) and newest stlink firmware. Reopen when issue still remains.

Nightwalker-87, 2020-02-27:
Reopening this for verification.

Nightwalker-87, 2020-03-24:
@navesta: Which version of the `stlink tools` did you use?

Nightwalker-87, 2020-07-05:
Is there any feedback on this from older versions of the toolset?

Nightwalker-87, 2021-03-13:
Fixed in #367.

closed 2021-03-13 by Nightwalker-87