stlink-org/stlink #356: st-flash run_flash_loader(0x08000000) failled

Almaz1c, 2015-12-09:
Hello. I try to load firmare in STM32F103CBT6 using compiled st-flash.exe utility under windows 8.1.

There is used command:
st-flash.exe write ${project_name}.bin 0x08000000

Here is result: 
...
Flash page at addr: 0x08000000 erased
Flash page at addr: 0x08000400 erased
Flash page at addr: 0x08000800 erased
Flash page at addr: 0x08000c00 erased
Flash page at addr: 0x08001000 erased
Flash page at addr: 0x08001400 erased
Flash page at addr: 0x08001800 erased
Flash page at addr: 0x08001c00 erased
Flash page at addr: 0x08002000 erased
Flash page at addr: 0x08002400 erased
Flash page at addr: 0x08002800 erased
Flash page at addr: 0x08002c00 erased
Flash page at addr: 0x08003000 erased
Flash page at addr: 0x08003400 erased
Flash page at addr: 0x08003800 erased
Flash page at addr: 0x08003c00 erased
Flash page at addr: 0x08004000 erased
Flash page at addr: 0x08004400 erased
Flash page at addr: 0x08004800 erased
Flash page at addr: 0x08004c00 erased
Flash page at addr: 0x08005000 erased
Flash page at addr: 0x08005400 erased
Flash page at addr: 0x08005800 erased
Flash page at addr: 0x08005c00 erased
Flash page at addr: 0x08006000 erased
Flash page at addr: 0x08006400 erased
Flash page at addr: 0x08006800 erased
Flash page at addr: 0x08006c00 erased
Flash page at addr: 0x08007000 erased
Flash page at addr: 0x08007400 erased
Flash page at addr: 0x08007800 erased
Flash page at addr: 0x08007c00 erased
Flash page at addr: 0x08008000 erased
Flash page at addr: 0x08008400 erased
Flash page at addr: 0x08008800 erased
Flash page at addr: 0x08008c00 erased
Flash page at addr: 0x08009000 erased
Flash page at addr: 0x08009400 erased
Flash page at addr: 0x08009800 erased
Flash page at addr: 0x08009c00 erased
Flash page at addr: 0x0800a000 erased
Flash page at addr: 0x0800a400 erased
stlink_fwrite_flash() == -1
2015-12-09T22:01:57 INFO src/stlink-common.c: Loading device parameters....
2015-12-09T22:01:57 INFO src/stlink-common.c: Device connected is: F1 Medium-density device, id 0x20036410
2015-12-09T22:01:57 INFO src/stlink-common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0xb4a000 bytes (11560 KiB) in pages of 1024 bytes
2015-12-09T22:01:57 INFO src/stlink-common.c: Attempting to write 42368 (0xa580) bytes to stm32 address: 134217728 (0x8000000)
2015-12-09T22:01:57 INFO src/stlink-common.c: Finished erasing 42 pages of 1024 (0x400) bytes
2015-12-09T22:01:57 INFO src/stlink-common.c: Starting Flash write for VL/F0/F3 core id
2015-12-09T22:01:57 INFO src/stlink-common.c: Successfully loaded flash loader in sram
2015-12-09T22:02:18 ERROR src/stlink-common.c: flash loader run error
2015-12-09T22:02:18 ERROR src/stlink-common.c: run_flash_loader(0x8000000) failed! == -1

What is the issue? Thank you)

xor-gate, 2016-05-17:
@Almaz1c could you test with `v1.2.0` or `master`, and see if the problem still exists?

vladiks, 2016-05-26:
I can confirm this bug, but I get it only occasionally - sometimes it passes, sometimes it fails.

- using the same command and params
- Running on raspberry Pi 2, using Linux raspberrypi 4.1.7+, armv6l
- chip STM32F091RCT6

I am using latest version, compiled yesterday. My output:
```
pi@raspberrypi ~/Desktop/Tester $ sudo st-flash write fw_v13.bin 0x08000000
2016-05-26T10:23:34 INFO src/common.c: Loading device parameters....
2016-05-26T10:23:34 INFO src/common.c: Device connected is: F09X device, id 0x10006442
2016-05-26T10:23:34 INFO src/common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
2016-05-26T10:23:34 INFO src/common.c: Attempting to write 196964 (0x30164) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08030000 erased
2016-05-26T10:23:37 INFO src/common.c: Finished erasing 97 pages of 2048 (0x800) bytes
2016-05-26T10:23:37 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2016-05-26T10:23:37 INFO src/flash_loader.c: Successfully loaded flash loader in sram
2016-05-26T10:23:47 ERROR src/flash_loader.c: flash loader run error
2016-05-26T10:23:47 ERROR src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

xor-gate, 2016-05-26:
@vladiks which version/git tag do you use of the st-flash tool?

xor-gate, 2016-05-26:
Current version is unable to compile with visual studio, probably it should work under MinGW but I have not tested this.

vladiks, 2016-05-26:
@xor-gate I'm sorry, but I don't know how to find out the git tag. I've downloaded master as zip and copied it to raspberry. I am retty sure it's 1.2.0, there are also news for 1.2.0 in changelog.

I've avoided the issue by erasing memory by separate command before flashing it again - the problem now dissappeared.

xor-gate, 2016-05-26:
@vladiks when you downloaded zipfile for master you running latest version (beyond 1.2.0). I have seen an exact issue where people erase the flash (mass erase/full?) and then it worked. This is a problem which I saw some other people have but have not confirmed and seen this myself. In the upcoming time I will prepare a continues hardware/stlink testing setup at [Dual Inventive](https://github.com/dualinventive) and hopefully I will catch this nasty bug. 

This needs some documentation somewhere if people tried to (mass)erased and then flashed and the problem resolved.

vladiks, 2016-05-26:
Good luck and many thanks for your work on this utility.

xor-gate, 2016-05-26:
I have put it in the README.md. So people will see it :+1:. Thanks all for the feedback, feel free to open a issue when something is not working as expected!

zokl, 2016-09-09:
Hello I have the same problem with STML0xx. This error occurs only with st-link utility in linux and mac os X. Utility from STM works fine.

Nightwalker-87, 2020-02-27:
Reopening this for verification.

Nightwalker-87, 2020-03-18:
STM32F0 und STM32F1 boards are **NOT** addressed by the linked PR.
Please refer to #468, #612, #642 instead if similar problems persist with these devices.

yeswolf, 2020-03-29:
I see the same issue always on macos and I can test it.

yeswolf, 2020-03-29:
Yep, finally I can test it. For me it's

```
2020-03-29T13:55:22 INFO flash_loader.c: Successfully loaded flash loader in sram
2020-03-29T13:55:22 ERROR flash_loader.c: write error, count == 1023
2020-03-29T13:55:22 ERROR common.c: stlink_flash_loader_run(0x8000800) failed! == -1
```
always on macOS, so it's pretty stable. Tired of mass erasing before, want to help %)

Nightwalker-87, 2020-04-04:
@yeswolf: Which version of `stlink` do you use?

yeswolf, 2020-04-04:
@Nightwalker-87 I use ```v1.6.0-dirty``` built from sources (downloaded from releases) with STM32F103RCT6 High-density.

Nightwalker-87, 2020-04-04:
I wonder if this could be related to a missing flash loader for F1 devices ...
We currently only have one for F0, F4, F7, L0, L1 and L4.

Nightwalker-87, 2020-06-24:
@yeswolf: Any changes with v1.6.1?

yeswolf, 2020-08-05:
@Nightwalker-87 Will test after 23th of August.

stappersg, 2020-08-09:
> Will test after 23th of August.

OK,  enjoy your vacation.
Please do the testing, it will help to reduce the amount of open issues of this cool project.

Besides @yeswolf  also high-lighting @Almaz1c and @vladiks .
See #1000 for other requests how to support this project.

yeswolf, 2020-09-05:
So, I've tried 1.6.1 using 2 approaches: a) installed from HomeBrew b) built from source. The result is strange: a) there is no issue with erasing the firmware as it was before b) but I can't launch debugging my firmware, I always have SIGTRAP in `startup_stm32f103xe.s`:

```
LoopFillZerobss:
  ldr r3, = _ebss
  cmp r2, r3
  bcc FillZerobss

/* Call the clock system intitialization function.*/
    bl  SystemInit
/* Call static constructors */
    bl __libc_init_array
/* Call the application's entry point.*/
  bl main
  bx lr // SIGTRAP here
.size Reset_Handler, .-Reset_Handler
```

If I use st-link 1.6.0 dirty (compiled from HEAD being `84f63d25bd77d61f8feccf08dd24d0d11895ef32`), there is no automatic mass-erasing (as I wrote in my previous comments), but debugging starts successfully without any problem. Same code, same project, same toolchain, same device - everything is the same except the path to st-util. 

I'll be happy to provide any logs to troubleshoot this issue.

ondrejspilka, 2021-01-23:
Hi,

I got the same issue on L496Disco board on Mac OS Catalina. 
I used mass erase prior flashing.
Interesting is AC6 System workbench works, but I want to migrate to VSCode.

st-link v.1.6.1. from Homebrew.

Debugging from VSCode using openOCD starts the project, but breaks into startup asm `startup_stm32l496xx.s`.
It doesn't run the code, looks like the code is not flashed.

```
➜  anemoss.vscode git:(master) ✗ st-flash erase; st-flash write build/anemoss.2.bin 0x8000000 
st-flash 1.6.1
2021-01-23T20:42:02 INFO common.c: L496x/L4A6x: 256 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
Mass erasing
st-flash 1.6.1
2021-01-23T20:42:02 INFO common.c: L496x/L4A6x: 256 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
file build/anemoss.2.bin md5 checksum: 3bf4f2cd944342fcf5012ecda24c954, stlink checksum: 0x00648757
2021-01-23T20:42:02 INFO common.c: Attempting to write 79656 (0x13728) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Page:0x0 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08000000 erased
EraseFlash - Page:0x1 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08000800 erased
EraseFlash - Page:0x2 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08001000 erased
EraseFlash - Page:0x3 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08001800 erased
EraseFlash - Page:0x4 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08002000 erased
EraseFlash - Page:0x5 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08002800 erased
EraseFlash - Page:0x6 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08003000 erased
EraseFlash - Page:0x7 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08003800 erased
EraseFlash - Page:0x8 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08004000 erased
EraseFlash - Page:0x9 Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08004800 erased
EraseFlash - Page:0xa Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08005000 erased
EraseFlash - Page:0xb Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08005800 erased
EraseFlash - Page:0xc Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08006000 erased
EraseFlash - Page:0xd Size:0x800 2021-01-23T20:42:02 INFO common.c: Flash page at addr: 0x08006800 erased
EraseFlash - Page:0xe Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08007000 erased
EraseFlash - Page:0xf Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08007800 erased
EraseFlash - Page:0x10 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08008000 erased
EraseFlash - Page:0x11 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08008800 erased
EraseFlash - Page:0x12 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08009000 erased
EraseFlash - Page:0x13 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08009800 erased
EraseFlash - Page:0x14 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800a000 erased
EraseFlash - Page:0x15 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800a800 erased
EraseFlash - Page:0x16 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800b000 erased
EraseFlash - Page:0x17 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800b800 erased
EraseFlash - Page:0x18 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800c000 erased
EraseFlash - Page:0x19 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800c800 erased
EraseFlash - Page:0x1a Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800d000 erased
EraseFlash - Page:0x1b Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800d800 erased
EraseFlash - Page:0x1c Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800e000 erased
EraseFlash - Page:0x1d Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800e800 erased
EraseFlash - Page:0x1e Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800f000 erased
EraseFlash - Page:0x1f Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x0800f800 erased
EraseFlash - Page:0x20 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08010000 erased
EraseFlash - Page:0x21 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08010800 erased
EraseFlash - Page:0x22 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08011000 erased
EraseFlash - Page:0x23 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08011800 erased
EraseFlash - Page:0x24 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08012000 erased
EraseFlash - Page:0x25 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08012800 erased
EraseFlash - Page:0x26 Size:0x800 2021-01-23T20:42:03 INFO common.c: Flash page at addr: 0x08013000 erased
2021-01-23T20:42:03 INFO common.c: Finished erasing 39 pages of 2048 (0x800) bytes
2021-01-23T20:42:03 INFO common.c: Starting Flash write for F2/F4/L4
2021-01-23T20:42:03 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
2021-01-23T20:42:04 ERROR flash_loader.c: flash loader run error
2021-01-23T20:42:04 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

Nightwalker-87, 2021-03-09:
@yeswolf: Your latest finding appears to be #1038. Please continue to track that issue there.
@ondrejspilka: Please continue to follow this issue in #607. This is not the correct topic for addressing this issue.

Nightwalker-87, 2021-03-09:
More detailed investigation:

On  `develop` commit 7829600c9af2cd354f61fc9ed2165b58733d54de (2020-05-28) (the last before Release v1.6.1) is still good.

```
st-flash 1.6.0-360-g7829600
2021-03-09T21:32:32 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file /[...]/Blinker.bin md5 checksum: 16277ff2c027d497361dbae4255081a1, stlink checksum: 0x0004644a
2021-03-09T21:32:32 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
2021-03-09T21:32:32 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-09T21:32:32 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-09T21:32:32 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-09T21:32:32 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-09T21:32:32 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-09T21:32:32 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2021-03-09T21:32:32 INFO flash_loader.c: Successfully loaded flash loader in sram
  4/4 pages written
2021-03-09T21:32:33 INFO common.c: Starting verification of write complete
2021-03-09T21:32:33 INFO common.c: Flash written and verified! jolly good!
```

Further known good commits:
- 1e2092173aae4ce0c256ff345cf647ec868f5435 (2020-10-21)
- 8560f931d4b9476cb00f59aa1f7ab784baa16699 (2020-11-12)

Known bad commits:
- Release v1.6.1 (`master`)
- b47f083736a6f6137e317e70998501205d44265f (2020-12-05) (`develop`) --> PR by @Ant-ON (Nov 2020) reintroduces the bug for a second time, while master remains affected since v1.6.1

We now need to find the exact commit and compare develop and master to each other at the v1.6.1 commit.

One key problem I see that facilitated such errors in the past is that `develop` was merged into master but continued to evolve further from its own branch instead from the master commit at v1.6.1 itself. We need to keep an eye on this in the future.

I would appreciate if someone could help to at least allocate this bug more precisely and to fix it (@Ant-ON maybe?).

Ant-ON, 2021-03-10:
@Nightwalker-87 Are there any failure logs with commits after my PR?
I added a processor status output to flash loader run failed in order to somehow find their reasons.

I tried to fix one of the possible reasons. If the MCU memory fill the garbage (and for some other reason), then it may be in the wrong command mode. In this case, switching occurs and the message "Go to Thumb mode" is displayed.

The last time I tried to figure it out in issue #1092. Perhaps the cause of the problem was in the MPU. But I never got an answer...

Nightwalker-87, 2021-03-10:
@Ant-ON: I think I can provide some additional output soon, but as mentioned in #1045 we may do a release first to resolve a divergence between `master` and `develop` instead and focus on a new common base for both branches and try to fix that up.

yeswolf, 2021-03-11:
@Nightwalker-87 Hate writing such comments, but I won't be able to verify - I already reflashed all my stlinks to JLink and switched to it's stack completely. Sorry for that.

Nightwalker-87, 2021-03-12:
@Ant-ON We have that merge now and can continue on a fresh `develop` branch.

Nightwalker-87, 2021-03-12:
I can **confirm** this bug. Flashing fails on every second attempt when the chip is not erased in-between:

```
$ sudo st-flash write /[...]/Blinker.bin 0x8000000
st-flash 1.6.1-217-g78e6eac
2021-03-12T15:39:44 WARN usb.c: NRST is not connected
2021-03-12T15:39:44 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file /[...]/Blinker.bin md5 checksum: 16277ff2c027d497361dbae4255081a1, stlink checksum: 0x0004644a
2021-03-12T15:39:44 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
2021-03-12T15:39:44 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-12T15:39:44 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-12T15:39:44 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-12T15:39:44 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-12T15:39:44 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-12T15:39:44 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-12T15:39:44 INFO flash_loader.c: Successfully loaded flash loader in sram
  4/  4 pages written
2021-03-12T15:39:44 INFO common.c: Starting verification of write complete
2021-03-12T15:39:44 INFO common.c: Flash written and verified! jolly good!

$ sudo st-flash write /[...]/Blinker.bin 0x8000000
st-flash 1.6.1-217-g78e6eac
2021-03-12T15:39:48 WARN usb.c: NRST is not connected
2021-03-12T15:39:48 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file /[...]/Blinker.bin md5 checksum: 16277ff2c027d497361dbae4255081a1, stlink checksum: 0x0004644a
2021-03-12T15:39:48 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
2021-03-12T15:39:48 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-12T15:39:48 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-12T15:39:48 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-12T15:39:48 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-12T15:39:48 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-12T15:39:48 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-12T15:39:48 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-12T15:39:48 ERROR flash_loader.c: Write error (R2 0x000001DE R15 0x08000304 DHCSR 0x00030003 DFSR 0x00000009)
2021-03-12T15:39:48 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1

$ sudo st-flash write /[...]/Blinker.bin 0x8000000
st-flash 1.6.1-217-g78e6eac
2021-03-12T15:39:50 WARN usb.c: NRST is not connected
2021-03-12T15:39:50 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file /[...]/Blinker.bin md5 checksum: 16277ff2c027d497361dbae4255081a1, stlink checksum: 0x0004644a
2021-03-12T15:39:50 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
2021-03-12T15:39:50 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-12T15:39:50 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-12T15:39:50 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-12T15:39:50 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-12T15:39:50 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-12T15:39:50 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-12T15:39:50 INFO flash_loader.c: Successfully loaded flash loader in sram
  4/  4 pages written
2021-03-12T15:39:50 INFO common.c: Starting verification of write complete
2021-03-12T15:39:50 INFO common.c: Flash written and verified! jolly good!

$ sudo st-flash write /home/nightwalker-87/Schreibtisch/Blinker/Release/Blinker.bin 0x8000000
st-flash 1.6.1-217-g78e6eac
2021-03-12T15:39:52 WARN usb.c: NRST is not connected
2021-03-12T15:39:52 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file /[...]/Blinker.bin md5 checksum: 16277ff2c027d497361dbae4255081a1, stlink checksum: 0x0004644a
2021-03-12T15:39:52 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
2021-03-12T15:39:52 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-12T15:39:52 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-12T15:39:52 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-12T15:39:52 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-12T15:39:52 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-12T15:39:52 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-12T15:39:52 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-12T15:39:52 ERROR flash_loader.c: Write error (R2 0x000001FC R15 0xFFFFFFFE DHCSR 0x00030003 DFSR 0x00000009)
2021-03-12T15:39:52 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

Setup:
- Programmer STLink/V2-clone / Firmware V2.J37.S7
- Board: STM32F103C8T6 ("Bluepill")
- OS: Debian (Bullseye [Testing])

There are several other issues dealing with this - it is one of the main bug in this toolset as far as I can see.
I'll list up these issues here soon.

Nightwalker-87, 2021-03-12:
Related issues are: #556 #597 #607 #612 #638 #690 #817 #967 #979 #1054

Ant-ON, 2021-03-19:
@Nightwalker-87 Can you test https://github.com/Ant-ON/stlink/tree/flash_loader_rework?

Nightwalker-87, 2021-03-19:
Yes, I will soon. Thx for the note.

Nightwalker-87, 2021-03-20:
@Ant-ON Unfortunately it doesn't make a difference. :confused:

Ant-ON, 2021-03-21:
@Nightwalker-87 Do you use --reset option? Can you upload firmware for bluepill? I will test it

Nightwalker-87, 2021-03-21:
No, but I retried and again there was no difference. The first attempt fails, the second one works, the next one fails and so on...
So in consequence one can upload firmware under these circumstances.

Ant-ON, 2021-03-21:
The problem is in setting up the MCU firmware, if the firmware goes once every other time. It is necessary to determine what exactly needs to be turned off.
The most easy option is to reboot the MK after erasing, but this is not the best option.

Nightwalker-87, 2021-03-21:
Hm, ok - Well there is no guarantee that the process is currently implemented well...
Should you find any flaws, just get rid of them if a reasonable alternative comes to you mind.

I'm still busy reviewing tickets and identifying already solved issues.
Nevertheless I can test things with my Bluepill or a Discovery VL board anytime. Just let me know. :wink:

Ant-ON, 2021-03-22:
@Nightwalker-87 I have looked at the logs from https://github.com/stlink-org/stlink/issues/356#issuecomment-803549064. You use `develop` branch of main repo. Can you try with `st-flash` from my fork https://github.com/Ant-ON/stlink/tree/flash_loader_rework?

Nightwalker-87, 2021-03-22:
No, I haven't: ` git checkout flash_loader_rework` (commit: bc9062c6c590e603e203567b6bfe14baf4c6a0af)
My command was `st-flash --reset write /[...]/Blinker.bin 0x8000000`, as requested.

Still can't report anything else. :confused:

Ant-ON, 2021-03-22:
@Nightwalker-87 I fixed the problem on the `flash_loader_rework` baranch of my fork. I want to make sure that the problem was solved not only for me. 
But the `--connect-under-reset` results were interesting. The error says that the MCU is a sleep...

Nightwalker-87, 2021-03-22:
Yeah, well, nobody claimed this would be easy to solve. :unamused:

Ant-ON, 2021-03-22:
@Nightwalker-87 
The `flash_loader_rework` branch
```console
anton@vivobook:~/stlink/build/Release/bin$ ./st-flash --reset --format ihex write ./STM32F103_Nucleo_Demo.hex 
st-flash 1.6.1-229-gbc9062c-dirty
2021-03-22T19:22:48 WARN usb.c: NRST is not connected
2021-03-22T19:22:48 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2021-03-22T19:22:48 WARN usb.c: NRST is not connected
2021-03-22T19:22:48 INFO common.c: Attempting to write 19328 (0x4b80) bytes to stm32 address: 134217728 (0x8000000)
2021-03-22T19:22:48 INFO common.c: Flash page at addr: 0x08000000 erased
...
2021-03-22T19:22:49 INFO common.c: Flash page at addr: 0x08004800 erased
2021-03-22T19:22:49 INFO common.c: Finished erasing 19 pages of 1024 (0x400) bytes
2021-03-22T19:22:49 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-22T19:22:49 INFO flash_loader.c: Successfully loaded flash loader in sram
 19/ 19 pages written
2021-03-22T19:22:50 INFO common.c: Starting verification of write complete
2021-03-22T19:22:50 INFO common.c: Flash written and verified! jolly good!
2021-03-22T19:22:50 WARN usb.c: NRST is not connected
anton@vivobook:~/stlink/build/Release/bin$ ./st-flash --reset --format ihex write ./STM32F103_Nucleo_Demo.hex 
st-flash 1.6.1-229-gbc9062c-dirty
2021-03-22T19:22:51 WARN usb.c: NRST is not connected
2021-03-22T19:22:51 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2021-03-22T19:22:51 WARN usb.c: NRST is not connected
2021-03-22T19:22:51 INFO common.c: Attempting to write 19328 (0x4b80) bytes to stm32 address: 134217728 (0x8000000)
2021-03-22T19:22:51 INFO common.c: Flash page at addr: 0x08000000 erased
...
2021-03-22T19:22:52 INFO common.c: Flash page at addr: 0x08004800 erased
2021-03-22T19:22:52 INFO common.c: Finished erasing 19 pages of 1024 (0x400) bytes
2021-03-22T19:22:52 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-22T19:22:52 INFO flash_loader.c: Successfully loaded flash loader in sram
 19/ 19 pages written
2021-03-22T19:22:53 INFO common.c: Starting verification of write complete
2021-03-22T19:22:53 INFO common.c: Flash written and verified! jolly good!
2021-03-22T19:22:53 WARN usb.c: NRST is not connected
```
The `develop` branch
```console
anton@vivobook:~/stlink/build/Release/bin$ ./st-flash --reset --format ihex write ./STM32F103_Nucleo_Demo.hex 
st-flash 1.6.1-229-gbc9062c-dirty
2021-03-22T19:27:36 WARN usb.c: NRST is not connected
2021-03-22T19:27:36 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2021-03-22T19:27:36 WARN usb.c: NRST is not connected
2021-03-22T19:27:36 INFO common.c: Attempting to write 19328 (0x4b80) bytes to stm32 address: 134217728 (0x8000000)
2021-03-22T19:27:36 INFO common.c: Flash page at addr: 0x08000000 erased
...
2021-03-22T19:27:36 INFO common.c: Flash page at addr: 0x08004800 erased
2021-03-22T19:27:36 INFO common.c: Finished erasing 19 pages of 1024 (0x400) bytes
2021-03-22T19:27:36 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-22T19:27:36 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-22T19:27:36 ERROR flash_loader.c: Write error (R2 0x000001F8 R15 0x080049D2 DHCSR 0x00030003 DFSR 0x00000009)
2021-03-22T19:27:36 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
anton@vivobook:~/stlink/build/Release/bin$ ./st-flash --reset --format ihex write ./STM32F103_Nucleo_Demo.hex 
st-flash 1.6.1-229-gbc9062c-dirty
2021-03-22T19:27:38 WARN usb.c: NRST is not connected
2021-03-22T19:27:38 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2021-03-22T19:27:38 WARN usb.c: NRST is not connected
2021-03-22T19:27:38 INFO common.c: Attempting to write 19328 (0x4b80) bytes to stm32 address: 134217728 (0x8000000)
2021-03-22T19:27:38 INFO common.c: Flash page at addr: 0x08000000 erased
...
2021-03-22T19:27:38 INFO common.c: Flash page at addr: 0x08004800 erased
2021-03-22T19:27:38 INFO common.c: Finished erasing 19 pages of 1024 (0x400) bytes
2021-03-22T19:27:38 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-22T19:27:38 INFO flash_loader.c: Successfully loaded flash loader in sram
 19/ 19 pages written
2021-03-22T19:27:39 INFO common.c: Starting verification of write complete
2021-03-22T19:27:40 INFO common.c: Flash written and verified! jolly good!
2021-03-22T19:27:40 WARN usb.c: NRST is not connected
```

Nightwalker-87, 2021-03-22:
@Ant-ON I can now reproduce after a clean uninstall.
I forgot that I had done a full system install of the develop branch head commit previously.
So of course there were leftovers. :facepalm: Sry for that - I should have known better.
The fix obviously solves the issue.

Nightwalker-87, 2021-03-22:
... so now it's about time to collect all related issues together, I guess. :1st_place_medal: :champagne:

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87