stlink-org/stlink #358: resetting a stm32f103 with a "kill + run" doesn't seem to work

l29ah, 2015-12-15:
(gdb) run
Starting program: blah

And that's it, it just hangs indefinitely. Nothing happens in the st-util log meanwhile. I've spotted an irregularity tho:

send: 
recv: vKill;a410
send: OK
recv: ?
send: OK
recv: vRun;
send: 
recv: R0
KARL - should read back as 0x03, not 60 02 00 00
init watchpoints

xor-gate, 2016-05-04:
Do you have some more details about your setup, programmer, target, debug output from `st-util`?

l29ah, 2016-05-04:
stm32f1discovery with the embedded stlink thing.
I've moved to openocd that works awesome since then.

xor-gate, 2016-05-04:
@l29ah thanks for your feedback, OpenOCD has more maintainers and users and has full ARM Semihosting support. Have you tried with latest `st-util` from master?

xor-gate, 2016-05-04:
This is somewhat related to issue #131

l29ah, 2016-05-04:
I was using master back then.

xor-gate, 2016-05-04:
Did you enable arm semihosting or using it from you target? As stlink tools don't support this.

l29ah, 2016-05-04:
On Wed, May 04, 2016 at 11:52:54AM -0700, Jerry Jacobs wrote:
> Did you enable arm semihosting or using it from you target? As stlink tools don't support this.

Didn't know about this thing, probably not.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

xor-gate, 2016-05-04:
As you are using OpenOCD now, should I close this issue?

xor-gate, 2016-05-17:
Closing due to inactivity and original issuer switched to use OpenOCD instead.

Nightwalker-87, 2020-03-14:
Reopening as problem remains unresolved.

Nightwalker-87, 2020-07-18:
Can somebody please verify this with a STM32F1 Discovery board on the `v1.3.0-patch` branch?

Nightwalker-87, 2021-04-04:
Fixed in #528.

closed 2021-04-04 by Nightwalker-87