stlink-org/stlink #378: st-flash: write with v2 device fails if size is specified

smithbone, 2016-02-29:
The v1/v2 checking code for st-flash is dependent on the number of arguments provided.  <size> is listed as an option argument but if you you provide that argument then the argument parser assumes you are talking to a v1 device because 4 arguments are provided.

gerryqd, 2016-05-14:
Just checked the code, seems the <size> info is useless when it is in "write" mode.

Anyway, this is a bug.

xor-gate, 2016-05-14:
Did you fixed this issue with pull request #413 ? I did review it and seems to be fine. I'm currently not very happy how the option parsing is implemented and it needs some love and reworking. Thanks for looking into it.

gerryqd, 2016-05-14:
Sorry, I did not fix it in #413. Personally, I think the opt parse function should be rewritten. Some cases should be taken into account.

xor-gate, 2016-05-14:
Yeah you are right, I will rename this issue because it is major. I take two small option parsing libraries into account which are both MIT license:

* https://github.com/clibs/flag
* or https://github.com/clibs/commander

xor-gate, 2016-05-20:
Related to #208

Nightwalker-87, 2020-03-21:
@gerryqd: What is the current state of this?

Nightwalker-87, 2020-04-11:
@grevaillot: Can you confirm if this resolves with your recent PR?

grevaillot, 2020-04-12:
i'm a bit puzzled by that piece of code, the v1/v2 stlink handling is a bit broken, the device path is not actually used, and i think it's useless now.

still it would be nice to replace the whole custom opt parsing by something more nice / standard..

closed 2020-04-14 by Nightwalker-87