stlink-org/stlink #379: STM32L152RB chip id not found in SWD mode

pengshulin, 2016-03-01:
I connected a standalone st-linkv2(not the board integrated one) with STM32L152RBT6 embedded in Discovery board in SWD mode (RST+CLK+DAT, I've disconnect the three pins with the integrated stlink), but failed to find the chip.  The integrated stlink works well.
In the debugging info, core_id is identified (I think the hardware connection no problem), but reads 0 from DBGMCU_IDCODE(@0xE0042000).
The stlink is upgraded to the latest firmware and st-flash tool is built from the master branch.
Can anyone help me?

Here's the debug error info:
`10:26:57 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)`
`10:26:57 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)`
`10:26:57 DEBUG src/stlink-common.c: *** stlink_reset ***`
`10:26:57 DEBUG src/stlink-common.c: *** looking up stlink version`
`10:26:57 DEBUG src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)`
`10:26:57 DEBUG src/stlink-common.c: stlink pid     = 0x3748`
`10:26:57 DEBUG src/stlink-common.c: stlink version = 0x2`
`10:26:57 DEBUG src/stlink-common.c: jtag version   = 0x18`
`10:26:57 DEBUG src/stlink-common.c: swim version   = 0x4`
`10:26:57 INFO src/stlink-common.c: Loading device parameters....`
`10:26:57 DEBUG src/stlink-common.c: *** stlink_core_id ***`
`10:26:57 DEBUG src/stlink-common.c: core_id = 0x2ba01477`
`10:26:57 DEBUG src/stlink-common.c: *** stlink_read_debug32 0 from 0xe0042000`
`10:26:57 DEBUG src/stlink-common.c: *** stlink_read_debug32 0 from 0x40015800`
`10:26:57 WARN src/stlink-common.c: unknown chip id! 0`
`10:26:57 DEBUG src/stlink-common.c: *** stlink_close ***`

If I use the integrated stlink, it works well, here's the right info:
`16:36:23 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)`
`16:36:23 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)`
`16:36:23 DEBUG src/stlink-common.c: *** stlink_reset ***`
`16:36:23 DEBUG src/stlink-common.c: *** looking up stlink version`
`16:36:23 DEBUG src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)`
`16:36:23 DEBUG src/stlink-common.c: stlink pid     = 0x3748`
`16:36:23 DEBUG src/stlink-common.c: stlink version = 0x2`
`16:36:23 DEBUG src/stlink-common.c: jtag version   = 0x18`
`16:36:23 DEBUG src/stlink-common.c: swim version   = 0x0`
`16:36:23 DEBUG src/stlink-common.c:     notice: the firmware doesn't support a swim interface`
`16:36:23 INFO src/stlink-common.c: Loading device parameters....`
`16:36:23 DEBUG src/stlink-common.c: *** stlink_core_id ***`
`16:36:23 DEBUG src/stlink-common.c: core_id = 0x2ba01477`
`16:36:23 DEBUG src/stlink-common.c: *** stlink_read_debug32 10186416 from 0xe0042000`
`16:36:23 DEBUG src/stlink-common.c: *** stlink_read_debug32 80 from 0x1ff8004c`
`16:36:23 INFO src/stlink-common.c: Device connected is: L1 Med-density device, id 0x10186416`
`16:36:23 INFO src/stlink-common.c: SRAM size: 0x4000 bytes (16 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 256 bytes`
`16:36:23 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)`
`16:36:23 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)`
`16:36:23 DEBUG src/stlink-common.c: *** stlink_force_debug_mode ***`
`16:36:23 DEBUG src/stlink-common.c: *** stlink_status ***`
`16:36:23 DEBUG src/stlink-common.c:   core status: halted`

pengshulin, 2016-03-01:
I found the reason: I forgot to connect VCC pin with stlink.
Now only VCC/CLK/IO/GND is needed, that's great!

xor-gate, 2016-05-02:
@pengshulin please close this issue if not applicable anymore. Thanks!

closed 2016-05-14 by xor-gate