stlink-org/stlink #380: STM32L031K6T6 unknown chip id (0x425)

raron, 2016-03-06:
Hello,

I try to connect to a NUCLEO-L031K6 board, but I get this message:
`WARN src/stlink-common.c: unknown chip id! 0x20006425`

There is a ST-LINK/V2-1 debugger in this board. It is "mbed compatible" and I can upload programs by copy the binary code file to the virtual USB mass-storage device.

There is only one chip id (0x417) in `stlink-common.h`, but in the [RM0377](http://www.st.com/st-web-ui/static/active/en/resource/technical/document/reference_manual/DM00108282.pdf) at 27.4.1 (page 799) there are multiple values for different category devices (0x425 for category 2).

I added a new device based on `STM32_CHIPID_L0` with the 0x425 device ID. I can read the flash content (witch is equal what I uploaded with the mass-storage method), but if I try to program the chip I get this message:
```
2016-03-06T18:16:10 INFO src/stlink-common.c: Attempting to write 17488 (0x4450) bytes to stm32 address: 134217728 (0x8000000)
2016-03-06T18:16:10 WARN src/stlink-common.c: unknown coreid bc11477, page erase failed
2016-03-06T18:16:10 ERROR src/stlink-common.c: Failed to erase_flash_page(0x8000000) == -1
stlink_fwrite_flash() == -1
```
I uploaded the full debug output [here](http://pastebin.com/xMTCTCBw).

Do I have to do something besides adding new device in `stlink-common.h`?
Does stlink work with these "mbed-enabled" debuggers, or maybe I have to upgrade the firmware of the debugger?

Thank you, this is a great tool

xor-gate, 2016-05-14:
Could you try with the new release `v1.2.0`?

raron, 2016-05-14:
It's the same.
After git checkout 1.2.0; and build: http://pastebin.com/S4X54Lvx
Still no `STM32_CHIPID_` definition for this chip (0x425) in the `stlink-common.h`.

xor-gate, 2016-05-15:
Could you provide a pull request to add the chip id?

raron, 2016-05-16:
I created a pull request. I tried it with the NUCLEO-L031K6 board. Maybe someone else should also test it.

commit 9cf48a76 (2016-05-16): Merge pull request #414 from raron/master

Add L0 Category 2 device (chip id: 0x425), fixes #380
closed 2016-05-16 by xor-gate