stlink-org/stlink #381: unknown chip id

wonkodv, 2016-03-06:
There are issues for this already but with less detail

I have a standalone green ST-Link V2 usb dongle.

I used the st-linkv2 for a while, debugging an stm32f030 and everything worked.
After I had to kill gdb today, `st-util` stopped working.
When I called `st-util` I got the following error message

    2016-03-06T20:50:39 INFO src/stlink-common.c: Loading device parameters....
    2016-03-06T20:50:39 WARN src/stlink-common.c: unknown chip id! 0xe0042000

Independent of connecting the stm32f0 or not, restarting, etc. I allways got the same message.

I used ST-LINK/V2 firmware upgrade (STSW-LINK005) to update the ST-Link V2 to the newest firmware (V2J24S4)
The message changed, I now get the following message: 

    2016-03-06T21:19:47 INFO src/stlink-common.c: Loading device parameters....
    2016-03-06T21:19:47 WARN src/stlink-common.c: unknown chip id! 0

It does *not* depend on the connected device, I can connect one, disconnect one, ... the message does not change.
I have Just pulled the newest stlink github version. No changes.

With verbosity I get the following output.
```
2016-03-06T21:24:49 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)
2016-03-06T21:24:49 DEBUG src/stlink-common.c: stlink current mode: debug (jtag or swd)
2016-03-06T21:24:49 DEBUG src/stlink-common.c: *** stlink_reset ***
2016-03-06T21:24:49 DEBUG src/stlink-common.c: *** looking up stlink version
2016-03-06T21:24:49 DEBUG src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)
2016-03-06T21:24:49 DEBUG src/stlink-common.c: stlink pid     = 0x3748
2016-03-06T21:24:49 DEBUG src/stlink-common.c: stlink version = 0x2
2016-03-06T21:24:49 DEBUG src/stlink-common.c: jtag version   = 0x18
2016-03-06T21:24:49 DEBUG src/stlink-common.c: swim version   = 0x4
2016-03-06T21:24:49 INFO src/stlink-common.c: Loading device parameters....
2016-03-06T21:24:49 DEBUG src/stlink-common.c: *** stlink_core_id ***
2016-03-06T21:24:49 DEBUG src/stlink-common.c: core_id = 0x00000000
2016-03-06T21:24:49 DEBUG src/stlink-common.c: *** stlink_read_debug32 0 is 0xe0042000
2016-03-06T21:24:49 DEBUG src/stlink-common.c: *** stlink_read_debug32 0 is 0x40015800
2016-03-06T21:24:49 WARN src/stlink-common.c: unknown chip id! 0
2016-03-06T21:24:49 DEBUG src/stlink-common.c: *** stlink_close ***
```

`st-info --chipid` does not print anything, return code is 255 but my dongle blinks for about a second.

simonjwright, 2016-03-11:
I’ve been building my own startup code (in Ada!) for stm32f429i-disco and got into exactly this situation. I fixed it by connecting up my J-Link Edu (as described in an application note [here](https://www.segger.com/downloads/appnotes) and loading a good image; after that st-util works OK.

When I say “OK”, until I upgraded to the latest stlink I had to unplug the board/plug it in again before st-util would connect; now it has 3 goes,

`$ st-util
[!] send_recv
[!] send_recv
2016-03-11T15:40:25 INFO src/stlink-common.c: Loading device parameters....
2016-03-11T15:40:25 INFO src/stlink-common.c: Device connected is: F42x and F43x device, id 0x10036419
2016-03-11T15:40:25 INFO src/stlink-common.c: SRAM size: 0x40000 bytes (256 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 16384 bytes
2016-03-11T15:40:25 INFO gdbserver/gdb-server.c: Chip ID is 00000419, Core ID is  2ba01477.
2016-03-11T15:40:25 INFO gdbserver/gdb-server.c: Target voltage is 2870 mV.
2016-03-11T15:40:25 INFO gdbserver/gdb-server.c: Listening at *:4242..
`

The COM LED is initially red, and there’s a couple of seconds pause before the first error message.

prusnak, 2016-03-18:
Most probably a duplicate of https://github.com/texane/stlink/issues/107

xor-gate, 2016-05-03:
Please test with latest stlink (master) and newest stlink firmware. Reopen when issue still remains.

Nightwalker-87, 2020-07-18:
This should be verified with the `v1.3.0-patch` branch and a recent ST-Link V2 firmware.

simonjwright, 2020-08-14:
The `v1.3.0-patch` branch correctly identified my STM32F40 (id `0413`) and STM32F429I (id `0419`) boards, in both cases using the embedded ST-LINK.

Nightwalker-87, 2020-08-14:
@simonjwright: Thx for testing this and providing feedback.
If someone could successfully verify with a STM32f030 we would be able to close this issue as resolved and fixed.

Nightwalker-87, 2021-03-13:
The original issuer likely used the stable version of the tools which was v1.1.0 by that time.
This issue was addressed during ongoing development in #367.

closed 2021-03-13 by Nightwalker-87