stlink-org/stlink #385: STM32F0x error

duro80, 2016-03-16:
After update  from 14/mar/2016 is broken support of STM32F0x.
The utility "st-info --chipid" (or other parameter) ends with no output. The utility st-flash exit with "unknown chipid" error.

texane, 2016-03-16:
did you try with latest fix ?

duro80, 2016-03-16:
Yes. 
Today, I pull the repository and make executables.

texane, 2016-03-16:
Can you confirm it is the current, head, 'TODAY' is too vague

duro80, 2016-03-16:
Sorry. The error is from commit 93e958f.

texane, 2016-03-16:
Thanks. But what I meant is, did you actually pulled the latest version
since I fixed something this afternoon (ie. stlink_version return code)

jape3, 2016-03-16:
Hi,

The problem seems to be linked with this piece of code from stlink-common.c;
`int stlink_chip_id(stlink_t *sl, uint32_t *chip_id) {
    int ret;

    //ret = stlink_read_debug32(sl, 0xE0042000, chip_id);
    ret = stlink_read_debug32(sl, 0x40015800, chip_id);
    if (ret == -1)
        return ret;

    if (chip_id == 0)
        ret = stlink_read_debug32(sl, 0x40015800, chip_id);    //Try Corex M0 DBGMCU_IDCODE register address

    return ret;
}`

So I replaced the 1st adress to try with the one from M0 and it worked. I haven't figured out yet if the problem is in the code section I posted or somewhere else related?

duro80, 2016-03-16:
Error is here:
file: stlink-common.c;

```
int stlink_chip_id(stlink_t *sl, uint32_t *chip_id) {
    int ret;
    ret = stlink_read_debug32(sl, 0xE0042000, chip_id);
    if (ret == -1)
        return ret;
//    if (chip_id == 0)  <---- chip_id is address!
/* correct way is use pointer!! */
    if (*chip_id == 0)
        ret = stlink_read_debug32(sl, 0x40015800, chip_id);    //Try Corex M0 DBGMCU_IDCODE register address
    return ret;
}
```

texane, 2016-03-16:
fixed, thanks

closed 2016-03-16 by duro80