stlink-org/stlink #387: Could you help to support STM32L072CZ for stink V2?

andy-y-li, 2016-03-23:
Could you help to support STM32L072CZ for stink V2?
For XP, it has support.
  On mac, it is fail at "WLOG("unknown chip id! %#x\n", chip_id);":
unknown chip id! xxxxxx447

gluedig, 2016-03-24:
you might try with my repo as i added support for L073RZ device which is pretty simmilar

xor-gate, 2016-05-03:
@gluedig could you prepare a reviewable pull request against master?

xor-gate, 2016-05-14:
@gluedig is the `STM32L072CZ` now supported with latest changes? @andy-y-li could you test the tooling from `master` branch?

gluedig, 2016-05-16:
@xor-gate i have no way of verifying on L072 as i have no such device, only L073
quick glance at datasheet indicates L072 is L073 minus LCD driver

xor-gate, 2016-05-16:
I understand, so basicly the current stlink tool flash loader should work out of the box.

Nightwalker-87, 2020-03-16:
Please verify with Release v1.6.0.

Nightwalker-87, 2020-03-20:
@andy-y-li: Please verify if this has been resolved in Release v1.2.0 and v1.3.0.

closed 2020-04-06 by Nightwalker-87