stlink-org/stlink #390: [breaking] Error: Data length doesn't have a 32 bit alignment: +2 byte.

Jegeva, 2016-04-05:
in the lasted version of stlink:
Error: Data length doesn't have a 32 bit alignment: +2 byte.

older version is working fine (same board, same main.bin) : 

OLDVERSION:
st-flash write main.bin 0x8000000
2016-04-06T00:12:51 INFO src/stlink-usb.c: -- exit_dfu_mode
2016-04-06T00:12:51 INFO src/stlink-common.c: Loading device parameters....
2016-04-06T00:12:51 INFO src/stlink-common.c: Device connected is: L1 Medium-Plus-density device, id 0x10086427
2016-04-06T00:12:51 INFO src/stlink-common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 256 bytes
2016-04-06T00:12:51 INFO src/stlink-common.c: Attempting to write 16816 (0x41b0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004100 erased
2016-04-06T00:12:51 INFO src/stlink-common.c: Finished erasing 66 pages of 256 (0x100) bytes
2016-04-06T00:12:51 INFO src/stlink-common.c: Starting Half page flash write for STM32L core id
2016-04-06T00:12:51 INFO src/stlink-common.c: Successfully loaded flash loader in sram
131/131 halfpages written
2016-04-06T00:12:53 INFO src/stlink-common.c: Starting verification of write complete
2016-04-06T00:12:53 INFO src/stlink-common.c: Flash written and verified! jolly good!

NEW VERSION: 
~/stlink/st-flash write main.bin 0x8000000
2016-04-06T00:13:28 INFO /home/jg/stlink/src/stlink-common.c: Loading device parameters....
2016-04-06T00:13:28 INFO /home/jg/stlink/src/stlink-common.c: Device connected is: L1 Medium-Plus-density device, id 0x10086427
2016-04-06T00:13:28 INFO /home/jg/stlink/src/stlink-common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 256 bytes
2016-04-06T00:13:28 INFO /home/jg/stlink/src/stlink-common.c: Attempting to write 16816 (0x41b0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004100 erased
2016-04-06T00:13:29 INFO /home/jg/stlink/src/stlink-common.c: Finished erasing 66 pages of 256 (0x100) bytes
2016-04-06T00:13:29 INFO /home/jg/stlink/src/stlink-common.c: Starting Half page flash write for STM32L core id
Error: Data length doesn't have a 32 bit alignment: +2 byte.
Aborted

gluedig, 2016-04-07:
what device are you using?
and what is the old version that works ok?

Jegeva, 2016-04-12:
stm32l100 discovery, 
git log --pretty=format:'%h' -n 1 -> 99ed475

tomdeboer, 2016-05-04:
I can confirm this. Last working commit is adbb6d270cd91de21307a1ad0c1644e321820207.

After merging #384, things got broken. Not just for the medium plus devices, but also for the STM32L151 I am using (which is a regular medium device) and the L152RE (a Nucleo board I happened to have).

xor-gate, 2016-05-04:
@mcoquelin-stm32 you refactored all code in PR #384, can you test this as described below with latest master. As I have no hardware laying around to verify/debug this and currently have no in-depth knowledge of the internal workings.

mcoquelin-stm32, 2016-05-05:
Hi,
Thanks for reporting the issue, and sorry for the regression I introduced.
I won't have any stm32 hw for the next two weeks to test any change,
but from code review, I think I found the cause of the problem.

I removed an instruction in the stm32l and stm32l0 loaders, so it is no more aligned on 32 bits.

Could you test the following patch, and let me know if it solved your issue?
https://github.com/mcoquelin-stm32/stlink/commit/1b8a5ed035566dafd9620b61f550aa8a7611964d

Thanks in advance,
Maxime

xor-gate, 2016-05-05:
@tomdeboer and @Jegeva could you verify on your hardware with this patch on current master it fixes the issue?

gluedig, 2016-05-06:
I tried the patched version on L073RZ device, without success,
the loader runs but does not finish.
I prepared a modified loader on https://github.com/gluedig/stlink/commit/aa54cc2f6d1ec7596849ffdd562b0686f03dbe25
but only for L0 devices as i cannot verify on any other.

Still there seems to be some issue (probably unrelated) with flashing using st-flash utility.
Loading via gdb/st-util works nice with full speed, st-flash sometimes works, sometimes falls back to slower speed (especially after whole device erase).

mcoquelin-stm32, 2016-05-06:
Hi @gluedig,

I pushed a v2 fixing the missing comma, but again, I cannot test it for now.
I see your comment saying it is not enough, so if you version works it is of course good to me.

The reason I modified the loaders was to use a common ABI with all version so that the code can be simplified.

Thanks,
Maxime

gluedig, 2016-05-06:
Hi @mcoquelin-stm32 
i think the culprit is probably 
line 1549: 0xf8, 0xd3, which i think is 'bcc' - so loop if lower (no carry)
it checks the result of comparison in previous line (0x00, 0x2a, - cmp	r2, #0)
and as r2 is used to specify number of words to write (and as a return value) the loop never actually runs and r2 still contains original value at exit, signifying an error
in my code the line is changed to to bhi - so loop if higher (carry, no zero)

xor-gate, 2016-05-07:
@Jegeva and @gluedig can you verify it has been fixed with PR #407, then we can call this issue a go 👍. Thanks all for the response and discussion to get this out of the way.

gluedig, 2016-05-08:
i tried on L053, it works ok

tomdeboer, 2016-05-08:
@gluedig I read your message after creating the previous fix. I overlooked the instruction you mentioned, but after trying it and verifying it works, I think it's should be preferred over my previous fix, which was basically a rollback to the last-working commit.

Also, looking at the stm32lx.S loader from OpenOCD, I believe the loaders for the STM32L1 and the STML0 are actually the same. Would you be able to make the L0 device also use the `loader_code_stm32l` loader and verify this? I only have L1 devices, so I am not able to try this myself.

gluedig, 2016-05-09:
@tomdeboer to understand correctly:
you took the version before merge #407, used changed L0 bootloader from my branch (in https://github.com/gluedig/stlink/commit/aa54cc2f6d1ec7596849ffdd562b0686f03dbe25) as L1 bootloader  and verified it on a L1 device?
if that is the case i cannot verify anything more - already made sure that L0 loader is working
one thing left to do is revert #407, apply fix for L0 bootloader and use this for both L0 and L1 devices

xor-gate, 2016-05-10:
Hi guys, can we call this issue done with the merge?

gluedig, 2016-05-10:
works for me

tomdeboer, 2016-05-10:
Yeah I think we can close it

xor-gate, 2016-05-10:
Thanks all for your time, effort and fixing!

valeros, 2016-07-06:
Hello!
One of our users [reported](https://community.platformio.org/t/upload-to-nucleo-l152re-failed-upload-protocol-stlink/468) that he has the same problem and looks like uploading to some `L0/L1` targets is broken in `1.2.0` release (tested on Mac, Windows, Linux with `NUCLEO-L152RE` and `NUCLEO-L053R8` boards).
```
2016-07-06T14:41:45 INFO src/stlink-common.c: Loading device parameters....
2016-07-06T14:41:45 INFO src/stlink-common.c: Device connected is: L0x3 device, id 0x10086417
2016-07-06T14:41:45 INFO src/stlink-common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 128 bytes
2016-07-06T14:41:45 INFO src/stlink-common.c: Ignoring 4 bytes of 0x00 at end of file
2016-07-06T14:41:45 INFO src/stlink-common.c: Attempting to write 19784 (0x4d48) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004d00 erased
2016-07-06T14:41:47 INFO src/stlink-common.c: Finished erasing 155 pages of 128 (0x80) bytes
2016-07-06T14:41:47 INFO src/stlink-common.c: Starting Half page flash write for STM32L core id
Error: Data length doesn't have a 32 bit alignment: +2 byte.

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
```
At the same time uploading to `NUCLEO-F401RE` works fine. Any hints?
Thanks in advance!

xor-gate, 2016-07-06:
@valeros  have you tried latest code in `master`?

valeros, 2016-07-06:
Hi @xor-gate !
Yes, I tried but uploading also fails with a different error:
```
2016-07-06T09:07:13 INFO src/common.c: Loading device parameters....
2016-07-06T09:07:13 INFO src/common.c: Device connected is: L152RE, id 0x10006437
2016-07-06T09:07:13 INFO src/common.c: SRAM size: 0x14000 bytes (80 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 256 bytes
2016-07-06T09:07:13 INFO src/common.c: Ignoring 4 bytes of 0x00 at end of file
2016-07-06T09:07:13 INFO src/common.c: Attempting to write 13388 (0x344c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08003100 erased[!] send_recv read reply failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
2016-07-06T09:07:21 WARN src/common.c: pecr.prglock not clear (0x6)
2016-07-06T09:07:21 ERROR src/common.c: Failed to erase_flash_page(0x8003200) == -1
[!] send_recv send request failed: LIBUSB_ERROR_NO_DEVICE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv read reply failed: LIBUSB_ERROR_NO_DEVICE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
stlink_fwrite_flash() == -1
```

xor-gate, 2016-07-06:
@valeros please open a clean new issue for this and describe your setup details and software version. Thanks!

closed 2020-04-06 by Nightwalker-87