stlink-org/stlink #399: Nucleo L476RG doesn't return ascii string... workaround

petertorelli, 2016-04-23:
I found that the Nucleo L476RG returns LIBUSB_ERROR_OTHER after the call to libusb_get_string_descriptor_ascii in stlink-common.c. No idea why, but if the code "break"s after closing the handle it works fine. Something simple like below. Any issues checking this in?

Thanks,
Peter

```
diff --git a/src/stlink-usb.c b/src/stlink-usb.c
index f3f41e3..ecbe9dd 100644
--- a/src/stlink-usb.c
+++ b/src/stlink-usb.c
@@ -776,6 +776,11 @@ stlink_t *stlink_open_usb(enum ugly_loglevel verbose, bool reset, char serial[16
                                                                  (unsigned char *)sl->serial, sizeof(sl->serial));
             libusb_close(handle);
 
+            if (sl->serial_size == LIBUSB_ERROR_OTHER) {
+                WLOG("Warning: STLink device didn't return an ASCII descriptor\n");
+                break;
+            }
+
             if (sl->serial_size < 0)
                  continue;
             if (serial == NULL)

```

texane, 2016-04-24:
OK. This is a valid error, so we cannot ignore it as you propose. What I
commit first check if there is
a serial to be checked. The top code has to take care of *serial = 0 or
serial = null. the test condition
could be made outside the if () too. Commit: 6fe3a02

texane, 2016-04-24:
Please close the issue if applicable

2016-04-24 9:32 GMT+02:00 lementec fabien <fabien.lementec@gmail.com>:

xor-gate, 2016-05-02:
@petertorelli, please test with latest master. Some refactoring has been done and close if applicable as already mentioned by @texane. Thanks!

petertorelli, 2016-05-02:
We're good. Thanks!

Closing.

FYI ---
There are few compiler warnings:

```
src/stlink-usb.c:950:19: warning: incompatible pointer to integer conversion assigning to 'char' from
      'void *' [-Wint-conversion]
          *serial = NULL;
                  ^ ~~~~

src/stlink-common.c:173:24: warning: unused function 'read_flash_wrpr' [-Wunused-function]
static inline uint32_t read_flash_wrpr(stlink_t *sl) {
                       ^
src/stlink-common.c:179:24: warning: unused function 'read_flash_obr' [-Wunused-function]
static inline uint32_t read_flash_obr(stlink_t *sl) {
                       ^
src/stlink-common.c:369:24: warning: unused function 'read_flash_acr' [-Wunused-function]
static inline uint32_t read_flash_acr(stlink_t *sl) {
```

closed 2016-05-02 by petertorelli