stlink-org/stlink #409: gui won't compile

KonstantinBerezenko, 2016-05-10:
stlink-gui.c references stlink-common.h which no longer exists.

xor-gate, 2016-05-11:
You are right, I have reorganized the files and not enabled building the GUI with Travis continuous integration and only tested on Mac OS X without GUI. I will fix this.

xor-gate, 2016-05-11:
@KonstantinBerezenko i reorganized some things, could you test the `travis-build-gui` branch ? I have only tested on Debian 8. Note: `./configure --with-gtk` has changed to `./configure --with-gtk-gui`.

KonstantinBerezenko, 2016-05-11:
@xor-gate Compiles fine under Ubuntu, with a few warnings.  Works fine.  Thanks.

xor-gate, 2016-05-11:
@KonstantinBerezenko could you post the warnings and which build system you are using: CMake or Autoconf. Thanks!

KonstantinBerezenko, 2016-05-11:
@xor-gate Autoconf on Ubuntu 12.04.

$ ./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force 
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: configure.ac: not using Autoheader
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:17: installing `./compile'
configure.ac:14: installing `./config.guess'
configure.ac:14: installing `./config.sub'
configure.ac:8: installing `./install-sh'
configure.ac:8: installing `./missing'
src/tools/gui/Makefile.am: installing `./depcomp'
Makefile.am: installing `./INSTALL'
autoreconf: Leaving directory `.'
$ ./configure --with-gtk-gui
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for ranlib... ranlib
checking whether gcc and cc understand -c and -o together... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking sys/mman.h usability... yes
checking sys/mman.h presence... yes
checking for sys/mman.h... yes
checking sys/poll.h usability... yes
checking sys/poll.h presence... yes
checking for sys/poll.h... yes
checking for mmap... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for USB... yes
checking for usb_path2devnum in -lusbpath... no
checking for GTK3... yes
checking for GLIB... yes
configure: creating ./config.status
config.status: creating src/tools/gui/Makefile
config.status: creating Makefile
config.status: executing depfiles commands
[... 21 lines cut ...]
gcc -DPACKAGE_NAME=\"stlink\" -DPACKAGE_TARNAME=\"stlink\" -DPACKAGE_VERSION=\"1.2.0\" -DPACKAGE_STRING=\"stlink\ 1.2.0\" -DPACKAGE_BUGREPORT=\"davem@devkitpro.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"stlink\" -DVERSION=\"1.2.0\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_SYS_POLL_H=1 -DHAVE_MMAP=1 -I.  -std=gnu99 -Wall -Wextra -O2 -I./include -I./src/mingw   -g -O2 -I/usr/include/libusb-1.0   -MT src/gdbserver/st_util-gdb-remote.o -MD -MP -MF src/gdbserver/.deps/st_util-gdb-remote.Tpo -c -o src/gdbserver/st_util-gdb-remote.o `test -f 'src/gdbserver/gdb-remote.c' || echo './'`src/gdbserver/gdb-remote.c
mv -f src/gdbserver/.deps/st_util-gdb-remote.Tpo src/gdbserver/.deps/st_util-gdb-remote.Po
gcc -DPACKAGE_NAME=\"stlink\" -DPACKAGE_TARNAME=\"stlink\" -DPACKAGE_VERSION=\"1.2.0\" -DPACKAGE_STRING=\"stlink\ 1.2.0\" -DPACKAGE_BUGREPORT=\"davem@devkitpro.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"stlink\" -DVERSION=\"1.2.0\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_SYS_POLL_H=1 -DHAVE_MMAP=1 -I.  -std=gnu99 -Wall -Wextra -O2 -I./include -I./src/mingw   -g -O2 -I/usr/include/libusb-1.0   -MT src/gdbserver/st_util-gdb-server.o -MD -MP -MF src/gdbserver/.deps/st_util-gdb-server.Tpo -c -o src/gdbserver/st_util-gdb-server.o `test -f 'src/gdbserver/gdb-server.c' || echo './'`src/gdbserver/gdb-server.c
mv -f src/gdbserver/.deps/st_util-gdb-server.Tpo src/gdbserver/.deps/st_util-gdb-server.Po
gcc -DPACKAGE_NAME=\"stlink\" -DPACKAGE_TARNAME=\"stlink\" -DPACKAGE_VERSION=\"1.2.0\" -DPACKAGE_STRING=\"stlink\ 1.2.0\" -DPACKAGE_BUGREPORT=\"davem@devkitpro.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"stlink\" -DVERSION=\"1.2.0\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_SYS_POLL_H=1 -DHAVE_MMAP=1 -I.  -std=gnu99 -Wall -Wextra -O2 -I./include -I./src/mingw   -g -O2 -I/usr/include/libusb-1.0   -MT src/mingw/st_util-mingw.o -MD -MP -MF src/mingw/.deps/st_util-mingw.Tpo -c -o src/mingw/st_util-mingw.o `test -f 'src/mingw/mingw.c' || echo './'`src/mingw/mingw.c
mv -f src/mingw/.deps/st_util-mingw.Tpo src/mingw/.deps/st_util-mingw.Po
[... cut ...]
stlink-gui.c:462:5: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 2 has type ‘gsize’ [-Wformat]
mv -f .deps/stlink_gui-stlink-gui.Tpo .deps/stlink_gui-stlink-gui.Po
gcc   -DSTLINK_UI_DIR="\"/usr/local/share/stlink/ui\"" -g -O2 -I/usr/include/libusb-1.0     -o stlink-gui stlink_gui-stlink-gui.o  ../../../libstlink.a -lgtk-3 -lgdk-3 -latk-1.0 -lgio-2.0 -lpangocairo-1.0 -lgdk_pixbuf-2.0 -lcairo-gobject -lpango-1.0 -lcairo -lgobject-2.0 -lglib-2.0   -lusb-1.0  
make[1]: Leaving directory `/home/konstantin/work/answ/stlink/src/tools/gui'

xor-gate, 2016-05-11:
Thanks for posting I will rephrase the important path and label/reopen the issue because this is not very nice. Go ahead and use master now I have merged the fix from the branch. I wil fix the warning later:

```
stlink-gui.c: In function ‘hexstr_to_guint32’:
stlink-gui.c:186:9: warning: large integer implicitly truncated to unsigned type [-Woverflow]
stlink-gui.c:193:9: warning: large integer implicitly truncated to unsigned type [-Woverflow]
stlink-gui.c: In function ‘dev_format_mem_size’:
stlink-gui.c:462:5: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 2 has type ‘gsize’ [-Wformat]
```

xor-gate, 2016-05-16:
Quoting from a private mail I got:

> Hi,
> a cast should be used to meet the data types,
>  
> stlink-gui.c:186:9: warning: large integer implicitly truncated to unsigned type [-Woverflow]
>          return LONG_MAX;
>          ^
> stlink-gui.c:193:9: warning: large integer implicitly truncated to unsigned type [-Woverflow]
>          return LONG_MAX;
>          ^
> remember
>  
> guint32
> typedef unsigned int guint32;
> An unsigned integer guaranteed to be 32 bits on all platforms. Values of this type can range from 0 to G_MAXUINT32 (= 4,294,967,295).
> To print or scan values of this type, use G_GINT32_MODIFIER and/or G_GUINT32_FORMAT.
> 
> #define LONG_MAX   2147483647L
> Description  LONG_MAX is the maximum value for an object of type long int.
>  
> so use
> return ( guint32) LONG_MAX;
>  
> cheers,
>  
> jau (Uwe-Jens Austel)

PR #410 (2016-05-11): Reorganize stlink-gui and update README, add building gui with Travis
Fixes #409
PR #416 (2016-05-16): stlink_gui: Fix compiler warning on overflow
Fixes #409
closed 2020-04-14 by Nightwalker-87