stlink-org/stlink #411: are there plans to support  the STMF746 Discovery Kit?

rahmanih, 2016-05-13:
I'm  working with a STMF746 Discovery Kit,
http://www2.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-discovery-kits/32f746gdiscovery.html

it seems that the stlink doesn't support it, are there plans to support it?
what is needed to get it supported?

thanks.

xor-gate, 2016-05-13:
I'm not sure but the chip should be detected by `st-info --probe`. See https://github.com/texane/stlink/blob/master/include/stlink.h#L190-L198

Please test with the stlink tool and give the output.

rahmanih, 2016-05-13:
Sorry for the issue bad description but indeed the board is detected by the stlink using the st-info --probe. the actual issue is when flashing binary files I'm getting scrambled output on the LCD.

I you want we can close this issue and I'll open a new one with detailed description.

regards

closed 2016-05-13 by rahmanih