stlink-org/stlink #417: Nucleo stlink v2 and stlink v2 serial number incompatibility

xor-gate, 2016-05-16:
- [X] Programmer/board type: NUCLEO-F401RE
- [X] Programmer firmware version: n/a
- [X] Operating system: Mac OS X
- [X] Stlink tools version and/or git commit hash: n/a
- [X] Stlink commandline tool name: `st-util` 
- [X] Target chip (and optional board): e.g STM32F401RE (NUCLEO-F401RE)

Output:

```
$ system_profiler SPUSBDataType
        STM32 STLink:

          Product ID: 0x374b
          Vendor ID: 0x0483  (STMicroelectronics)
          Version: 1.00
          Serial Number: 0672FF494951785087023310
          Speed: Up to 12 Mb/sec
          Manufacturer: STMicroelectronics
$ ./st-info --probe
Found 1 stlink programmers
 serial: 303637324646343934393531373835
openocd: "\x30\x36\x37\x32\x46\x46\x34\x39\x34\x39\x35\x31\x37\x38\x35"
  flash: 524288 (pagesize: 16384)
   sram: 98304
 chipid: 0x0433
  descr: F4 device (Dynamic Efficency)
```

The current stlink tools/library is unable to detect a binary or hexstring serial number. The best fix is to detect if the serial is already hex encoded or encode it into a hex string. We need to have a generic `hexstr_from_bin`, `hexstr_to_bin` and `hexstr_is_valid`.

Nightwalker-87, 2020-04-18:
Who is able to contribute here?

Nightwalker-87, 2020-12-10:
@slyshykO Any ideas on this one?

slyshykO, 2020-12-10:
If convert openocd: "\x30\x36\x37\x32\x46\x46\x34\x39\x34\x39\x35\x31\x37\x38\x35" to "normal" ASCII we can see this string
"0672FF494951785". Looks similarly to 0672FF494951785087023310, at least first 15 signs.

tarek-bochkati, 2020-12-10:
please check http://openocd.zylin.com/#/c/5396/, it could be linked to this

Nightwalker-87, 2020-12-10:
Thanks @tarek-bochkati. Would you fancy dealing with this issue then?

Nightwalker-87, 2021-03-12:
@tarek-bochkati: Could you open a PR for this, as you were involved in this topic anyway?

tarek-bochkati, 2021-03-16:
PR #1106

PR #1106 (2021-03-20): Fixed old DFU serial number for STLINK programmers

(Closes #417, #494)
closed 2021-03-20 by Nightwalker-87