stlink-org/stlink #418: STM32F410 not supported

thebiguno, 2016-05-19:
Hello all,

I am trying to program an STM32F410RBTx chip with stlink, and it is failing with error:

wyatt@behemoth:~/stm32f410/led$ /opt/stlink/st-flash write led.bin 0x8000000
2016-05-18T21:59:59 INFO /src/common.c: Loading device parameters....
2016-05-18T21:59:59 WARN /src/common.c: unknown chip id! 0x10006458

Looking at include/stlink/chipid.h I can see that there is no entry for ID 458.  Unfortunately I am not clear on the different parameters in src/chipid.c to add this chip myself, as I am quite new to ARM programming.

I am programming the chip on a custom board via the NUCLEO F411RE dev board in target mode.  I know the programmer works, as I can program it correctly using st-link.  I know the custom board + chip works as I can program it via the Nucleo in USB mass storage mode.

I would prefer to use this command line tool for programming, as it integrates with Makefiles better than the USB mass storage programming mode; plus it will open up my options for other programmers (not just using the Nucleo).
- [ X ] Programmer/board type: NUCLEO F411RE board in target mode (CN2 not bridged)
- [ X ] Operating system: Debian Jessie
- [ X ] Stlink tools version and/or git commit hash: 1.2.0-27-gad39787
- [ X ] Stlink commandline tool name: st-flash
- [ X ] Target chip (and optional board): STM32F410RBTx

Output:

```
wyatt@behemoth:~/stm32f410/led$ /opt/stlink/st-flash write led.bin 0x8000000
2016-05-18T21:59:59 INFO /src/common.c: Loading device parameters....
2016-05-18T21:59:59 WARN /src/common.c: unknown chip id! 0x10006458

```

Expected/description:
`st-link does not appear to support the STM32F410RBTx chip.`

Thank you,
The stlink project maintainers

xor-gate, 2016-05-19:
Could you test with https://github.com/texane/stlink/commit/9c635e419deca697ff823000aad2e39d47ec8d6c I have not tested it on real hardware.

thebiguno, 2016-05-20:
Sorry for the delay in testing, yesterday was crazy.

Anyway, it is getting closer, but not quite there yet.  This is the output now:

```
wyatt@behemoth:~/Documents/Electronics/projects/samples/arm/stm32f410/led$ /opt/stlink/st-flash write led.bin 0x8000000
2016-05-20T09:24:48 INFO /home/wyatt/Downloads/stlink/src/common.c: Loading device parameters....
2016-05-20T09:24:48 INFO /home/wyatt/Downloads/stlink/src/common.c: Device connected is: F410 device, id 0x10006458
2016-05-20T09:24:48 INFO /home/wyatt/Downloads/stlink/src/common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 16384 bytes
2016-05-20T09:24:48 INFO /home/wyatt/Downloads/stlink/src/common.c: Attempting to write 8024 (0x1f58) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2016-05-20T09:24:49 INFO /home/wyatt/Downloads/stlink/src/common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2016-05-20T09:24:49 INFO /home/wyatt/Downloads/stlink/src/common.c: Starting Flash write for F2/F4/L4
2016-05-20T09:24:49 ERROR /home/wyatt/Downloads/stlink/src/common.c: unknown coreid, not sure what flash loader to use, aborting!: 2ba01477
2016-05-20T09:24:49 WARN /home/wyatt/Downloads/stlink/src/common.c: Failed to write flash loader to sram!
2016-05-20T09:24:49 ERROR /home/wyatt/Downloads/stlink/src/common.c: init_flash_loader() == -1
stlink_fwrite_flash() == -1
```

Compiled version:
```
wyatt@behemoth:~/Downloads/stlink$ git describe --long --tags
1.2.0-30-g1507e34
```

Thanks!

xor-gate, 2016-05-20:
There is something wrong now with selection of the correct flash loader because the coreid is not recognized.

thebiguno, 2016-05-20:
Is this something that you need from me?  I would be happy to help but am not clear on what you need.

xor-gate, 2016-05-20:
I'm not sure yet either because the flash loader mechanism is build by other people. I'm trying to figure out why it doesn't recognized correctly.

thebiguno, 2016-05-20:
Gotcha.  When you figure it out, let me know what I can do to help.

xor-gate, 2016-05-20:
https://github.com/texane/stlink/blob/1507e34dd6455636724bb68b7c5c2253dd2fb77b/src/common.c#L1620-L1622

We need to add the F410 chipid to that flash-loader selector, hopefully fixed here:
https://github.com/texane/stlink/commit/8da1467bb1e1f28f3e7cf830a48c8cc61090e0d2

Let me know if it works (or not)

thebiguno, 2016-05-20:
Beautiful!

```
wyatt@behemoth:~/Documents/Electronics/projects/samples/arm/stm32f410/led$ /opt/stlink/st-flash write led.bin 0x8000000
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: Loading device parameters....
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: Device connected is: F410 device, id 0x10006458
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 16384 bytes
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: Attempting to write 8024 (0x1f58) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: Starting Flash write for F2/F4/L4
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 8024
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: Starting verification of write complete
2016-05-20T09:53:21 INFO /home/wyatt/Downloads/stlink/src/common.c: Flash written and verified! jolly good!
```

This looks to be working as far as I am concerned.  Thank you very much for your quick fix!

Cheers

closed 2016-05-20 by thebiguno

xor-gate, 2016-05-20:
Very nice, have fun 👍