stlink-org/stlink #422: STM32F4 Discovery Unknown Chip ID! 0

tlamb96, 2016-05-25:
- [Stlink/v2] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [STSW-STM32068 1.1.0] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [Ubuntu 14.04 ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ Most recent (downloaded it today)] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [st-flash] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util` 
- [STM32f4 Discovery] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.
I am following the basic tutorial found here: http://makeict.org/wiki/STM32F4_Discovery_on_Linux and when I get to "sudo st-flash write bin/main.bin 0x8000000" the LD1 and COM lights toggle on/off with a blue color and I get the output shown below. I have tried using your tool on other projects and it still doesn't work. Any help would be appreciated :)

Output:
2016-05-24T23:22:01 INFO src/common.c: Loading device parameters....
2016-05-24T23:22:01 WARN src/common.c: unknown chip id! 0
```

xor-gate, 2016-05-25:
Have you tried `v1.2.0` as you write you use old `v1.1.0` ?

tlamb96, 2016-05-25:
Hi xor-gate,
Thanks for the timely response! Where do I get v1.2.0 when ST has v1.1.0 as the most recent firmware? 
Found here:
http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32-standard-peripheral-libraries-expansions/stsw-stm32068.html 
And if it is my hardware, is there a complete hardware and firmware reset? When I connect my board via usb cable, the COM light is solid blue, LD2 is solid green, and all other LDs are off. When I try to erase or load via st-flash, the LD1 and COM LEDs toggle a blue light between each other. Does this indicate a hardware or firmware error?

xor-gate, 2016-05-25:
I think we had a miscommunication. With the:

> [ Most recent (downloaded it today)] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056

The tools/library version from `texane/stlink`. The programmer firmware version as you state is `STSW-LINK007 2.27.15`.

tlamb96, 2016-05-25:
Yes I have the STSW-LINK007 2.27.15 found from:
http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link007.html
Here is a screenshot of the output:
http://imgur.com/pz0qGaf
Do you know what could be wrong?

xor-gate, 2016-05-25:
The firmware seems good. Could you try with `st-info --probe` tool from `texane/stlink`?

tlamb96, 2016-05-25:
The output of the command:
Found 1 stlink programmers
Any other ideas? (And thank you so much for helping me, I have spent a lot of time trying to get this to work)

xor-gate, 2016-05-25:
Probably the stlink programmer is claimed by another application. The probe tool scans all the usb devices and tries to open them. Currently the ones which fail will not be displayed.

tlamb96, 2016-05-25:
I have tried to program the board from several machines (one Windows environment and two Unix environments) could that effect it? I also tried to use the offical ST-Link utility when I was on my windows machine, could that effect it? And is there a way to find which application is claiming the stlink and/or reset the stlink programmer?

tlamb96, 2016-05-25:
[Update] When I have nothing connected to my computer, I have:
Found 0 stlink programmers
But when I connect my board via usb:
Found 1 stlink programmers
Is this programmer stored on the board?

xor-gate, 2016-05-25:
No claiming means when the usb device is opened by an application. The programmer is located as the first chip on the board with usb connection. I have no clue what your problem is. Maybe you have no permissions to access the device. Could you try to run `sudo st-probe --info`?

tlamb96, 2016-05-25:
[Update] I realized I had the CN3 jumpers on which enables on board programming (rookie mistake) which is why st-info was detecting a programmer. Once I removed the jumpers, the st-info --probe command output:
Found 0 stlink programmers
When I ran sudo st-flash write blinky.bin 0x8000000 I got the following output:
2016-05-25T11:00:54 INFO src/common.c: Loading device parameters....
2016-05-25T11:00:54 WARN src/common.c: unknown chip id! 0
Once again I have an unknown chip id. Anything else that could be wrong?

tlamb96, 2016-05-25:
I fixed the issues by removing qstlink2 from my system which was probably the 1 programmer that st-info --probe was detecting. Now when I run sudo st-flash write blinky.bin 0x8000000 I get the following output:
http://pastebin.com/HAe5MAXs
which I saw other users were getting too. Unfortunately, I did not find a definite answer on how to fix this. Any advice?

xor-gate, 2016-05-30:
Have you tried to mass erase the flash first and then program? As related to: https://github.com/texane/stlink#sometimes-flashing-only-works-after-a-mass-erase

xor-gate, 2016-06-07:
@tlamb96 could you verify that after mass erase it works?

tlamb96, 2016-06-07:
@xor-gate My STM32F4 board was shipped with Read Out Protection to level 1 which disables any writing to the memory. As an STM32 novice, this took quite a lot of time to figure out but is now fixed. Thank you for your support and the stlink repo.

closed 2016-06-07 by tlamb96

xor-gate, 2016-06-07:
Damn, thats a shame -_-". Good luck, thank you for your contribution.

sw17ch, 2016-08-17:
I had the same issue. I used the Windows ST-Link utility to do an erase of the chip. (I also updated the firmware, but I don't think this is related.)

Is it possible to use this project to do the mass erase like the Windows ST-Link utility does without being able to read out the chip ID and core ID?

Nightwalker-87, 2020-03-16:
@sw17ch: Please open a new issue if you still desire an answer to your question or have it addressed. Thx. This issue is now permanently closed.

Nightwalker-87, 2020-04-06:
> Is it possible to use this project to do the mass erase like the Windows ST-Link utility does without being able to read out the chip ID and core ID?

No, not as far as I am aware of.