stlink-org/stlink #423: STM32F100 development kit does not work since change to libusb

galexander1, 2016-06-04:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropiate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropiate.

- [ ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util` 
- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.

Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

galexander1, 2016-06-04:
The call to that crashes my STM32F100 is happening from gdb-server, because it outputs the voltage for informational purposes (ILOG).  In other places, the target voltaeg is determined only for certain kinds of boards where that is necessary, and it does not cause this problem.

Pull request 424 fixes this issue for me.

And thank you for adding the libusb support!  Now that I'm past this issue, it works way more reliably than the sg mode interface did in the past.

xor-gate, 2016-06-04:
Which hardware target do you use ? The STM32VLDiscovery with STM32F100RBT6B ?
http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-discovery-kits/stm32vldiscovery.html

galexander1, 2016-06-04:
Yes, that's the one.  Though my board is old (2011), so I don't know if there are different revisions since then.

galexander1, 2016-06-04:
It is this chipid: STLINK_CHIPID_STM32_F1_VL_MEDIUM_LOW = 0x420,

xor-gate, 2016-06-04:
I'm currently don't have a test setup, in the upcoming weeks I would be able to verify. When removing reading of target voltage for gdb-server would not change the behaviour of the gdb interface. But i'm wondering how many people rely on it currently. In my opinion the gdb-server should not do this.

closed 2020-03-17 by Nightwalker-87