stlink-org/stlink #425: "[!] send_recv" error message is a little cryptic

galexander1, 2016-06-04:
To diagnose a recent problem (#423), I found screens full of "[!] send_recv", which was not meaningful to me - I could not even be certain it was an error message.  Pull request forthcoming.

galexander1, 2016-06-04:
I made pull request #426 to address this issue.  When the send_recv() fails, now it outputs the libusb error name and what it was trying to do (send request/read reply) when the failure happened.  And when the caller notices send_recv() failed, it also outputs the STLINK_... command name that it was attempting.  Makes it particularly easy to unambiguously diagnose when one stage in stlink access causes future requests to fail.

xor-gate, 2016-06-04:
Hi @galexander1 looks very good to me. @texane could you also review the PR #426? I would like to merge this.

texane, 2016-06-05:
sorry for delay, but looks good to me too

xor-gate, 2016-06-05:
Good, so my "silent" merge effort is not lost :=)

texane, 2016-06-05:
Of course it is not :)

closed 2020-03-17 by Nightwalker-87