stlink-org/stlink #427: libusb_set_configuration() failed

multivac61, 2016-06-04:
I have a [STMicroelectronics NUCLEO-L152RE](http://eu.mouser.com/Search/ProductDetail.aspx?R=NUCLEO-L152REvirtualkey51120000virtualkey511-NUCLEO-L152RE) development board. My OS is Ubuntu, I'm running the 4.4.0-22-generic and I just pulled the newes stlink v1.2.0 branch. When running `./st-util` I receiven an error saying that "libusb_set_configuration() failed" (see Output). My default user (and root) is both in the `dialout`̣ and `̣plugdev`̣ groups as you ask about in [this issue](https://github.com/texane/stlink/issues/312).

Output:

```
sudo ./st-util 
setting new configuration (0 -> 1)
2016-06-04T22:47:22 WARN src/stlink-usb.c: libusb_set_configuration() failed
[!] send_recv
2016-06-04T22:47:22 INFO gdbserver/gdb-server.c: Chip ID is 00000000, Core ID is  00000000.
[!] send_recv
2016-06-04T22:47:22 INFO gdbserver/gdb-server.c: Listening at *:4242...
^C[!] send_recv
```

Please let me know what I can do to help.

multivac61, 2016-06-04:
It should be noted that I have the newest version of libusb available on apt, 2:1.0.20-1.

xor-gate, 2016-06-05:
Could you try with latest `master` ? There have been some changes in the loader for STM32L0xx. It seems a different problem but maybe it works. As stated by the libusb documentation itself, it seems another application has claimed the usb device: [libusb_set_configuration doc](http://libusb.sourceforge.net/api-1.0/group__dev.html#ga186593ecae576dad6cd9679f45a2aa43). I'm not sure which causes this. Could you check `/etc/udev*` folder if there are rules which probably claim the device?

multivac61, 2016-06-06:
The only file now in `/etc/udev/rules.d` is `49-stlinkv2.rules` and I've executed the `udevadm` commands as well. On the latest `master`  I get the exact same error

```
sudo ./st-util 
setting new configuration (0 -> 1)
2016-06-06T09:59:43 WARN src/usb.c: libusb_set_configuration() failed
[!] send_recv
2016-06-06T09:59:43 INFO src/gdbserver/gdb-server.c: Chip ID is 00000000, Core ID is  00000000.
[!] send_recv
2016-06-06T09:59:43 INFO src/gdbserver/gdb-server.c: Listening at *:4242...
```

xor-gate, 2016-06-06:
I have no real clue what causes the problem. Have you tried `st-info --probe` ?

multivac61, 2016-06-06:
When running `st-info --probe` I get `Found 1 stlink programmers`.

multivac61, 2016-06-07:
Found the bug... When I run `dmesg | tail` after plugging in my development board I got this error message:

```
[  583.583636] usb 1-3: new full-speed USB device number 11 using xhci_hcd
[  583.713138] usb 1-3: New USB device found, idVendor=0483, idProduct=374b
[  583.713147] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  583.713151] usb 1-3: Product: STM32 STLink
[  583.713155] usb 1-3: Manufacturer: STMicroelectronics
[  583.713158] usb 1-3: SerialNumber: 0670FF495056805087193849
[  583.713556] usb 1-3: ep 0x84 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[  583.714124] usb 1-3: can't set config #1, error -71
```

which meant that the kernel wasn't picking up the device. [Googled around](https://bbs.archlinux.org/viewtopic.php?id=198044) for similar bugs and tried another usb cable. Now the kernel picks up the board and the output from st-util is the following:

```
sudo ./st-util 
2016-06-07T11:18:39 INFO src/common.c: Loading device parameters....
2016-06-07T11:18:39 INFO src/common.c: Device connected is: L152RE, id 0x10006437
2016-06-07T11:18:39 INFO src/common.c: SRAM size: 0x14000 bytes (80 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 256 bytes
2016-06-07T11:18:39 INFO src/gdbserver/gdb-server.c: Chip ID is 00000437, Core ID is  2ba01477.
2016-06-07T11:18:39 INFO src/gdbserver/gdb-server.c: Target voltage is 3238 mV.
2016-06-07T11:18:39 INFO src/gdbserver/gdb-server.c: Listening at *:4242...
```

Thanks for the help @xor-gate, appreciate it!

xor-gate, 2016-06-07:
I'm happy it has finaly been found. Sometimes it takes some investigation. Thank you for your contribution!

closed 2016-06-07 by xor-gate