stlink-org/stlink #428: Flashing does not work in deep sleep (stop) mode

andyg24, 2016-06-06:
First, thank you very much for developing this tool. It works great most of the time, and is orders of magnitude faster than OpenOCD for programming STM32 microcontrollers.

One issue I ran across is that st-flash does not seem to work when my CPU (stm32l031k6) is in stop mode. A workaround that I found is to press and release the reset button at exactly the same time as running the st-flash command; this doesn't work every time though.

I am on Linux (Ubuntu Trusty), have fresh st-flash (compiled from the latest snapshot in git), and am using a Nucleo STM32L031 board. I can attach a simple bin file that demonstrates the problem if it's helpful.

The command that I use is

st-flash --reset write bin/test.bin 0x8000000

The output that I get is

2016-06-06T12:39:08 INFO src/common.c: Loading device parameters....
2016-06-06T12:39:08 WARN src/common.c: unknown chip id! 0

but only if the CPU is in stop mode. Otherwise everything works as expected.

Apologies if this is a known issue and I haven't done enough searching to find it.

xor-gate, 2016-06-06:
I have seen this kind of issue before, which is reset-related and based on the state of the core. This needs to be investigated and reproduced, and of course fixed.

andyg24, 2016-06-06:
Thank you for the quick follow up. I am not familiar with the code, but could it be that you try to read chip ID prior to a JTAG reset? If so, that would explain the error, as the DBG interface is disabled in stop mode. A proposed fix then is to do a reset prior to reading the chip ID.

xor-gate, 2016-06-06:
The flow is in this order:

* src/tools/flash.c
* https://github.com/texane/stlink/blob/master/src/tools/flash.c#L184
* https://github.com/texane/stlink/blob/master/src/usb.c#L730
* https://github.com/texane/stlink/blob/master/src/usb.c#L879
* https://github.com/texane/stlink/blob/master/src/usb.c#L885
* https://github.com/texane/stlink/blob/c0177ec3872244fb04fc517397059f742e2b1cc3/src/common.c#L550
* https://github.com/texane/stlink/blob/c0177ec3872244fb04fc517397059f742e2b1cc3/src/common.c#L569

andyg24, 2016-06-07:
Thanks, that helps. Looks like the fix is to replace

usb.c, L879

`    if (reset) {
        stlink_reset(sl);
        usleep(10000);
    }`

with

`if (reset) {
        usleep(10000);
        stlink_jtag_reset(sl, 2);
        stlink_reset(sl);
    }
`

andyg24, 2016-06-07:
Please see #430

xor-gate, 2016-06-07:
Seems good to me, @texane what do you think?
There are also some other reset-related issues still open: 
https://github.com/texane/stlink/labels/error%2Freset

xor-gate, 2016-06-09:
@texane ping? :-).

Nightwalker-87, 2020-02-27:
Closed by PR #430.

commit 8fadfa91 (2016-08-07): Merge pull request #430 from andyg24/stop_mode_fix

Do a JTAG reset prior to reading CPU information, fixes #428 and #451.
closed 2020-03-17 by Nightwalker-87