stlink-org/stlink #431: Discussion: Autotools will be deprecated in 2.0.0

xor-gate, 2016-06-07:
As per 2.0.0 I will deprecate the use of autotools (autoconf, automake). As it is not as cross-platform minded as CMake. And I am unable to maintain two build systems. Installing on windows autotools is a pain and I want to create IDE or console projects. The reasons why I only want to support CMake:

* No obsecure m4-macro system, and have a decent programming language for the build system 
* Easy to maintain different targets with different compiler flags
* Generate files (from templates) in a cross-platform fashion (not using sed,grep awk etc)
* Builtin testing facility (CTest)
* Builtin packaging facility (CPack)
* Multiple output generators (Make, Xcode (OSX), Visual Studio)

One current use-case is:
Generate a `version.h` from `version.h.in` template and use it to notify users which version they run. Completely automatic created from git tags or `version.txt` file.

texane, 2016-06-07:
I agree with you. So lets deprecate autotools, unless there is a strong
argument about keeping it.

xor-gate, 2016-06-07:
Thank you @texane for the feedback, there are tons of other nice features why we should stick with cmake. I will keep things posted here when I need some discussion.

rikvdh, 2016-06-07:
👍  for CMake, I see no point in keeping autotools as well.

xor-gate, 2016-06-29:
Fixed by PR #434

closed 2020-03-17 by Nightwalker-87