stlink-org/stlink #433: ST-Link does not recognize STM32F767ZI

BirdBare, 2016-06-18:
- [ ] Programmer/board type: Stlink/v2-onboard
- [ ] Operating system: Mac OS X,
- [ ] Stlink commandline tool name: `st-util` 
- [ ] Target chip: STM32F767ZI Nucleo-144

Output:
2016-06-18T02:02:33 INFO src/stlink-common.c: Loading device parameters....
2016-06-18T02:02:33 WARN src/stlink-common.c: unknown chip id! 0x10006451

Expected/description:
I cannot get st-link to connect to my STM32F767ZI chip. I am wondering if this chip is supported at this time? 

Thank you,
Bradley

xor-gate, 2016-06-18:
@BirdBare could you try the custom branch, I have no hardware to test but it should work:
https://github.com/texane/stlink/tree/support_stm32f7x7x

BirdBare, 2016-06-18:
It is not working. I get the same Problem. Is my device ID incorrect?

Bradleys-MBP-2:stlink-support_stm32f7x7x bradleybare$ st-util
2016-06-18T10:45:59 INFO src/stlink-common.c: Loading device parameters....
2016-06-18T10:45:59 WARN src/stlink-common.c: unknown chip id! 0x10006451

Nightwalker-87, 2020-03-16:
Closed by #436.

PR #436 (2016-06-18): Support stm32f7x7x
Fixes #433
commit 3f7d0f9d (2016-06-18): Merge pull request #436 from texane/support_stm32f7x7x

Support stm32f7x7x, fixes #433
closed 2020-03-16 by Nightwalker-87