stlink-org/stlink #435: ST-Link cannot load flash loader into STM32F767ZI

BirdBare, 2016-06-18:
- [ ] Programmer/board type: Stlink/v2-onboard
- [ ] Operating system: Mac OS X,
- [ ] Stlink commandline tool name: `st-util` 
Using the STM32F7x7x support branch
- [ ] Target chip: STM32F767ZI Nucleo-144

Output:
During st-flash:
2016-06-18T11:57:03 INFO src/common.c: Loading device parameters....
2016-06-18T11:57:03 INFO src/common.c: Device connected is: F76xxx device, id 0x10006451
2016-06-18T11:57:03 INFO src/common.c: SRAM size: 0x5c000 bytes (368 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 2048 bytes
2016-06-18T11:57:03 INFO src/common.c: Attempting to write 2240 (0x8c0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000800 erasedEraseFlash - Sector:0x0 Size:0x800 
2016-06-18T11:57:03 INFO src/common.c: Finished erasing 2 pages of 2048 (0x800) bytes
2016-06-18T11:57:03 INFO src/common.c: Starting Flash write for F2/F4/L4
2016-06-18T11:57:03 ERROR src/flash_loader.c: unknown coreid, not sure what flash loader to use, aborting!: 5ba02477
2016-06-18T11:57:03 WARN src/flash_loader.c: Failed to write flash loader to sram!
2016-06-18T11:57:03 ERROR src/common.c: stlink_flash_loader_init() == -1
stlink_fwrite_flash() == -1

during st-util:
2016-06-18T11:57:48 INFO src/common.c: Loading device parameters....
2016-06-18T11:57:48 INFO src/common.c: Device connected is: F76xxx device, id 0x10006451
2016-06-18T11:57:48 INFO src/common.c: SRAM size: 0x5c000 bytes (368 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 2048 bytes
2016-06-18T11:57:48 INFO src/gdbserver/gdb-server.c: Chip ID is 00000451, Core ID is  5ba02477.
2016-06-18T11:57:48 INFO src/gdbserver/gdb-server.c: Listening at *:4242...

Expected/description:
St link cannot load flash loader into stm32f767. 

Thank you,
Bradley

xor-gate, 2016-06-18:
Hi @BirdBare could you test with latest commit from the branch (pull and make). It should work now because stlink was unable to detect the flash loader.

BirdBare, 2016-06-18:
You fixed it.

Thank you

closed 2020-04-14 by Nightwalker-87