stlink-org/stlink #439: Regression when programming STM32F4xx device from a STM32VL Discovery kit

dflogeras, 2016-06-21:
Hi, I often use a STM32VL Discovery as a programmer, because it was cheap and I don't really have any other use for it :)  Today I revived a project where the target is a homebrew STM32F407 board.  I tried to use the STM32VL Discovery as the USB programmer.  I know this used to work, but today I was getting a "Failed to read target voltage" and then a bunch of USB errors.  After which I would have to unplug the USB cable before I could communicate again.  This was on the current master.

I tried 1.1.0 and it worked, so I bisected and I think the problem was introduced in 71535104e47f2a96ce6a87a6da0fc4519bc64d15

If it matters, this is on Gentoo Linux, compiled with autotools, GCC 4.9.3.

The device is a  STM32F405RGT6W, and st-util -1 reports:

2016-06-21T17:04:33 INFO src/stlink-common.c: Device connected is: F4 device, id 0x10016413
2016-06-21T17:04:33 INFO src/stlink-common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
2016-06-21T17:04:33 INFO src/stlink-sg.c: Successfully opened a stlink v1 debugger
2016-06-21T17:04:33 INFO gdbserver/gdb-server.c: Chip ID is 00000413, Core ID is  2ba01477.

xor-gate, 2016-06-21:
I'm not sure, but I think stlink v1 is unable to read target voltage. Which firmware version for the program are do you use? Probably you should first try to upgrade as I have seen weird issues with broken stlink programmer firmware. Over here for v1:
http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link001.html

dflogeras, 2016-06-21:
Well, if I plug it into the Windows STM32 ST-LINK utility, and click upgrade->device connect, a dialog pops up and says "Old firmware loader detected on ST-Link/V2.  For full operation (especially SWV support), please contect ST for upgrading."

Then it reports the Firmware Version: V0.J0.S0 STM32 Debugger

And says it wants to upgrade to "V2.J24.S0".

All of that makes me nervous.  It seems to want to upgrade to V2, and I'm not sure if this device is supposed to have V2.  Also, on previous times I've upgraded the STLink firmware, I didn't get the warning dialog, but they were already V2 devices.

Any ideas?  The link you sent shows me a version "1.12.3"

dflogeras, 2016-06-21:
The plot thickens, if I download the utility from the link that you provided, it says my current firmware is "V4.J48.S61 JTAG Debugger" and it wants to upgrade to "V2.J14.S0". UH?

xor-gate, 2016-06-21:
I'm not sure, but it seems things are not working as expected even for the st tools themselves ;+(. I think in some time the stlink v1 is deprecated anyway. Its only a shame you can not use it. Have you tried openocd or to much hassle?

dflogeras, 2016-06-21:
I am not a huge fan of openocd.  I guess I can always just revert to 1.1.0 when using this board, unless you would put in an override for the voltage check if stlink detects a V1 interface?  I realize that is probably a request to muddy the codebase

xor-gate, 2016-06-21:
The voltage is used to select the correct flash loader algorithms because in low-power mode things are a little different and only byte-based writes are possible (if I remember correctly). So when programming a low-power device it is not possible with stlink v1. But the F4 should just program as expected as there are no low-power versions as I know.

I'm not sure how this would work for other people, and indeed it messes up the codebase. Could you try yourself by removing the part which breaks programming?

dflogeras, 2016-06-21:
Well, if I hack out the stlink_target_voltage() call and related decision in both stlink_write_flash() and stlink_flash_loader_write_to_sram() on master it works again.  Note, even defaulting to the old behaviour, but calling stlink_target_voltage() still breaks and results in a ton of USB errors.  Is there a way to tell whether the backend is V1 or V2 in those two functions, and simply revert to the old behaviour (32bit writes) if it is otherwise impossible to detect?  I'll attempt the patch, but I need guidance on if/how to detect V1/V2

xor-gate, 2016-06-21:
The `stlink_t` has a field `version.stlink_v` when `1`, it uses V1 API, or else when `2` it uses `V2` API.

dflogeras, 2016-06-21:
I created a pull request, please have a look.  I got an error message concerning travis-ci not having a valid yml file.

closed 2016-06-25 by xor-gate