stlink-org/stlink #443: Waveshare Mini ST-LINK/V2 unknown chip id! 0xe0042000

husnoo, 2016-07-04:
OS X 10.11.5 (15F34).

http://www.waveshare.com/ST-LINK-V2-mini-STM32.htm

```
uname -a
Darwin Nawals-MacBook-Air.local 15.5.0 Darwin Kernel Version 15.5.0: Tue Apr 19 18:36:36 PDT 2016; root:xnu-3248.50.21~8/RELEASE_X86_64 x86_64

```
```
st-util -v99
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: stlink current mode: debug (jtag or swd)
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: stlink current mode: debug (jtag or swd)
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** stlink_reset ***
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** looking up stlink version
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: st vid         = 0x0483 (expect 0x0483)
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: stlink pid     = 0x3748
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: stlink version = 0x2
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: jtag version   = 0x17
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: swim version   = 0x4
2016-07-04T11:06:46 INFO /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: Loading device parameters....
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** stlink_core_id ***
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: core_id = 0x00000000
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** stlink_read_debug32 e0042000 is 0xe0042000
2016-07-04T11:06:46 WARN /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: unknown chip id! 0xe0042000
2016-07-04T11:06:46 DEBUG /tmp/stlink-20160608-18806-p6zk82/stlink-1.2.0/src/stlink-common.c: *** stlink_close ***

```

xor-gate, 2016-07-04:
I see you are using a stlink clone, make sure you have te latest stlink firmware running. The `texane/stlink` software is untested with (chinese) clones and only community supported with `stlink/v2, stlink/v2-1`. Which are standalone programmers, nucleo and discovery boards.

xor-gate, 2016-07-04:
I'm sorry but I now see in the log you are using the following firmware version of the programmer:
`V2J23S4`, the latest is `V2J27Sxx`.

husnoo, 2016-07-04:
I've upgraded the firmware...

```
./st-util -v99
2016-07-04T13:33:47 DEBUG src/common.c: stlink current mode: mass
2016-07-04T13:33:47 DEBUG src/common.c: stlink current mode: mass
2016-07-04T13:33:47 DEBUG src/common.c: *** stlink_enter_swd_mode ***
2016-07-04T13:33:47 DEBUG src/common.c: *** stlink_reset ***
2016-07-04T13:33:47 DEBUG src/common.c: *** looking up stlink version
2016-07-04T13:33:47 DEBUG src/common.c: st vid         = 0x0483 (expect 0x0483)
2016-07-04T13:33:47 DEBUG src/common.c: stlink pid     = 0x3748
2016-07-04T13:33:47 DEBUG src/common.c: stlink version = 0x2
2016-07-04T13:33:47 DEBUG src/common.c: jtag version   = 0x1b
2016-07-04T13:33:47 DEBUG src/common.c: swim version   = 0x6
2016-07-04T13:33:47 INFO src/common.c: Loading device parameters....
2016-07-04T13:33:47 DEBUG src/common.c: *** stlink_core_id ***
2016-07-04T13:33:47 DEBUG src/common.c: core_id = 0x00000000
2016-07-04T13:33:47 DEBUG src/common.c: *** stlink_read_debug32 0 is 0xe0042000
2016-07-04T13:33:47 DEBUG src/common.c: *** stlink_read_debug32 0 is 0x40015800
2016-07-04T13:33:47 WARN src/common.c: unknown chip id! 0
2016-07-04T13:33:47 DEBUG src/common.c: *** stlink_close ***
```

xor-gate, 2016-07-04:
Maybe there are some hardware problems, make sure you have also connected the reset. If it is possible also try with the tool from ST itself, the STLink Utility under windows. You also don't specify you hardware setup (target CPU), voltages etc.

husnoo, 2016-07-04:
Hi, thanks for taking the time! I'm a newbie with STM widgets - I'm trying to reproduce this:

https://davidgf.net/page/41/e-ink-wifi-display

So my initial attempts are at accessing the STM through the ST-Link.

xor-gate, 2016-07-04:
I'm not sure but after a quick look it seems the connector is missing the nRST (MCU reset pin). And it is possible the CPU is not reset/initialized correctly and no chipid can be read. @texane does the nRST always be connected to the STM32xxx to get access?

husnoo, 2016-07-04:
Aha, it was a dry joint where I soldered a connector to device! It's connected now! Woop!

xor-gate, 2016-07-04:
Very very nice! You could also close the issue yourself. Thanks for your feedback, and good luck. Closing this now.

closed 2016-07-04 by xor-gate