stlink-org/stlink #446: OS/X 10.11.6 - no kext available - using 10.10 version causes kernel panic

fustbariclation, 2016-07-22:
- [X ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard

Stlink/v2 - ideally

- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15

- [ X] Operating system: e.g Linux, Mac OS X, Windows (with specific version)

OS/X 10.11.6 El Capitan 

- [X ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056

v1.2.0
======

Release date: 16 may 2016

- [X ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util` 

st-flash

st-flash will read - but not write

- [X ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

STM32F746G-DISCO

st-util
2016-07-22T10:20:49 INFO src/common.c: Loading device parameters....
2016-07-22T10:20:49 INFO src/common.c: Device connected is: F7 device, id 0x10016449
2016-07-22T10:20:49 INFO src/common.c: SRAM size: 0x50000 bytes (320 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c: Chip ID is 00000449, Core ID is  5ba02477.
2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c: Chip clidr: 09000003, I-Cache: off, D-Cache: off
2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c:  cache: LoUU: 1, LoC: 1, LoUIS: 0
2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c:  cache: ctr: 8303c003, DminLine: 32 bytes, IminLine: 32 bytes
2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c: D-Cache L0: 2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c: f003e019 LineSize: 8, ways: 4, sets: 32 (width: 10)
2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c: I-Cache L0: 2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c: f007e009 LineSize: 8, ways: 2, sets: 64 (width: 11)
2016-07-22T10:20:49 INFO src/gdbserver/gdb-server.c: Listening at *:4242...

The kexts available are:

stlink_shield10_10.kext	
stlink_shield10_6.kext	
stlink_shield10_8.kext	
stlink_shield10_11.kext	
stlink_shield10_7.kext	
stlink_shield10_9.kext

Missing is stlink_shield10_11.kext 

When I tried to modify and use stlink_shield10_10.kext - the system failed with a kernel panic.

xor-gate, 2016-07-22:
Hi @fustbariclation, I'm not sure but on my Macbook with OSX 10.11 I didn't need a kernel module at all. And I used only stlink/v2 and stlink/v2-1. There is no real "error message" from the tools. But "only" a kernel panic. Have you tried without the module? I think this is only necessary because OS X will claim the stlink/v1 which is also not supported by ST anymore.

fustbariclation, 2016-07-22:
Dear Jerry,

Thank you very much for those comments. I've been having trouble getting
this to work for some time, actually. I wanted to get it working on my
Raspberry Pi to begin with - and it did flash the board once, but I've not
managed to do it again - the serial /dev files are not there.

So, I tried on the Mac.

It gets the stats from the card  OK:

st-info --flash

0x100000

st-info --sram

0x50000

st-info --descr

F7 device

st-info --pagesize

0x800

st-info --chipid

0x0449

st-info --serial

303636434646353735323536383637

st-info --hla-serial

"\x30\x36\x36\x43\x46\x46\x35\x37\x35\x32\x35\x36\x38\x36\x37"

st-info --probe

Found 1 stink programmers

It even reads OK:

st-flash read code.out 0x800000 0x007777

2016-07-22T12:14:57 INFO src/common.c: Loading device parameters....

2016-07-22T12:14:57 INFO src/common.c: Device connected is: F7 device, id
0x10016449

2016-07-22T12:14:57 INFO src/common.c: SRAM size: 0x50000 bytes (320 KiB),
Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes

The problem comes trying to flash it:

st-flash read code.out 0x800000 0x007777

2016-07-22T12:14:57 INFO src/common.c: Loading device parameters....

2016-07-22T12:14:57 INFO src/common.c: Device connected is: F7 device, id
0x10016449

2016-07-22T12:14:57 INFO src/common.c: SRAM size: 0x50000 bytes (320 KiB),
Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes

sh-3.2#  st-flash write code.out 0x800000 0x007777

2016-07-22T12:15:37 WARN src/sg.c: Failed to find an stlink v1 by VID:PID

2016-07-22T12:15:37 ERROR src/sg.c: Could not open stlink devic

As you can see, it looks as if it's trying, and failing, to use stink v1.

This is why I tried to load the kext for the stlinkv1_macosx_driver

Regards,
Peter

yagamy4680, 2016-07-22:
Sorry for a stupid question. Where can I download those built kernel module files? E.g. `stlink_shield10_11.kext`

xor-gate, 2016-07-22:
@yagamy4680 these are located in the repository

yagamy4680, 2016-07-22:
Thanks for feedbacks. However, I've tried to clone the entire repository, and try to find any "kext" file but nothing.

```text
# yagamy @ grandia in /tmp [21:07:26]
$  git clone https://github.com/texane/stlink.git
Cloning into 'stlink'...
remote: Counting objects: 5419, done.
remote: Total 5419 (delta 0), reused 0 (delta 0), pack-reused 5419
Receiving objects: 100% (5419/5419), 14.51 MiB | 1.81 MiB/s, done.
Resolving deltas: 100% (3354/3354), done.
Checking connectivity... done.

# yagamy @ grandia in /tmp [21:08:58] C:1
$  find ./stlink | grep "kext"
```

Do I miss something, or shall I check out any branch to find those pre-built kernel modules?

xor-gate, 2016-07-22:
True, these are located in a tarball:
https://github.com/texane/stlink/tree/master/stlinkv1_macosx_driver

yagamy4680, 2016-07-22:
Thanks again. I've download that tarball file, and extract it, but still cannot find `stlink_shield10_11.kext`. (Other kernel module files are found: 10.10, 10.9, ...)

```text
$  find . | grep "kext"
./stlink_shield/build/Release/stlink_shield.kext
./stlink_shield/build/Release/stlink_shield.kext/Contents
./stlink_shield/build/Release/stlink_shield.kext/Contents/Info.plist
./stlink_shield/build/Release/stlink_shield.kext/Contents/Resources
./stlink_shield/build/Release/stlink_shield.kext/Contents/Resources/English.lproj
./stlink_shield/build/Release/stlink_shield.kext/Contents/Resources/English.lproj/InfoPlist.strings
./stlink_shield/build/Release/stlink_shield10_7.kext
./stlink_shield/build/Release/stlink_shield10_7.kext/Contents
./stlink_shield/build/Release/stlink_shield10_7.kext/Contents/Info.plist
./stlink_shield/build/Release/stlink_shield10_7.kext/Contents/Resources
./stlink_shield/build/Release/stlink_shield10_7.kext/Contents/Resources/English.lproj
./stlink_shield/build/Release/stlink_shield10_7.kext/Contents/Resources/English.lproj/InfoPlist.strings
./stlink_shield10_10.kext
./stlink_shield10_10.kext/Contents
./stlink_shield10_10.kext/Contents/Info.plist
./stlink_shield10_10.kext/Contents/Resources
./stlink_shield10_10.kext/Contents/Resources/English.lproj
./stlink_shield10_10.kext/Contents/Resources/English.lproj/InfoPlist.strings
./stlink_shield10_6.kext
./stlink_shield10_6.kext/Contents
./stlink_shield10_6.kext/Contents/Info.plist
./stlink_shield10_6.kext/Contents/Resources
./stlink_shield10_6.kext/Contents/Resources/English.lproj
./stlink_shield10_6.kext/Contents/Resources/English.lproj/InfoPlist.strings
./stlink_shield10_7.kext
./stlink_shield10_7.kext/Contents
./stlink_shield10_7.kext/Contents/Info.plist
./stlink_shield10_7.kext/Contents/Resources
./stlink_shield10_7.kext/Contents/Resources/English.lproj
./stlink_shield10_7.kext/Contents/Resources/English.lproj/InfoPlist.strings
./stlink_shield10_8.kext
./stlink_shield10_8.kext/Contents
./stlink_shield10_8.kext/Contents/Info.plist
./stlink_shield10_8.kext/Contents/Resources
./stlink_shield10_8.kext/Contents/Resources/English.lproj
./stlink_shield10_8.kext/Contents/Resources/English.lproj/InfoPlist.strings
./stlink_shield10_9.kext
./stlink_shield10_9.kext/Contents
./stlink_shield10_9.kext/Contents/Info.plist
./stlink_shield10_9.kext/Contents/Resources
./stlink_shield10_9.kext/Contents/Resources/English.lproj
./stlink_shield10_9.kext/Contents/Resources/English.lproj/InfoPlist.strings
```

fustbariclation, 2016-07-22:
No, it's not there.

That's part of the problem. I'm running 10.11, so the script fails. I tried
copying the kext for 10.10 and fixing up the references - and that's what
caused the kernel panic.

So, either, it'd be nice to have it working with out the kext, or, it'd be
nice to have a kext that works for 10.11.

Regards,
Peter

xor-gate, 2016-07-22:
I'm not sure because you have a F7 board and a stlink/v2 programmer and you should not need a kernel module at all. If I remember correctly this is only used to unclaim the usb-disk programmer from the kernel and let libusb use it.

yagamy4680, 2016-07-22:
Just notice that using stlink/v2 programmer without OSX kernel module can work normally. I will buy stlink/v2 programmer to test. Thanks for supports. 👍

fustbariclation, 2016-07-22:
Thank you for that thought.

I hope so. I thought so, to begin with. However, as I said, now I've used
Mac/OS, Linux (through VirtualBox), and my Raspberry Pi. In each case, this
USB <-> serial connection to flash the board has been the problem.

It's not clear to me why it's able to read from the board, easily, but
simply trying to flash things back doesn't work.

To read from the board, it must know the protocol, it must know the /dev/
file. It must know the serial addressing.

Why can't it write?

X-Ryl669, 2016-08-01:
The main reason is because of two errors you've made. First, the flash address is 0x8000000 (not 0x800000). Second, the command line parser differentiate from st-link v2 to v1 by the number of arguments, so the commands you need to perform are:
```
st-flash read code.out 0x8000000 0x007777

st-flash write code.out 0x8000000
```
And that's it, it should work (and you don't need any kext on ElCapitan)

xor-gate, 2016-08-01:
Thanks @X-Ryl669 for your reaction and verification!

xor-gate, 2016-08-03:
Thanks all for your contributions. I'm closing this now.

closed 2016-08-03 by xor-gate