stlink-org/stlink #448: every time i unplug the usb and replug again in mac : WARN src/usb.c: Couldn't find any ST-Link/V2 devices

mcheung63, 2016-07-25:
every time i unplug the usb and replug again in mac : WARN src/usb.c: Couldn't find any ST-Link/V2 devices. No way to make my mac detect nucleo after replug in second time.

- [ ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard

Stlink/v2-onboard

- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15

Nucleo-L476RG

- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)

Max OSX EI Captain , 10.11.5

- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

Nucleo-L476RG

xor-gate, 2016-07-26:
I'm not sure, have you tried upgrading the firmware of the onboard programmer with the java tools which runs under OS X? http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link005.html

Did you tried another USB cable? Probably this is something with osx and not with `texane/stlink`.

xor-gate, 2016-08-03:
Have you tried to upgrade the firmware as mentioned?

osk3091, 2016-09-01:
Try to eject ST-Link drive (in my case NODE_F411RE) from Ubuntu Devices.

xor-gate, 2016-10-29:
Please try with the latest `texane/stlink` version and programmer firmware. Feel free to comment.

kubark42, 2017-03-20:
> Did you tried another USB cable? Probably this is something with osx and not with texane/stlink.

Yup, that was my problem. Can confirm this works perfectly on macOS 10.12.2 Sierra. Thanks!

closed 2020-02-23 by Nightwalker-87