stlink-org/stlink #449: Nothing works after MCU hot unplug-plug

poglazoff, 2016-07-26:
Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropiate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropiate.

- [X] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version: don't know how to get it
- [X ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [X] Stlink tools version and/or git commit hash: a78159440544c27fb6dcd0eb9ff3480f81239e3d
- [X] Stlink commandline tool name: `st-info`, `st-flash`, probably any 
- [X] Target chip (and optional board): STM32F030C6T6

Steps to reproduce:
1. Connected my device using STM32F030C6T6 to Stlink/v2 programmer.
2. Flashed it succesfully.
3. Disconnected my device from programmer, DON'T disconnect programmer from laptop's USB.
4. Connected the device to programmer again.
5. Tried to flash it with the same command or even just got any information and get error.

Output:

```
pavel@pavel-UX31A:~/stlink$ ./st-info --descr
[!] send_recv
[!] send_recv
[!] send_recv
[!] send_recv
[!] send_recv
[!] send_recv
[!] send_recv
```

Expected/description:
`If disconnect programmer from USB and connect again error goes away.
Now I use special script from here to reset USB port before flashing: http://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line`

xor-gate, 2016-07-26:
I think this has something todo with the programmer firmware itself, as it is in a state where it has the old target remembered and you hot-plug your target. Currently the tools can not recover from this, as the programmer needs probably a jtag reset. Normaly when this occurs the tools should try to recover and reset the programmer into a know state. I have seen this kind of behaviour.

When you reset the usb port with a script it reinitialized the programmer and the stlink tools then run from a clean state. 

This is related to the older issue #244.

poglazoff, 2016-07-26:
Thank you for your answer.
1. Is it correct that only some programmers are subjected to this error, and some are not?
2. Do you know if it can be resolved with programmer reflashing?
3. Though the solution I use isn't fully correct, may it be helpful for other users? Certainly it's platform-dependent but should be short and simple on each.

xor-gate, 2016-07-26:
1. I think this is applicable to all programmers with all firmwares (but better try if it remains after upgrade)
2. I think a firmware upgrade will not resolve the issue
3. It is not platform-dependent, but a special use-case which not many users have. Still the tooling should be able to recover from this.

poglazoff, 2016-07-27:
Ok.
So what do you think? If USB reset functionality should be included in Stlink Tools or not?

xor-gate, 2016-07-27:
I'm not 100% how the reset will behave with the programmer when calling [`libusb_reset_device`](http://libusb.sourceforge.net/api-1.0/group__dev.html#ga7321bd8dc28e9a20b411bf18e6d0e9aa). The tools/library need to rescan after programmer reset and create a new context and is currently not supported. Doing a USB reset is also a workaround instead of a real fix.

poglazoff, 2016-07-27:
I don't know how often typical Stlink Tools user deals with this bug, but for me it happens dozens of times every day and takes tens minutes daily.

Also I still don't understand if it possible to create real fix (not workaround) without complete rewrite of programmers firmware (which looks almost impossible).

xor-gate, 2016-07-27:
You could try to patch and add `libusb_reset_device` and then rescan all devices or exit/rerun the tool. And see if it works for you. It is currently difficult to implement a reset mechanism so the library/tools pick-up again on the same serial number (programmer) after a USB reset. As there are two backends (mass-storage and raw usb) and alot of code duplication in the `usb.c` backend.

I'm not sure where in the flow we should place the reset currently.

xor-gate, 2017-01-14:
Not many people use it this way, and it is not a very easy fix. I will close it for now. If somebody feels like to implement this feature feel free and I will reopen this issue. Thanks for reporting!

Nightwalker-87, 2020-02-26:
Reopening this "lost" request.

Nightwalker-87, 2021-04-11:
@Ant-ON This may also be addressed in your recent PR.

Nightwalker-87, 2022-01-03:
Related to #888 and #1151.

Nightwalker-87, 2022-01-09:
@Ant-ON Would you fancy to work on this topic?

Nightwalker-87, 2022-10-23:
This issue is now closed due to inactivity.
Please also note that any version prior to `v1.7.0` is unsupported according to our [#Security Policy](https://github.com/stlink-org/stlink/security/policy#security-policy).
Should the problem persist, please retry with the latest version in the `develop` branch.
If this is the case, one should then open a new ticket.

closed 2022-10-23 by Nightwalker-87