stlink-org/stlink #450: st-flash --enable-expat can't find v2 device

ervion-n, 2016-08-02:
When I use st-flash, it complains that memory map failed. If y add --enable-expat as suggested in the readme, it warns, that it can't find v1 device and then fails. 

It is expected, that it can't find v1 device, as my device is v2-1. It should correctly detect the v2-1 device and flash it with firmware.bin.

- Programmer/board type:  ST-LINK/V2-1 
- Operating system: Linux Raspbian
- Stlink tools version: 1.2.0-66-gdf1522b git commit hash: df1522b7dbf9cbf47c1b435bd3fe0e34550e9a36
- Stlink commandline tool name: st-flash
- Target chip (and board): e.g STM32F746ZG (NUCLEO-F746ZG )

Output:
```
pi@raspberrypi:~ $ st-flash write firmware.bin 0x08000000
2016-08-02T12:55:10 INFO /usr/local/src/stlink/src/common.c: Loading device parameters....
2016-08-02T12:55:10 INFO /usr/local/src/stlink/src/common.c: Device connected is: F7 device, id 0x10016449
2016-08-02T12:55:10 INFO /usr/local/src/stlink/src/common.c: SRAM size: 0x50000 bytes (320 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
mmap() == MAP_FAILED
2016-08-02T12:55:10 ERROR /usr/local/src/stlink/src/common.c: map_file() == -1
stlink_fwrite_flash() == -1
pi@raspberrypi:~ $ st-flash write firmware.bin 0x08000000 --enable-expat
2016-08-02T12:55:15 WARN /usr/local/src/stlink/src/sg.c: Failed to find an stlink v1 by VID:PID
2016-08-02T12:55:15 ERROR /usr/local/src/stlink/src/sg.c: Could not open stlink device
```

If I use `st-util --enable-expat` and `gdb`, and enter commands `target remote :4242`, `file firmware.elf` and `load firmware.elf`, then close gdb and reset the board, then new code gets executed.

xor-gate, 2016-08-02:
Probably you look at the wrong place, you have no permission to access the device as user pi.
You need to add a udev rule so the permissions are set correctly. See:

https://github.com/texane/stlink#setting-up-udev-rules
https://github.com/texane/stlink/tree/master/etc/udev/rules.d

Could you try as a quick check to run as root and check it works and it is a permissions.

ervion-n, 2016-08-02:
No, I did add correct udev rules. But just for good measure, I executed the code as root and got the same results:
```
root@raspberrypi:/home/pi# st-flash write firmware.bin 0x08000000
2016-08-02T13:42:32 INFO /usr/local/src/stlink/src/common.c: Loading device parameters....
2016-08-02T13:42:32 INFO /usr/local/src/stlink/src/common.c: Device connected is: F7 device, id 0x10016449
2016-08-02T13:42:32 INFO /usr/local/src/stlink/src/common.c: SRAM size: 0x50000 bytes (320 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
mmap() == MAP_FAILED
2016-08-02T13:42:32 ERROR /usr/local/src/stlink/src/common.c: map_file() == -1
stlink_fwrite_flash() == -1
root@raspberrypi:/home/pi# st-flash write firmware.bin 0x08000000 --enable-expat
2016-08-02T13:42:43 WARN /usr/local/src/stlink/src/sg.c: Failed to find an stlink v1 by VID:PID
2016-08-02T13:42:43 ERROR /usr/local/src/stlink/src/sg.c: Could not open stlink device
root@raspberrypi:/home/pi# 
```
It is also worth noting, that `st-util` found the devince without problems, and flashing and debugging via gdb worked.

xor-gate, 2016-08-02:
Are you sure st-util is not running while st-flash is invoked?

ervion-n, 2016-08-02:
yes

xor-gate, 2016-08-02:
I think memory mapping the firmware file under raspberry is broken. Could you try to remove everything from mmap.h except this and recompile?

https://github.com/texane/stlink/blob/e454fe98191095333258cfcfa813585086f669ed/include/stlink/mmap.h#L5

Did you build with cmake or autoconf?

ervion-n, 2016-08-03:
Just double checked, it happens exactly like i described before when st-util is not running.

ervion-n, 2016-08-03:
I used cmake.
I commented out all lines expect the 5., and re-built everything.

Everything works perfectly! :D

xor-gate, 2016-08-03:
I think the binary interface of mmap flags is different. We should not implement our own mmap function as this can be problematic. I will fix this as soon as I have time.

Thanks for your feedback @ervion-n.

closed 2016-08-03 by xor-gate

xor-gate, 2016-08-03:
@ervion-n could you test with latest master?

ervion-n, 2016-08-03:
Yes, latest master works perfectly, thank you!

xor-gate, 2016-08-03:
@ervion-n thank you for your verification!

commit 57cf59a8 (2016-08-03): cmake: Add proper detection of sys/mman.h. Fixes #450