stlink-org/stlink #451: Unknown Chip ID 0xe0042000

karora, 2016-08-06:
How can I help to fix "stlink/src/common.c: unknown chip id! 0xe0042000" ?

- [ ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
ST-LINK / V2
B 2015 30

This looks like a normal standard Stlink/v2, as far as I can see from the various pictures on the internet.

- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
Unknown.  Perhaps the lsusb output is useful?
```
$ lsusb -v
 ...
Bus 001 Device 011: ID 0483:3748 STMicroelectronics ST-LINK/V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0x3748 ST-LINK/V2
  bcdDevice            1.00
  iManufacturer           1 STMicroelectronics
  iProduct                2 STM32 STLink
  iSerial                 3 HÿopQH34g
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              4 ST Link
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)

```

- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
Linux, Debian Sid, x86-64

- [ ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
git ee2681ecebc8b2284cab02e1dd40536dfb91fc51

- [ ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util` 
st-flash (& others)

- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)
The top of the chip is printed in full with:
  STM32F405
  RGT6
  7B357 VQ
  PHL 7B 527

A as-detailed description possible of the problem with debug output when available.

Output:

```
$ ./st-flash --debug read out.bin 0x8000000 4096
2016-08-06T17:43:34 DEBUG stlink/src/common.c: stlink current mode: debug (jtag or swd)
2016-08-06T17:43:34 DEBUG stlink/src/common.c: stlink current mode: debug (jtag or swd)
2016-08-06T17:43:34 DEBUG stlink/src/common.c: *** stlink_reset ***
2016-08-06T17:43:34 DEBUG stlink/src/common.c: *** looking up stlink version
2016-08-06T17:43:34 DEBUG stlink/src/common.c: st vid         = 0x0483 (expect 0x0483)
2016-08-06T17:43:34 DEBUG stlink/src/common.c: stlink pid     = 0x3748
2016-08-06T17:43:34 DEBUG stlink/src/common.c: stlink version = 0x2
2016-08-06T17:43:34 DEBUG stlink/src/common.c: jtag version   = 0x17
2016-08-06T17:43:34 DEBUG stlink/src/common.c: swim version   = 0x4
2016-08-06T17:43:34 INFO stlink/src/common.c: Loading device parameters....
2016-08-06T17:43:34 DEBUG stlink/src/common.c: *** stlink_core_id ***
2016-08-06T17:43:34 DEBUG stlink/src/common.c: core_id = 0x2ba01477
2016-08-06T17:43:34 DEBUG stlink/src/common.c: *** stlink_read_debug32 e0042000 is 0xe0042000
2016-08-06T17:43:34 WARN stlink/src/common.c: unknown chip id! 0xe0042000
2016-08-06T17:43:34 DEBUG stlink/src/common.c: *** stlink_close ***
$
```

Expected/description:

I'd hoped the chip ID would be supported :-)

Thank you,
Andrew McMillan

karora, 2016-08-06:
OK, looks like this is some kind of standard error that I have to fix by finding someone with a Windows system, and reprogramming the adapter in some way?  It would be great if there were a clear statement somewhere about what needs to be done.  Unfortunately I don't know anyone using  Windows who is geeky enough to be able to help (or who I could borrow their computer) so it would be *amazing* if there were some instructions about how to fix this from Linux...

Or OSX - I guess I can find geeks using OSX as well, or borrow a laptop from one.

karora, 2016-08-06:
Hunting through the issues a little more, I see the patch in issue #428 and given the way this board is designed to work, I guess that's the problem.

After applying that patch I can now:

```
$ ./st-flash --reset read out.bin 0x8000000 4096
2016-08-06T20:33:27 INFO /home/karora/projects/HUB/stlink/src/common.c: Loading device parameters....
2016-08-06T20:33:27 INFO /home/karora/projects/HUB/stlink/src/common.c: Device connected is: F4 device, id 0x10076413
2016-08-06T20:33:27 INFO /home/karora/projects/HUB/stlink/src/common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
```

So perhaps the issue is that when a chipid of 0xe0042000 is detected the error message should suggest trying again with a --reset option (especially after this patch is added).

Thanks,
Andrew.

xor-gate, 2016-08-07:
Fixed by merge of #430.

maxim25111, 2018-05-04:
I have this same problem. After installing on linux libusb-1.0.0 for :i386 on machine 64bit, I get message 
unknown chip id! 0xe0042000.
Someone said that stm32 is not recognizable, I check one more time cable connection from stlink to my device, and I did something terible, I put cable from swclk to swdio and no wonder that Arduino couldn't communicate with sdm32f1. 
 It fix for me. And I didn't need hold reset button. Is just wrong cable connection. :)

closed 2020-03-17 by Nightwalker-87