stlink-org/stlink #453: Wrong memory size hardcoded for STM32F7xxx in memory_map_template_F7

zulusw, 2016-08-09:
The issue shows up when using STM32F769. In conjunction with st-utils.
(gdb0
Output:

```
(gdb) target extended-remote :4242
Remote debugging using :4242
2016-08-08T23:45:14 INFO .../stlink/src/gdbserver/gdb-server.c: Found 8 hw breakpoint registers
2016-08-08T23:45:14 INFO .../stlink/src/gdbserver/gdb-server.c: GDB connected.
0x08000340 in ?? ()
(gdb) x/x 0x2004f000
0x2004f000:	0x0a843b1c
(gdb) x/x 0x20050000
0x20050000:	Cannot access memory at address 0x20050000
```

Expected/description:
`I should be able to dump memory all the way up to 0x2007ffff`

I can look into a fix if you want.

zulusw, 2016-08-09:
I believe (part of) the issue is this line in memory_map_template_F7:
```
    "  <memory type=\"ram\" start=\"0x20000000\" length=\"0x50000\"/>"      // sram
```

brabo, 2017-11-11:
I believe this was fixed in https://github.com/texane/stlink/commit/1118f8ed498a1c0eff280d36c97a14248b91b73b#diff-bb08b95a80a0b60cd043251a0e2a559a
Cheers!

Nightwalker-87, 2020-02-21:
Please verify if the problem still exists in Release v1.6.0.

zulusw, 2020-02-25:
I don't have a board handy at the moment but it must have been fixed by my
change as suggested by @brabo. I had reported the bug and also provided a
fix for it.

Nightwalker-87, 2020-02-25:
Thank you for coming back to this topic. I'll check as soon as I am done with other reviewing tasks. :+1:

Nightwalker-87, 2020-03-17:
Fixed in PR #456.

closed 2020-03-17 by Nightwalker-87