stlink-org/stlink #457: Help downloading code from STM32 via `st-flash`  (getting `write() != aligned_size' & 'stlink_fread() == -1`)

jacobq, 2016-08-18:
I am trying to figure out how to use the `st-flash` command-line flash loader utility. After building these `stlink` tools I can see that the board is attached (`st-info --probe` outputs "Found 1 stlink programmers", `st-info --descr` outputs "F334 device"), but when I try to download the example code that comes loaded onto the board into a hex file by running `st-flash read /dev/sg2 demo.hex ` it fails with:
```
2016-08-18T16:30:18 INFO stlink/src/common.c: Loading device parameters....
2016-08-18T16:30:18 INFO stlink/src/common.c: Device connected is: F334 device, id 0x10016438
2016-08-18T16:30:18 INFO stlink/src/common.c: SRAM size: 0x3000 bytes (12 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 2048 bytes
write() != aligned_size
stlink_fread() == -1
``` (absolute path prefix removed)
Could someone help me figure out what's wrong? Perhaps I am not using the command properly? I could not find much documentation explaining how to use it, and I also have not done much with STM32 devices before. 

Details about my setup are given below.

* ST-Link/v2-1 built-into STM32F3348-DISCOVERY board (http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-discovery-kits/32f3348discovery.html)
* Loaded latest firmware from ST (STSW-LINK007 v2.27.17 --> V2.J27.M15 shown in updater tool)
* Running Linux (Debian stretch/sid x86_64)
* Compiled stlink tools at commit `1ca526ca4d5a1c0658812720690ebefdeb259e3d`
* Target device is  STM32F334C8T6 (STM32F3348-DISCO)

```
$ st-flash --debug --reset read /dev/sg2 demo.hex 
2016-08-18T16:08:15 DEBUG stlink/src/common.c: stlink current mode: mass
2016-08-18T16:08:15 DEBUG stlink/src/common.c: stlink current mode: mass
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_enter_swd_mode ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_jtag_reset ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_reset ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** looking up stlink version
2016-08-18T16:08:15 DEBUG stlink/src/common.c: st vid         = 0x0483 (expect 0x0483)
2016-08-18T16:08:15 DEBUG stlink/src/common.c: stlink pid     = 0x374b
2016-08-18T16:08:15 DEBUG stlink/src/common.c: stlink version = 0x2
2016-08-18T16:08:15 DEBUG stlink/src/common.c: jtag version   = 0x1b
2016-08-18T16:08:15 DEBUG stlink/src/common.c: swim version   = 0xf
2016-08-18T16:08:15 INFO stlink/src/common.c: Loading device parameters....
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_core_id ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c: core_id = 0x2ba01477
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_read_debug32 10016438 is 0xe0042000
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_read_debug32 ffff0040 is 0x1ffff7cc
2016-08-18T16:08:15 INFO stlink/src/common.c: Device connected is: F334 device, id 0x10016438
2016-08-18T16:08:15 INFO stlink/src/common.c: SRAM size: 0x3000 bytes (12 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 2048 bytes
2016-08-18T16:08:15 DEBUG stlink/src/common.c: stlink current mode: debug (jtag or swd)
2016-08-18T16:08:15 DEBUG stlink/src/common.c: stlink current mode: debug (jtag or swd)
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_jtag_reset ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_reset ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_force_debug_mode ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_status ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c:   core status: halted
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_read_mem32 ***
data_len = 2048 0x800
 d5 46 00 08 d9 46 00 08 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 dd 46 00 00 e1 46 00 08 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e5 46 00 00 e9 46 00 08 00 00 00 08 ff 42 00 00 cd 42 00 08 fd 42 00 08 ed 46 00 08 f1 46 00 08 f5 46 00 08 f9 46 00 08 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fd 46 00 00 10 b4 00 08 00 28 40 b2 0a 01 09 d5 10 f0 40 b2 df f8 0f 03 1b 19 c0 4b 04 2c 03 f8 0a 01 04 e0 df f8 40 b2 c2 54 b4 3b 70 47 10 bc 04 00 10 b5 b0 f1 60 1e 01 d3 80 7f 11 e0 01 20 df f8 60 1e 08 60 9c 1b 5f f0 0f 21 ff f7 ff 30 00 20 da ff 90 1b df f8 07 20 08 60 8c 1b df f8 00 20 08 60 10 b5 10 bd 00 24 86 b0 df f8 00 20 cd e9 7c 1b 00 20 02 01 cd e9 00 21 df f8 04 01 01 f0 38 0e df f8 16 fc 40 68 30 0e 30 1e df f8 f1 f0 b0 fb cc ff ff f7 df f8 00 20 08 70 54 1b 01 f0 00 20 01 20 0a ff 07 ff 01 f0 01 f0 02 20 03 20 04 ff 01 ff 01 f0 00 20 01 21 33 ff 01 f0 01 f0 00 20 01 28 80 ff 00 20 25 d1 7b ff 01 f0 14 d1 01 28 01 f0 03 20 41 f6 12 ff 00 f0 58 30 03 20 5a fd 15 ff 01 f0 01 f0 02 20 41 f6 08 ff 00 f0 58 30 02 20 50 fd 0b ff 01 f0 03 20 e5 e7 07 ff 01 f0 01 f0 02 20 01 20 04 ff e8 1a df f8 00 f0 08 80 00 f0 94 fd 00 f0 c8 fa df f8 bf fc 00 78 d8 0a 08 d0 00 28 00 f0 02 28 c0 f0 fc 80 03 28 9b 80 51 82 00 f0 00 22 f1 e7 00 31 5f f4 bc 0a df f8 c1 ff 01 f0 01 f0 00 20 01 20 e2 fe df fe 01 f0 01 f0 02 20 03 20 dc fe d9 fe 01 f0 df f8 00 20 08 70 9c 1a df f8 40 21 01 f0 90 0a 00 20 7f ff 27 ff 01 f0 38 d1 01 28 88 0a df f8 01 28 00 78 03 20 33 d1 b9 fe 01 f0 01 f0 00 20 00 20 c0 fe 5c 1a df f8 dd e9 08 70 02 f0 02 01 00 ee 44 fb 00 f0 10 0a dd e9 ed fc 00 20 02 23 58 1a df f8 81 fb 02 f0 02 01 cd e9 7a 70 4f f4 e9 fc 00 f0 02 01 dd e9 df f8 00 22 02 f0 20 3a ce d8 e8 fb df f8 00 20 cd e9 14 1a 03 20 02 01 97 fe 01 f0 10 70 42 f2 d5 fc 00 f0 00 20 c1 e7 e9 fe 01 f0 36 d1 01 28 0c 0a df f8 00 28 00 78 03 20 31 d1 85 fe 01 f0 01 f0 00 20 00 20 78 fe e0 19 df f8 dd e9 08 70 02 f0 02 01 00 ee 06 fb 00 f0 10 0a dd e9 af fc 00 20 02 23 e0 19 df f8 43 fb 02 f0 02 01 cd e9 7a 70 4f f4 ab fc 00 f0 02 01 dd e9 00 23 00 22 d3 fb 02 f0 00 20 cf d8 cd e9 00 21 00 20 02 01 5b fe 01 f0 10 70 42 f2 99 fc 00 f0 5a e7 c3 e7 5f f4 00 22 df f8 00 31 01 f0 8c 09 01 20 2a ff 8c 19 df f8 00 20 08 70 6c 19 df f8 02 01 cd e9 e4 0a 9f ed 7a fc 00 f0 df f8 00 20 08 70 5c 19 01 f0 00 20 df f8 30 fe 00 68 70 09 77 fc 00 f0 01 f0 00 20 02 20 32 fe 25 fe 01 f0 5c 09 df f8 00 f0 00 68 02 20 6c fc 27 fe 01 f0 01 f0 03 20 df f8 1a fe 00 68 44 09 61 fc 00 f0 01 f0 03 20 01 20 1c fe 0f fe 01 f0 30 09 df f8 00 f0 00 68 01 20 56 fc 11 fe 01 f0 df f8 01 20 08 70 0c 19 df f8 40 21 01 f0 00 09 df f8 b4 fe 00 68 0c 09 df f8 40 08 08 60 04 19 00 09 df f8 01 28 00 68 49 f6 04 d1 df f8 40 40 08 60 f0 18 01 20 fb e6 dc 18 df f8 00 20 08 70 bc 18 df f8 02 01 cd e9 01 f0 00 20 01 20 ea fd e7 fd 01 f0 01 f0 02 20 03 20 e4 fd e1 fd 01 f0 df f8 00 20 08 70 ac 18 df f8 40 21 01 f0 a0 08 00 20 87 fe 8c 18 df f8 00 20 08 70 c7 fd 01 f0 01 f0 03 20 50 20 ce fd 9c 18 df f8 28 20 08 60 94 18 df f8 00 20 48 60 8c 18 df f8 a0 20 88 60 84 18 df f8 78 20 c8 60 7c 18 df f8 00 20 08 61 78 18 df f8 00 20 08 60 70 18 df f8 00 20 48 60 68 18 df f8 00 20 88 60 60 18 df f8 00 20 c8 60 58 18 df f8 df f8 08 61 93 49 54 08 01 22 01 60 00 31 5f f4 28 08 df f8 77 fe 01 f0 40 08 df f8 01 60 8e 49 38 08 df f8 00 0a 90 ed 40 0a b5 ee 10 fa f1 ee df f8 6d db 00 78 f8 07 68 d1 00 28 1c 08 df f8 00 0a 90 ed a2 0a df ed 60 0a b4 ee 10 fa f1 ee 00 20 23 db 00 98 00 90 1e d2 05 28 df f8 00 98 51 f8 f0 17 49 1c 20 10 e8 27 df f8 20 10 42 f8 00 99 00 98 dc 27 df f8 21 10 52 f8 80 52 4f f4 40 f6 51 43 b1 fb e4 42 df f8 f2 f1 42 f8 c8 27 00 98 20 10 00 90 40 1c df f8 dd e7 90 ed bc 07 df ed 00 0a 80 ee 8b 0a 10 ee 20 0a 02 f0 10 0a df f8 d6 fa df f8 a8 27 02 f0 a8 37 02 f0 f8 fa 01 90 d2 f9 02 f0 01 98 41 ec ca fa 02 f0 10 0b 51 ec f8 fb 02 f0 10 0b 01 90 c6 f9 01 0a 9d ed c0 0a bc ee 10 0a 10 ee 73 fb 00 f0 6c 07 df f8 02 f0 00 68 df f8 b4 fa df f8 64 27 02 f0 68 37 02 f0 fa f9 df f8 b0 f9 08 60 50 17 03 20 88 e7 13 fd 01 f0 01 f0 00 20 df f8 1a fd 00 21 3c 07 df f8 01 60 90 ed 34 07 df ed 00 0a b4 ee 6a 0a f1 ee 60 0a 6d d5 10 fa f4 06 df f8 00 28 00 78 df f8 68 d1 90 ed 14 07 df ed 00 0a b4 ee 63 0a f1 ee 60 0a 23 db 10 fa 00 90 00 20 05 28 00 98 00 98 1e d2 ec 16 df f8 20 10 51 f8 df f8 49 1e 42 f8 e0 26 00 98 20 10 df f8 00 99 52 f8 d4 26 4f f4 21 10 51 43 80 52 e4 42 40 f6 f2 f1 b1 fb c4 26 df f8 20 10 42 f8 40 1c 00 98 dd e7 00 90 b8 06 df f8 00 0a 90 ed 4a 0a df ed 20 0a 80 ee 10 0a 10 ee 53 fa 02 f0 a4 26 df f8 a4 36 df f8 75 fa 02 f0 4f f9 02 f0 01 98 01 90 47 fa 02 f0 10 0b 41 ec 75 fb 02 f0 10 0b 51 ec 43 f9 02 f0 9d ed 01 9 [... cut ...] 00 60 38 90 20 20 00 50 4f f4 00 20 39 90 30 a9 3a 90 08 05 df f8 38 fb 02 f0 df f8 01 21 02 f0 fc 04 01 21 52 fb 00 30 5f f4 68 fb 01 f0 00 90 10 20 8d f8 02 20 00 20 04 00 06 00 8d f8 8d f8 03 20 00 20 05 00

write() != aligned_size
stlink_fread() == -1
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_exit_debug_mode ***
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2016-08-18T16:08:15 DEBUG stlink/src/common.c: *** stlink_close ***
```

xor-gate, 2016-08-20:
@jacobq it is currently not possible to load hex files into the target with the `st-flash` tool as there is no code to read the contents of hex files.

xor-gate, 2016-08-20:
Closing due to duplicate of issue #110.

closed 2020-04-14 by Nightwalker-87