stlink-org/stlink #458: Writing option bytes of STM32F091RCT6

vladiks, 2016-08-19:
Hello, I am trying to write option bytes of STM32F091RCT6 using Stlink/v2. Bytes start at address 0x1FFF F800 and occupie 1024bytes. I need to write just first four bytes, but writing 1024 would not be problem.

I thought, I could do this using command:

```
st-flash write optionBytes.bin 0x1FFFF800
```
but the output is:
```
2016-08-19T11:31:25 INFO src/common.c: Loading device parameters....
2016-08-19T11:31:25 INFO src/common.c: Device connected is: F09X device, id 0x10006442
2016-08-19T11:31:25 INFO src/common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
```

St-flash version is 1.2.0. OS Linux raspberry  4.1.7+ armv6l

Basic question - is it ven possible to write option bytes using this tool?

Thanks for help

xor-gate, 2016-08-20:
I'm not 100% sure, but writing outside the flash/ram region is currently not supported and untested.

xor-gate, 2016-08-20:
This issue is related to #333.

Nightwalker-87, 2020-03-20:
@vladiks: I believe this feature is still missing. Any update on this?

vladiks, 2020-03-26:
@Nightwalker-87 I am sorry, but I can not give you any feedback, cause the project for which i needed this, is already outside my reach.

Nightwalker-87, 2020-03-26:
@vladiks: Thanks for coming back to this. Never mind, it was an answer to expect though due to the long timespan. That's ok. Likely it wasn't very clever not to stick to it when it was relevant for you in earlier days. I'm sorry to see that, but just got to know about this recently during maintenance tasks. I'm one of the new main contributors of this project.

Anyway I'll leave this open for now, until someone else is able to verify or I hopefully run across a commit that directly resolves this issue.

Nightwalker-87, 2020-04-16:
@grevaillot: Would it be possible to add that to the current commit? Since your rewrite this should only be a few lines or am I wrong here? I assume though you may not be able to test on real hardware...

grevaillot, 2020-04-17:
yeah, i've got some f091 from a project somewhere, but i think the flash option stuff is the same for all f0 and f1, Adding read is easy, adding write need a slight more complicated logic : apparently, write accesses are to be done 16bit wide, and option flash need to be erased first, so need to backed up and patched if we want to write only the first option reg. I'll have a look.

Ant-ON, 2020-04-17:
@grevaillot 
Yes, it's same

jdbruijn, 2020-09-02:
I have hardware I can test this with if needed. Unfortunately I don't have time ATM to look into the code and potentially fix it. Would be happy to help test it if there is a branch/commit/patch that can be used to test with. I have an STM32F205RG of which I want to write the option bytes, readout protection (RDP) specifically. I have found that I can write the option bytes from the embedded software so it isn't something that is essential, but as said happy to help test this feature!

stappersg, 2020-09-02:
On Wed, Sep 02, 2020 at 02:40:18AM -0700, Jeroen de Bruijn wrote:
> Would be happy to help test it
> if there is a branch/commit/patch
> that can be used to test with.

"git fork" and reproduce the problem ...

ApiumJacob, 2021-04-01:
In an effort to not create duplicate tickets, I too would like to be able to set and read option bytes.  Specifically the RDP byte.

Ant-ON, 2021-04-01:
@ApiumJacob `st-flash` support the option bytes (RDP is the part of this) write for F4 target. How to write see https://github.com/stlink-org/stlink/blob/develop/doc/tutorial.md

Nightwalker-87, 2021-04-02:
Please refer to the now linked PR for a proposed solution.
There is a possibility for testing with a STM32F0-Discovery board within the next two weeks.

PR #1112 (2021-05-10): Support for writing option bytes on STM32F0/F1/F3
Implementation to write option bytes for STM32F0/F1/F3.

(Closes #346, Closes #458, Closes #808, Closes #1084)
closed 2021-05-10 by Nightwalker-87