stlink-org/stlink #461: Custom continuous integration with Jenkins

xor-gate, 2016-09-08:
We can continue discussion in this issue from PR #444.

nekromant, 2016-09-09:
@xor-gate FYI: I've set up a branch pipeline test with a sample Jenkinsfile here:

http://github.com/nekromant/stlink (see pipeline-test)
And a relevant project in ci
https://jenkins.ncrmnt.org/job/GithubCI/job/stlink-branch-ci/

It's not yet working, I'll be playing with DSL  when I have a spare minute. 

It looks like scan-build you've been missing there was because of this bug: 
https://issues.jenkins-ci.org/browse/JENKINS-29872

Should be trivial to fix in scan-build plugin if they haven't done already.

xor-gate, 2017-01-14:
IMHO you may just remove the pipeline build. If we need this in the future we could always figure out something. Thank you for your time @nekromant !

Nightwalker-87, 2020-03-16:
Is there still any demand for this feature request?

Nightwalker-87, 2020-03-30:
Travis CI is being used for continuous integration since Apr 2016 (commit ca08bed617efc86a7765554c14a448b731f97eb3).
Since no further demand has been registered for this feature, this ticket is now finally closed.

closed 2020-03-30 by Nightwalker-87